espressif/esp-idf #4396: cmake menuconfig missing entries

paoloach, 2019-11-25:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      it doesn't matter
- Kit version:  it doesn't matter
- Module or chip used:   it doesn't matter
- IDF version: v4.1-dev-1086-g93a8603c5
- Build System:    CMake
- Compiler version: crosstool-NG esp32-2019r1-rc2) 8.2.0
- Operating System:     Linux
- Power Supply:        it doesn't matter

## Problem Description

Running menuconfig from the cmake build folder, the Serial Flasher config miss the first 3 entries:

- default serial port
- default baud rate
- use compressed upload

### Expected Behavior
These 3 entries shuold be present

### Actual Behavior
These 3 entries are not present
### Steps to repropduce

got into the folder examples/get-started/hello_world
mkdir build
cd build
cmake ../
make menuconfig
select the voice "Serial Flasher config"

projectgus, 2019-11-26:
Hi @paoloach ,

When we moved from GNU Make to CMake, the serial flasher settings were removed from the project configuration. It was a frequent source of (legitimate) complaints that changing serial port settings should not cause the entire project to rebuild.

The Getting Started guide explains how to set the serial port if working with idf.py:
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#step-9-flash-onto-the-device

If not using idf.py then you can set the serial port details via environment variables ESPPORT & ESPBAUD:
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#flashing-with-ninja-or-make

There is a minor bug in the CMake build system though, which is that the "Compressed Upload" config option is still being used to generate the flashing commands, even though it's disabled in CMake. Will fix ASAP (this option was removed because there is not really any reason to disable compressed uploads in normal operation).

paoloach, 2019-11-26:
Hi @projectgus,

thank for the explanations.

So, I think I can close this issue.

Regards,
Paolo

closed 2019-11-26 by paoloach