espressif/esp-idf #4397: calling esp_tls_conn_write multiple time doesn't works.

B-Temia, 2019-11-25:
Hi,

I need to upload files located in the sd card (between 1 and 10 megs) using esp_tls_conn_write. I read one file at a time by chunk and post it to the .net core web api. One file, one upload. 

All the required headers are there including Content_length. My .net core web api receive the post and returns the response without problem and save it to blob storage.

I start this test project from the sample projet https_request, but I move the esp_conn_write in a function to be able to reuse it more than once 

```c
static int send_chunk(struct esp_tls *tls, const void *data, size_t datalen)
{
    int ret;
    size_t written_bytes = 0;
    do
    {
        ret = esp_tls_conn_write(tls, data + written_bytes, datalen - written_bytes);
        if (ret >= 0)
        {
            written_bytes += ret;
        }
        else if (ret != MBEDTLS_ERR_SSL_WANT_READ && ret != MBEDTLS_ERR_SSL_WANT_WRITE)
        {
            ESP_LOGE(TAG, "esp_tls_conn_write  returned 0x%x", ret);
            return -1;
        }
    } while (written_bytes < datalen);

    return 0;
}
```

I call this function many times without problem to set diffent headers from another function.

```c
        char *header_start = "POST " WEB_URL " HTTP/1.0\r\n"
                             "Host: " WEB_SERVER "\r\n"
                             "User-Agent: esp-idf/1.0 esp32\r\n";
        if (send_chunk(tls, header_start, strlen(header_start)) == -1)
        {
            goto exit;
        }

        char *header_content_type = "Content-Type: text/html\r\n";
        if (send_chunk(tls, header_content_type, strlen(header_content_type)) == -1)
        {
            goto exit;
        }

        char *header_filename = "X-Filename: 00000000.bin\r\n";
        if (send_chunk(tls, header_filename, strlen(header_filename)) == -1)
        {
            goto exit;
        }

        FILE *fh = fopen("/sdcard/00000000.bin", "r");
        if (!fh)
        {
            return;
        }

        unsigned long fsize;

        fseek(fh, 0, SEEK_END);
        fsize = ftell(fh);
        rewind(fh);

        char header_content_length[40];
        sprintf(header_content_length, "Content-Length: %lu\r\n", fsize);
        if (send_chunk(tls, (const char *)header_content_length, strlen(header_content_length)) == -1)
        {
            goto exit;
        }

        char *header_end = "\r\n";
        if (send_chunk(tls, header_end, strlen(header_end)) == -1)
        {
            goto exit;
        }
```

The problem begin when I send the file content by reading the file by chunk (I tested different chunk size and it is always the same problem). I also add vTaskDelay(... in the do.

```c
        char *chunk = malloc(SCRATCH_BUFSIZE);
        size_t chunksize;
        do
        {
            chunksize = fread(chunk, 1, SCRATCH_BUFSIZE, fh);
            if (send_chunk(tls, chunk, chunksize))
            {
                goto exit;
            }
        } while (chunksize != 0);

        fclose(fh);
```

When I open the file in vscode using hexdump for VSCode, only the first part of the file is good (first read of the file), everything else is 00 in hexa.

I did another test with long string and the first few calls to send_chunk finish good but not all.

```c
        unsigned long fsize;
        int iterationTest = 20;
        char *body_long_content = "write something long here...";
        fsize = strlen(body_long_content) * iterationTest;

        char header_content_length[40];
        sprintf(header_content_length, "Content-Length: %lu\r\n", fsize);
        if (send_chunk(tls, (const char *)header_content_length, strlen(header_content_length)) == -1)
        {
            goto exit;
        }

        char *header_end = "\r\n";
        if (send_chunk(tls, header_end, strlen(header_end)) == -1)
        {
            goto exit;
        }

        // send the string many times
        for (int i = 0; i < iterationTest; i++)
        {
            if (send_chunk(tls, body_long_content, strlen(body_long_content)) == -1)
            {
                goto exit;
            }
        }
```

Thanks,

AdityaHPatwardhan, 2019-11-27:
@B-Temia , In the example when you read the file by chunks , have you confirmed that all the chunks are properly read and the error is only in sending the file.
Also, When you call send_chunk function repetitively , can you print the bytes written on the tls connection on monitor

AdityaHPatwardhan, 2019-12-02:
@B-Temia , is your problem solved ?

B-Temia, 2019-12-02:
@AdityaHPatwardhan 

I tested the same functionnalty with the http_request sample and it works with this one. I will have to test it again with the https_request and make sure I read the stream correctly on server side since the body is not send fully when the call arrived.

The reason why I do it, it is because I try to figure what is the problem with another bug with esp-azure and the upload to blob functionnality. 

Thanks!!

AdityaHPatwardhan, 2019-12-27:
@B-Temia , can we close this issue ?

mahavirj, 2020-01-03:
Closing due to lack of response. Please feel free to re-open if issue still exists.

closed 2020-01-03 by mahavirj