espressif/esp-idf #4398: Backslashes of windows paths causes CMake to fail in v4.0

cima, 2019-11-25:
## Environment

- Development Kit: N/A
- Kit version (for WroverKit/PicoKit/DevKitC): N/A
- Module or chip used: ESP32 LORA TT GO
- IDF version: release/v4.0
- Build System: CMake
- Compiler version: (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Windows 10 64-bit
- Power Supply: USB

## Problem Description
Using ESP-IDF v4.0 and environment installed using _esp-idf-tools-setup-2.0.exe_ gives me no chance to even perform `idf.py menuconfig`. It always crashes CMake execution on *Invalid escape sequence \U*. Paths to various tools (python, IDF_PATH, ...) are generated by windows batch scripts and contains backslashes. However sometimes these are processed by CMake and CMake treats backslashes as an escape sequence which obviously fails. I have tried to fight it by resetting some variables like *IDF_PATH* to forward slashed paths, but then different paths appeared with backslashes. So manually fighting is not a solution.

### Expected Behavior

CMake is configured in a way it survives windows path conventions.

### Actual Behavior

CMake crashes at some points when windows path with backslashes is present.

### Steps to repropduce

1. Checkout _esp-idf from_ github from branch _release/v4.0_ to e.g. _c:/Users/z608328/ESP32/esp-idf/_
2. Install toolchain using _esp-idf-tools-setup-2.0.exe_
3. Select prviously checked out esp-idf
4. execute
```
cd c:\Users\z608328\ESP32\esp-idf 
export.bat
set CC=xtensa-esp32-elf-gcc.exe
set CXX=xtensa-esp32-elf-g++.exe 
cd c:\src\incubator\myProject\
idf.py menuconfig
```

### Code to reproduce this issue

```CMakeLists.txt
cmake_minimum_required(VERSION 3.5)

set(CMAKE_MODULE_PATH $ENV{IDF_PATH})
set(CMAKE_SYSTEM_NAME Generic)

project(CIMA-ESP32-AZURE)
```

## Debug Logs

```
c:\src\incubator\myProject>idf.py menuconfig
Checking Python dependencies...
Python requirements from c:\Users\z608328\ESP32\esp-idf\requirements.txt are satisfied.
Executing action: menuconfig
Running cmake in directory c:\src\incubator\myproject\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 c:\src\incubator\myproject"...
Warn about uninitialized values.
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- Check for working C compiler: C:/Users/z608328/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
CMake Error at C:/src/incubator/myProject/build/CMakeFiles/CMakeTmp/CMakeLists.txt:2 (set):
  Syntax error in cmake code at

    C:/src/incubator/myProject/build/CMakeFiles/CMakeTmp/CMakeLists.txt:2

  when parsing string

    c:\Users\z608328\ESP32\esp-idf

  Invalid character escape '\U'.

CMake Error at C:/Users/z608328/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/CMakeTestCCompiler.cmake:37 (try_compile):
  Failed to configure test project build system.
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/src/incubator/myProject/build/CMakeFiles/CMakeOutput.log".
See also "C:/src/incubator/myProject/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1
```

igrr, 2019-11-26:
Hi @cima, could you please try changing the project CMakeLists file to look like this, and see if that helps?

https://github.com/espressif/esp-idf/blob/93a8603c545fb8e54741d6685146e2f3b874378d/examples/get-started/hello_world/CMakeLists.txt#L5-L6

cima, 2019-11-26:
Adding the `include($ENV{IDF_PATH}/tools/cmake/project.cmake)` line alone didn't changed the behavior.  However removal of `set(CMAKE_MODULE_PATH $ENV{IDF_PATH})` make my example project to configure just fine.

```
cmake_minimum_required(VERSION 3.5)
set(CMAKE_SYSTEM_NAME Generic)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(CIMA-ESP32-AZURE)
```

I have a bigger project that I am migrating from old C++ to C++17 and to v4.0 so I reflect these changes in its CMake files and see if the same approach make it compilable in v4.0.

cima, 2019-11-27:
My bigger project now compiles fine. It turned out that `include($ENV{IDF_PATH}/tools/cmake/project.cmake)` was already there. So the root cause was presence of `set(CMAKE_MODULE_PATH $ENV{IDF_PATH})`. After its removal even the bigger project compiled just fine.

igrr, 2019-12-23:
Okay, thanks, will consider this resolved.

closed 2019-12-23 by igrr

kashyap2690, 2020-06-27:
The latest hello_world is still getting this error in windows 10.
@igrr 
It does not have set(CMAKE_MODULE_PATH $ENV{IDF_PATH}) as @cima mentioned so I can't find any workaround.

Any help?

igrr, 2020-06-27:
@kashyap2690 Please open a new issue and fill in the information requested in the template, we'll try to help.