espressif/esp-idf #4399: Provisioning Documentation.

monkeytronics, 2019-11-25:
Is there any documentation describing how to provision the SSID and Password over BLE & AP. I know there is an Android and iOS app available for reference. But for those who prefer to work in Ionic, this is not very useful. 

Scanning and finding the device  (BLE provisioning) is working fine, as expected. From here, it would be great to have a clear list of the steps required, ideally with as much detail as possible, including the expected responses at each stage.

monkeytronics, 2019-11-26:
For completeness, here's the details of what I see when I connect.

```
 [console.log]: {
   "name": "PROV_F00574",
   "id": "24:6F:28:F0:05:76",
   "advertising": {},
   "rssi": -55,
   "services": [
     "1801",
     "1800",
     "12345678-90ab-cdef-fedc-ba0987654321"
   ],
   "characteristics": [
     {
       "service": "1801",
       "characteristic": "2a05",
       "properties": [
         "Indicate"
       ],
       "descriptors": [
         {
           "uuid": "2902"
         }
       ]
     },
     {
       "service": "1800",
       "characteristic": "2a00",
       "properties": [
         "Read"
       ]
     },
     {
       "service": "1800",
       "characteristic": "2a01",
       "properties": [
         "Read"
       ]
     },
     {
       "service": "1800",
       "characteristic": "2aa6",
       "properties": [
         "Read"
       ]
     },
     {
       "service": "12345678-90ab-cdef-fedc-ba0987654321",
       "characteristic": "1234ff50-90ab-cdef-fedc-ba0987654321",
       "properties": [
         "Read",
         "Write"
       ],
       "descriptors": [
         {
           "uuid": "2901"
         }
       ]
     },
     {
       "service": "12345678-90ab-cdef-fedc-ba0987654321",
       "characteristic": "1234ff51-90ab-cdef-fedc-ba0987654321",
       "properties": [
         "Read",
         "Write"
       ],
       "descriptors": [
         {
           "uuid": "2901"
         }
       ]
     },
     {
       "service": "12345678-90ab-cdef-fedc-ba0987654321",
       "characteristic": "1234ff52-90ab-cdef-fedc-ba0987654321",
       "properties": [
         "Read",
         "Write"
       ],
       "descriptors": [
         {
           "uuid": "2901"
         }
       ]
     },
     {
       "service": "12345678-90ab-cdef-fedc-ba0987654321",
       "characteristic": "1234ff53-90ab-cdef-fedc-ba0987654321",
       "properties": [
         "Read",
         "Write"
       ],
       "descriptors": [
         {
           "uuid": "2901"
         }
       ]
     }
   ]
 }
```

mahavirj, 2019-11-26:
@monkeytronics You may find following pointers useful:

https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/provisioning/provisioning.html
https://github.com/espressif/esp-idf/tree/master/tools/esp_prov
https://github.com/espressif/esp-idf/tree/master/examples/provisioning/console_prov

monkeytronics, 2019-11-28:
@mahavirj Those are helpful. Unfortunately #3 doesn't run because of: 

`ImportError: No module named google.protobuf.internal`

but never mind that. I've started with the app. I send the first command to write and then read back over BLE from the ff53 characteristic. I am getting this which I don't expect: 

```
"Write Version V0.1"
"OK"
"Reading Version"
"{\n\t\"prov\":\t{\n\t\t\"ver\":\t\"v1.1\",\n\t\t\"cap\":\t[\"wifi_scan\"]\n\t}\n}"
```

Straight away I can't write & read the characteristic. Which is what I interpreted was done in esp_prov.py.  

This is an up hill struggle. The python code is really tough going to follow with clever FSMs to track to find the execution path. Is there any more linear documentation for the BLE use case or even pseudo code:

```
Write xxx to characteristic yyyy.
Read back from yyyy. It should be zzzz or else this thing is wrong. 
Create private / pubic key pair (with details)
etc... 
```
The console example is fine for debug, but it doesn't really help at all with recreating the application.

mahavirj, 2019-11-28:
@monkeytronics I am afraid but beyond phone application source SDKs (for Android/iOS) and host side python tool (which is mentioned above), there are no additional references available (at the time design we thought that this should cover most users and their customization requirements).

monkeytronics, 2019-11-28:
@mahavirj That makes sense. Is there anything available that we could use? Maybe a bullet point list showing the sequence of writes, reads, and security functions. It would be hugely helpful in getting us to market quicker.

monkeytronics, 2019-12-02:
Some specific questions:

1.a) When I write "V0.1" to char UUID (1234ff53-90ab-cdef-fedc-ba0987654321) it responds "OK". I was expecting "SUCCESS" based on the guide. Possibly a version issue in which case, some guidance on other differences of this version would be great.
2.b) When I read back from that same UUID, I get `{\n\t\"prov\":\t{\n\t\t\"ver\":\t\"v1.1\",\n\t\t\"cap\":\t[\"wifi_scan\"]\n\t}\n}`. I was expecting it to respond with "V0.1". Any idea what's going on. As per 1a) are the readme notes no longer up to date. 

3. The next example step involves sending to characteristic with UUID '0000ff51-0000-1000-8000-00805f9b34fb' payload = '10015a25a201220a20ae6d9d5d1029f8c366892252d2d5a0ffa7ce1ee5829312545dd5f2aba057294d'. I assume that in this step, we create a private and public key pair with curve25519, and send the public key over to the device. But the string is too long (it's 82 bytes instead of 64), so I'm not clear what it actually is. Would perhaps the group that developed the android app care to briefly list the sequence of steps and the packet contents for each one. It would be much more efficient than customers trying to reverse engineer the code. 

Sorry to be a pain, but it's in everyone's interest to make it easier to access this product. Thanks guys.

monkeytronics, 2019-12-03:
@mahavirj Is there any hope of getting some support on this. What's needed is a break down of the packets as sent and received. In question 2, I'm going to guess that the last 64 hex characters are the client public key. So, my question is really what are the other 18 hex chars. They look like cleartext given that the response contains something very similar, so it's got to be ble control stuff?? A little help or a clue would be great.

And the same for the remaining exchanges. What are the contents of each packet? We could really use some help finishing this, so we can focus on our core application. 

Thanks.

ameyinamdar, 2019-12-03:
Hi @monkeytronics. The protocol for WiFi provisioning is command-response based and uses Protobufs for data encoding (except for getting version). The specific command-response data structures can be found in To make the provisioning work with multiple transports (BLE, SoftAP), it relies on different endpoints and the command-response communication works on specific end-points.
1. "Proto-ver" end-point is used for getting protocol version using JSON encoding. This is always plaintext and user-readable for simplicity.
2. "Prov-session" end-point is used for secure session setup. In case of security method "security0" this becomes optional as no security is used at the protocol level.
3. "Prov-config" is an end-point where all the Wi-Fi configuration related command responses take place.
4. "Prov-scan" end-point provides optional Wi-Fi scan results service.

For BLE transport, these end-points are associated with various BLE characteristics (Please see https://github.com/espressif/esp-idf/blob/master/components/wifi_provisioning/src/manager.c#L1233). Write operation can be used to send command whereas read operation can fetch blocking response. For actual command-responses structure, you can check files present in https://github.com/espressif/esp-idf/tree/master/components/wifi_provisioning/proto. Wi-Fi configuration command-responses are listed in https://github.com/espressif/esp-idf/blob/master/components/wifi_provisioning/proto/wifi_config.proto. You can extend the protocol by adding your own end-point or by adding additional command-response in existing end-point. Then you have to regenerate "C" protobuf files using protoc-c compiler.

I hope this helps. Please let me know if I can help further.

monkeytronics, 2019-12-19:
Still working on this. Based on the example in : [ble_prov](https://github.com/espressif/esp-idf/tree/master/examples/provisioning/ble_prov) I was epecting to be able to write something like this:

```
 if (characteristic === 'prov-session') {
      uuid_frag = 'ff51';
 }

    this.ble.write(
      this.peripheral.id,
      '12345678-90ab-cdef-fedc-ba0987654321', // sevice id uuid
      '1234' + uuid_frag + '-90ab-cdef-fedc-ba0987654321', // characteristic uuid : proto-ver
      write_buffer).then( 
        response => this.setStatus('Write Response = ' + response),
        //() => this.setStatus('Write Successful.'),
        e => console.log('Oh no, ', e )
      );
```
Whe I do this, I can see the error tripping in the monitor:

```
E (856888) security1: Unable to unpack setup_req
E (856888) protocomm_ble: Invalid content received, killing connection
E (856898) BT_GATT: GATTS_SendRsp Unknown  conn_id: 3

E (856898) BT_APPL: Sending response failed
```
which is means the potobuf wasn't decoded successfully. So the provided example string isn't actually right. It would be great if I could get a sample that worked to start with. I've got prototbuf working, but it's not much use to me if I don't know what format to encode... 

Console log in my app looks like this:

```
[ng] [console.log]: {}
[ng] [console.log]: "Write Version 10015a25a201220a20ae6d9d5d1029f8c366892252d2d5a0ffa7ce1ee5829312545dd5f2aba057294d"
[ng] [console.log]: "Oh no, " 133
```

ameyinamdar, 2019-12-19:
@monkeytronics please note that the 'prov-session' endpoint is used only for session establishment as per the security protocol that is pre-defined between the device and the provisioner (python client in your case). I am not sure if you are using this end-point to send some different custom data across. If you wish to send custom data, you can create additional end-point in the device and add support for it in the provisioner. The example for sending the custom data can be found at [https://github.com/espressif/esp-idf/tree/master/examples/provisioning/custom_config](https://github.com/espressif/esp-idf/tree/master/examples/provisioning/custom_config)

The code to add custom endpoint can be found at [https://github.com/espressif/esp-idf/blob/master/examples/provisioning/custom_config/main/app_prov.c#L97](https://github.com/espressif/esp-idf/blob/master/examples/provisioning/custom_config/main/app_prov.c#L97) and actual custom data handler function can be found at [https://github.com/espressif/esp-idf/blob/master/examples/provisioning/custom_config/main/app_prov_handlers.c#L51](https://github.com/espressif/esp-idf/blob/master/examples/provisioning/custom_config/main/app_prov_handlers.c#L51)

The example also contains the proto files and host side python script as well to send the custom data. Essentially, you can use this example and change the custom_config proto [https://github.com/espressif/esp-idf/blob/master/examples/provisioning/custom_config/components/custom_provisioning/proto/custom_config.proto](https://github.com/espressif/esp-idf/blob/master/examples/provisioning/custom_config/components/custom_provisioning/proto/custom_config.proto) to contain data that you wish to send across or get from device.

monkeytronics, 2019-12-20:
@ameyinamdar thanks. I am sending the session setup request message. I've created a valid payload of type Sec1Payload. But it is being rejected by the device with a 133 error. 

```
        var payload = new messages.Sec1Payload();
        payload.setMsg(0);
        payload.setSc0(cmd0); 

        console.log(payload.getMsg());
        console.log(payload.hasSc0());
        console.log(payload.getSc0());
        console.log(payload.hasSc1());
        var bytesOfStuff =  payload.serializeBinary();
```

The binary packet is like this (as a Uint8Array). 

`"162,1,34,10,32,251,250,47,97,175,206,247,185,147,253,198,80,121,163,199,213,45,239,76,141,51,27,248,200,238,37,251,23,216,59,49,48"`

I notice my packet is significantly shorter than the values in the example below taken from the provisoining example. There are 41 characters there. Am I using the wrong .proto or something?

```
==== Starting Session ====
BLECLI >> Write following data to characteristic with UUID '0000ff51-0000-1000-8000-00805f9b34fb' :
    >> 10015a25a201220a20ae6d9d5d1029f8c366892252d2d5a0ffa7ce1ee5829312545dd5f2aba057294d
BLECLI >> Enter data read from characteristic (in hex) :
    << 10015a390801aa0134122048008bfc365fad4753dc75912e0c764d60749cb26dd609595b6fbc72e12614031a1089733af233c7448e7d7fb7963682c6d8
BLECLI >> Write following data to characteristic with UUID '0000ff51-0000-1000-8000-00805f9b34fb' :
    >> 10015a270802b2012212204051088dc294fe4621fac934a8ea22e948fcc3e8ac458aac088ce705c65dbfb9
BLECLI >> Enter data read from characteristic (in hex) :
    << 10015a270803ba01221a20c8d38059d5206a3d92642973ac6ba8ac2f6ecf2b7a3632964eb35a0f20133adb
==== Session Established ====
```

In fact I tried to push the example packets back through the protobuf deserialiser and they didn't register as valid.

ameyinamdar, 2019-12-20:
@monkeytronics I think you need to start with SessionData that wraps Sec1Payload. If you want to send a security1 session request from Python script, there is a python tool available [https://github.com/espressif/esp-idf/tree/master/tools/esp_prov](https://github.com/espressif/esp-idf/tree/master/tools/esp_prov) that has the relevant code that can be found at [https://github.com/espressif/esp-idf/blob/master/tools/esp_prov/security/security1.py#L132](https://github.com/espressif/esp-idf/blob/master/tools/esp_prov/security/security1.py#L132)

monkeytronics, 2019-12-27:
@ameyinamdar Yes. I did all of that already. I can now send data successfully. However, the shared key generation is not working. I have added some hexdumps into the code and enabled debug logging. From this, I can see the PoP SHA265 hashing function matches my implementation. However, the shared key does not. I have grabbed the hex dumps of private and public keys to debug, but as yet cannot successfully recreate the shared key! I've tried reversing the endian-ness of the private, public and shared key (8 combinations) without success (noticing that there is a lot of flip_endian() going on in the code. It really would be helpful if this was better documented. 

From the device code:

```
D (14049) security1: Device pubkey:
D (14059) security1: 1c 9d 55 ef c2 ad 4e 03 97 37 3c f5 5b b1 e1 f5
D (14059) security1: b1 dd de e2 e9 9a b3 e3 93 21 9b 86 2d b0 0c 77
D (14059) security1: Client pubkey:
D (14059) security1: dc 75 7e 8c 7b 7a cc 47 cb 88 f6 59 94 cf 73 2b
D (14069) security1: cc 3e d5 a1 94 69 87 e3 7d 43 77 99 41 69 b1 55
D (14469) security1: Adding proof of possession
D (14469) security1: PoP     =:
D (14469) security1: 61 62 63 64 31 32 33 34
D (14469) security1: SHA PoP =:
D (14479) security1: e9 ce e7 1a b9 32 fd e8 63 33 8d 08 be 4d e9 df
D (14479) security1: e3 9e a0 49 bd af b3 42 ce 65 9e c5 45 0b 69 ae
D (14489) security1: Shared key (no ^ SHA):
D (14489) security1: 21 03 0b 84 be 43 a2 5b 73 7c 04 40 84 be a5 44
D (14499) security1: f8 82 4a 81 91 65 95 2c 7b 4f 53 ce 26 01 35 1e
D (14509) security1: Shared key:
D (14509) security1: c8 cd ec 9e 07 71 5f b3 10 4f 89 48 3a f3 4c 9b
D (14519) security1: 1b 1c ea c8 2c ca 26 6e b5 2a cd 0b 63 0a 5c b0
```

From my implementation. My code pulls the device public key over the BLE connection, and logs this:
```
[ng] [console.log]: "device Public = " "1c9d55efc2ad4e0397373cf55bb1e1f5b1dddee2e99ab3e393219b862db00c77"
[ng] [console.log]: "Client Private  = " "684ea2dd64ae40bc46accb8455a9e900c89ec68092ef9b12e3b7fdf0b5fcae51"
[ng] [console.log]: "Shr (noSHA) " "52d3f1e808c935d0008e7a527fd409cc2c462e67144fd5cfb7b9a3f2dcd68263"
[ng] [console.log]: "SHA PoP   = " "e9cee71ab932fde863338d08be4de9dfe39ea049bdafb342ce659ec5450b69ae"
[ng] [console.log]: "Shr(final)= " "bb1d16f2b1fbc83863bdf75ac199e013cfd88e2ea9e0668d79dc3d3799ddebcd"
```

I've checked the pop & SHA of this. I've also double checked the XOR conversion. Those all look good. It's the calculation of Shared Key (before adding the PoP) that does not tie up ("Shr (noSHA) "). As I' say, I've tried reversing the endianness of everything. This is a standard function. What am I missing that explains why the result of the shared key generation do not match the expected value?

ameyinamdar, 2019-12-27:
@monkeytronics It uses Curve25519 to generate the shared secret and it seems to be not matching somehow with the device side implementation. On the device, it uses mbedTLS curve25519 API and the python scripts use "crypography.hazmat" X25519 python package. Please see: https://github.com/espressif/esp-idf/blob/30372f5a4ff2c0dfdaad95f544dc36bcdda30b75/tools/esp_prov/security/security1.py#L115

I suppose the best way to verify this may be to use device private key and client public key hardcoded in your Curve25519 call to ensure that it generates the same shared secret. May I ask which crypto implementation are you using?

ameyinamdar, 2019-12-27:
BTW, the security1 scheme documentation is available at https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/provisioning/provisioning.html

monkeytronics, 2020-01-06:
OK. So I've managed to establish the session. Wasted a week of time doing it. 

I don't suppose you could advise what's wrong with the following. I'm writing to the Start Scan endpoint. My code looks like this:

     ```
      let payload = new wifi_pb.WiFiScanPayload();
      
      if ( cmd ==='scan_start') {
        payload.setMsg(0);
        // payload.setStatus(0);
        let cmdScanStart = new wifi_pb.CmdScanStart();
        cmdScanStart.setBlocking(1);
        cmdScanStart.setPassive(0);       // hard coded in py code
        cmdScanStart.setGroupChannels(5);
        cmdScanStart.setPeriodMs(120);    // hard coded in py code
        payload.setCmdScanStart(cmdScanStart);
      }
      
      var bytesOfStuff =  payload.serializeBinary();
```

I am getting the following 8 bytes of payload: bytesOfStuff = {82,6,8,1,24,5,32,120} as a Uint8Array. This appears to be incorrect. When I use the provisioning example, I see only 6 bytes every time. Do you have an expected protocbuf payload I could compare against?

ameyinamdar, 2020-01-09:
@monkeytronics 

1) This appears to be issue with AES-CTR implementation on your client side. In the protocomm (which is the base for Wi-Fi provisioning), once the session is setup, the subsequent communication on all end-points happens with the same context of AES-CTR. Typically, you don't have to manage the nonce value manually and the crypto library should take care of correctly incrementing it internally.
2) Your second observation also points to the same probably that result in failed decoding of response. It is correct that if you read the value multiple times on an end-point (BLE characteristic), you won't see the value changing. Because as per the protocol, the command has to be executed by writing to the endpoint to receive the response. Only reading the response multiple times won't trigger the command automatically. The purpose behind this design is to make the protocol work across different transports like BLE and HTTP.

Essentially mapping the crypto correctly to match the server side implementation is hard. We understand that and that's where we have provided C, Java, Objective-C and Python code. But it looks like you're using a different client. I found that using MbedTLS's AES implementation is easily portable if your platform supports C/C++. Please see if that can work in your case.

monkeytronics, 2020-01-09:
@ameyinamdar 
1. I agree. My implementation has an issue. When you say communication happens with the same context, does that means the nonce is increased once per 16 byte block? The code I'm using appears to require manual control of the nonce, which is a pain. But my efforts to match up the nonce with the expected seem to fail for some reason. 
2. Interesting. I don't suppose you know what behaviour to expect for the nonce on repeated reads. 

Perhaps there's a way to get mbedtls to print out the nonce during the encryption / decryption process so I can compare them. I've tried to do this but couldn't figure it out. I set the flag in menuconfig -> mbedtls debug. And the tried to set the debug level to 4 in the code. But I didn't see anything in the console. With your obvious expertise with mbedtls, I don't suppose you know how to do this?

ameyinamdar, 2020-01-09:
@monkeytronics Please note that there are two nonce related parameters for AES-CTR - nonce and nonce_offset. Nonce is a 16 byte random (but shared) data that is incremented and nonce_offset keeps track of current offset which is (total_data_len_so_far % 16). I added a print in the AES-CTR encrypt function and performed Wi-Fi scanlist get operation couple of times and this is how I see the nonce and nonce_offset progressing. Please note that the nonce offset and nonce is printed before AES-CTR operation. Nonce value increases by 1 (nonce[15] being least-significant). Please see below log. Hope this helps.

```
esp_aes_crypt_ctr len 32 nc_offset 0 NC: BB 56 6F 41 C5 13 56 54 F7 AF 61 AD 54 B5 7A A8
esp_aes_crypt_ctr len 32 nc_offset 0 NC: BB 56 6F 41 C5 13 56 54 F7 AF 61 AD 54 B5 7A AA
esp_aes_crypt_ctr len 6 nc_offset 0 NC: BB 56 6F 41 C5 13 56 54 F7 AF 61 AD 54 B5 7A AC
esp_aes_crypt_ctr len 4 nc_offset 6 NC: BB 56 6F 41 C5 13 56 54 F7 AF 61 AD 54 B5 7A AD
esp_aes_crypt_ctr len 4 nc_offset 10 NC: BB 56 6F 41 C5 13 56 54 F7 AF 61 AD 54 B5 7A AD
esp_aes_crypt_ctr len 8 nc_offset 14 NC: BB 56 6F 41 C5 13 56 54 F7 AF 61 AD 54 B5 7A AD
esp_aes_crypt_ctr len 6 nc_offset 6 NC: BB 56 6F 41 C5 13 56 54 F7 AF 61 AD 54 B5 7A AE
esp_aes_crypt_ctr len 182 nc_offset 12 NC: BB 56 6F 41 C5 13 56 54 F7 AF 61 AD 54 B5 7A AE
esp_aes_crypt_ctr len 8 nc_offset 2 NC: BB 56 6F 41 C5 13 56 54 F7 AF 61 AD 54 B5 7A BA
esp_aes_crypt_ctr len 159 nc_offset 10 NC: BB 56 6F 41 C5 13 56 54 F7 AF 61 AD 54 B5 7A BA
esp_aes_crypt_ctr len 8 nc_offset 9 NC: BB 56 6F 41 C5 13 56 54 F7 AF 61 AD 54 B5 7A C4
esp_aes_crypt_ctr len 153 nc_offset 1 NC: BB 56 6F 41 C5 13 56 54 F7 AF 61 AD 54 B5 7A C5
esp_aes_crypt_ctr len 8 nc_offset 10 NC: BB 56 6F 41 C5 13 56 54 F7 AF 61 AD 54 B5 7A CE
esp_aes_crypt_ctr len 81 nc_offset 2 NC: BB 56 6F 41 C5 13 56 54 F7 AF 61 AD 54 B5 7A CF
esp_aes_crypt_ctr len 6 nc_offset 3 NC: BB 56 6F 41 C5 13 56 54 F7 AF 61 AD 54 B5 7A D4
esp_aes_crypt_ctr len 4 nc_offset 9 NC: BB 56 6F 41 C5 13 56 54 F7 AF 61 AD 54 B5 7A D4
esp_aes_crypt_ctr len 4 nc_offset 13 NC: BB 56 6F 41 C5 13 56 54 F7 AF 61 AD 54 B5 7A D4
esp_aes_crypt_ctr len 8 nc_offset 1 NC: BB 56 6F 41 C5 13 56 54 F7 AF 61 AD 54 B5 7A D5
esp_aes_crypt_ctr len 6 nc_offset 9 NC: BB 56 6F 41 C5 13 56 54 F7 AF 61 AD 54 B5 7A D5
esp_aes_crypt_ctr len 187 nc_offset 15 NC: BB 56 6F 41 C5 13 56 54 F7 AF 61 AD 54 B5 7A D5
esp_aes_crypt_ctr len 8 nc_offset 10 NC: BB 56 6F 41 C5 13 56 54 F7 AF 61 AD 54 B5 7A E1
esp_aes_crypt_ctr len 154 nc_offset 2 NC: BB 56 6F 41 C5 13 56 54 F7 AF 61 AD 54 B5 7A E2
```

monkeytronics, 2020-01-09:
That's amazing! Thanks for doing that. I don't suppose you could share the code you added so I can duplicate the output here. It would be so helpful debugging the last few parts of this process.

ameyinamdar, 2020-01-10:
@monkeytronics I can indeed provide the debug patch here. Please use:

```c
diff --git a/components/esp32/hwcrypto/aes.c b/components/esp32/hwcrypto/aes.c
index 3bf90a9bc4..1f36f6dde4 100644
--- a/components/esp32/hwcrypto/aes.c
+++ b/components/esp32/hwcrypto/aes.c
@@ -450,6 +450,14 @@ int esp_aes_crypt_ctr( esp_aes_context *ctx,
         return MBEDTLS_ERR_AES_INVALID_KEY_LENGTH;
     }

+    printf("esp_aes_crypt_ctr len %d nc_offset %d ", length, *nc_off);
+    printf("NC: ");
+    for (i = 0; i < 16; i++) {
+        printf("%2X ", nonce_counter[i]);
+    }
+
+    printf("\n");
+
     esp_aes_acquire_hardware();
     ctx->key_in_hardware = 0;
```

monkeytronics, 2020-01-15:
Awesome stuff. Got it all working. Thanks for all your help! I have learned quite a few things in the process! You guys are hardcore.

closed 2020-01-15 by monkeytronics

galah92, 2021-02-03:
@monkeytronics I stumbled this thread while searching for a provisioning flow implementation for the web (JS & Web Bluetooth). Any chance you can share your code with us?

CSHaze, 2021-02-09:
I am looking for a js implementation as well. Some code would be very helpful

sz-po, 2021-02-10:
@ameyinamdar  @monkeytronics I'm facing the same issue. Maybe espressif should consider JS implementation?

CSHaze, 2021-03-05:
@ameyinamdar @monkeytronics I think I am pretty close. I have the correct nonce and nonce offset. I just do not understand how to combine the two values? I am getting stuck at the step where the nonce_offset is 6, the first call where there is an offset. Does anyone know how to correctly apply the offset to the nonce value?

ameyinamdar, 2021-03-05:
Hi @CSHaze does the above comment - [https://github.com/espressif/esp-idf/issues/4399#issuecomment-572470170](https://github.com/espressif/esp-idf/issues/4399#issuecomment-572470170) help that describes the relationship between size of data and nonce_offset? It also shows how nonce increments. What are you looking for specifically?

monkeytronics, 2021-03-05:
@CSHaze It's been a while since I had my sleeves rolled up in that code, but as far as I recall, (and @ameyinamdar is the boss here) the offset is basically just a base 16 decimal place of the nonce. An aside on nomenclature. I've just upset myself my calling something a base 16 decimal. I'm not 100% is there's a name for it... Maybe a hex fraction? Essentially it keeps track of the portion of a 16 byte block that has been consumed. So, when the offset rolls over 0xF, the nonce increases by one! That make sense? The point is that the nonce does not increase if the block size is small enough to fall inside the same block as the previous chunk. That got me for sure!

CSHaze, 2021-03-09:
Hi @ameyinamdar and @monkeytronics, that definitely helps, I really appreciate the guidance. @monkeytronics this thread and some of your stack overflow questions have really helped me out. I think my nonce is correct, right offset, right nonce for each step according to the output I am getting from the esp32. My problem must lie somewhere else. Error handling on CryptoJS and google protobuff isn't the best so I was thinking it must have been the nonce but after you both explaining it I think I had that part correct. I'll keep trying and post back here if I discover anything meaningful that might help someone else.

marcelombc, 2022-04-20:
I'm trying to set the public key but I get an error (**Invalid public key length**).
Here is my code:

```js
const sessionCommand0 = new session.SessionCmd0();
sessionCommand0.setClientPubkey(PUBLIC_KEY);

const sec1Payload = new session.Sec1Payload();
sec1Payload.setSc0(sessionCommand0);

const sessionData = new session.SessionData();
sessionData.setSecVer(session.SecSchemeVersion.SECSCHEME1);
sessionData.setSec1(sec1Payload);

const bytes = await sessionData.serializeBinary();

await BleClient.write(
  connectedDevice.deviceId,
  '021a9004-0382-4aea-bff4-6b3f1c5adfb4',
  '021aff51-0382-4aea-bff4-6b3f1c5adfb4',
  numbersToDataView(bytesOfStuff),
  {
    timeout: 10000,
  }
);
```

The PUBLIC_KEY was generated using node crypto.

```js
const crypto = require('crypto');

const { publicKey, privateKey } = crypto.generateKeyPairSync('x25519', {
  publicKeyEncoding: {
    type: 'spki',
    format: 'pem',
  },
  privateKeyEncoding: {
    type: 'pkcs8',
    format: 'pem',
  },
});

console.log(publicKey);
console.log(privateKey);
```

marcelombc, 2022-04-22:
I'm now using https://www.npmjs.com/package/curve25519-js to generate the key pair and everything is working.