espressif/esp-idf #4400: [BLE Mesh] wifi & bt coexistence

isDerek, 2019-11-26:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it): 
    // v4.1-dev-1086-g93a8603c5
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [macOS]
- Power Supply:         [USB]

## Problem Description

when the device is running in wifi mode , i trigger the bt init, the device restart. 

### Expected Behavior

init success

### Actual Behavior

device restart

### Steps to repropduce

1. wifi is running in STA Mode

2. bt init 
```
  esp_err_t err;

  ESP_LOGI(TAG, "Initializing...");

  err = nvs_flash_init();
  if (err == ESP_ERR_NVS_NO_FREE_PAGES)
  {
    ESP_ERROR_CHECK(nvs_flash_erase());
    err = nvs_flash_init();
  }
  ESP_ERROR_CHECK(err);

  err = bluetooth_init();
  if (err)
  {
    ESP_LOGE(TAG, "esp32_bluetooth_init failed (err %d)", err);
    return;
  }

  ble_mesh_get_dev_uuid(dev_uuid);
```
3. the device restart
```
pda] 05:59:49.400 i c mod: demo_prov_enable: prov_enable set to 0
[pda] 05:59:56.514 i c client: ANS check event
I (28281) ble_mesh_provisioner: Initializing...
I (28291) BTDM_INIT: BT controller compile version [340d4d4]
I (28291) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40099ab1  PS      : 0x00060d33  A0      : 0x800980d1  A1      : 0x3fffd950  
0x40099ab1: uxListRemove at /Users/derek/Desktop/ayla-espressif/espressif/esp-idf/components/freertos/list.c:214

A2      : 0x3ffffff4  A3      : 0x3ffda070  A4      : 0x00060d20  A5      : 0x000000fe  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0xe52049d5  A9      : 0x00000000  
A10     : 0x00000001  A11     : 0x00000004  A12     : 0x00000008  A13     : 0x3fffbf48  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000009  LBEG    : 0x400905bc  LEND    : 0x400905c7  LCOUNT  : 0x00000000  
0x400905bc: memset at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:133

0x400905c7: memset at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:143

ELF file SHA256: 8b8685c62f27a40048892cc338a762afa4c98a49317794526baf09173b7807a7

Backtrace: 0x40099aae:0x3fffd950 0x400980ce:0x3fffd970 0x401b3184:0x3fffd990 0x401856e1:0x3fffd9e0 0x401cbacf:0x3fffda00 0x401cbb5a:0x3fffda20 0x401870b0:0x3fffda40 0x4018715c:0x3fffda60 0x4016c0cd:0x3fffda90 0x401b2ffd:0x3fffdab0 0x40096415:0x3fffdad0
0x40099aae: uxListRemove at /Users/derek/Desktop/ayla-espressif/espressif/esp-idf/components/freertos/list.c:214

0x400980ce: vTaskDelete at /Users/derek/Desktop/ayla-espressif/espressif/esp-idf/components/freertos/tasks.c:1225 (discriminator 4)

0x401b3184: osi_thread_create at /Users/derek/Desktop/ayla-espressif/espressif/esp-idf/components/bt/common/osi/thread.c:174

0x401856e1: hci_start_up at /Users/derek/Desktop/ayla-espressif/espressif/esp-idf/components/bt/host/bluedroid/hci/hci_layer.c:108

0x401cbacf: bte_main_enable at /Users/derek/Desktop/ayla-espressif/espressif/esp-idf/components/bt/host/bluedroid/main/bte_main.c:126

0x401cbb5a: bte_main_boot_entry at /Users/derek/Desktop/ayla-espressif/espressif/esp-idf/components/bt/host/bluedroid/main/bte_main.c:89

0x401870b0: btc_init_bluetooth at /Users/derek/Desktop/ayla-espressif/espressif/esp-idf/components/bt/host/bluedroid/btc/core/btc_main.c:64

0x4018715c: btc_main_call_handler at /Users/derek/Desktop/ayla-espressif/espressif/esp-idf/components/bt/host/bluedroid/btc/core/btc_main.c:103

0x4016c0cd: btc_thread_handler at /Users/derek/Desktop/ayla-espressif/espressif/esp-idf/components/bt/common/btc/core/btc_task.c:155

0x401b2ffd: osi_thread_run at /Users/derek/Desktop/ayla-espressif/espressif/esp-idf/components/bt/common/osi/thread.c:68

0x40096415: vPortTaskWrapper at /Users/derek/Desktop/ayla-espressif/espressif/esp-idf/components/freertos/port.c:143

Rebooting...
ets Jun  8 2016 00:22:57
```
menuconfig, i check the config ,the wifi coexistence is open
```
software controls wifi/bluetooth coexistence
```

Campou, 2019-11-29:
Hi @isDerek ,

Thanks for reporting this issue, will look into it.

Thanks.

Campou, 2019-12-02:
Hi @isDerek 

I tried with the BLE Mesh and Wi-Fi coexistence example, and made some minor changes to the example, it worked fine. The following is the screenshot of my example:
[image]

 If convenient, please attach the example on your side, and I will look into it.

Thanks.

isDerek, 2019-12-17:
sorry , i have modified the question , it occur becasue it have no memory. but the log has no tips so that i spend long time.

closed 2019-12-17 by isDerek

redradist, 2020-10-10:
> 
> 
> sorry , i have modified the question , it occur becasue it have no memory. but the log has no tips so that i spend long time.

@isDerek Could you explain how you fixed issue with memory ?