espressif/esp-idf #4401: ESP32 AP模式下，有sta退出wifi的时候，触发事件非常慢

fzboffice, 2019-11-26:
有sta退出wifi的时候
WIFI_EVENT_AP_STADISCONNECTED这个事件要过2分钟才触发，请问哪里可以设置快一些触发?

liuzfesp, 2019-11-26:
@fzboffice we have two different scenario to trigger this event:

- The event will be triggered immediately when it receives the deauth/disassociation frame from the  STA.
- The event will be triggered when the 5 minutes keep-alive timer expires, the timer expire if the AP doesn't receive any packets from STA for 5 minutes. Generally this scenario happens when the STA do hardware reset/power-off, or when the disassociation/de-auth frame is failed received by AP. Currently we don't have API to configure the keep-alive timer.

fzboffice, 2019-11-26:
> @fzboffice we have two different scenario to trigger this event:
>
> * The event will be triggered immediately when it receives the deauth/disassociation frame from the  STA.
> * The event will be [...]

@liuzfesp 能不能加入api让我可以修改这个计时器呢？谢谢

liuzfesp, 2019-11-26:
@fzboffice sure we can add API to do it, but we don't have time to do that in 2019 because we have other higher priority work to do. Maybe we can consider to add it in IDF v4.2.

closed 2020-08-20 by projectgus