espressif/esp-idf #4402: Facing compilation related issue while porting websocket example ino ESP32 IDF 2.1

Ritesh236, 2019-11-26:
Hello,

We are currently using ESp32 IDF 2.1 for our product development purpose and delivered few products as well to client.

Now We want to implement Web Socket example which uses esp-idf version V4.0-beta2 Pre-Release and we don't have any vision or option to move that version of IDF.

Because of older version it is hard to merge websocket example with my code. do you have any solution for that?

igrr, 2019-11-26:
Hi @Ritesh236,

ESP-IDF versions <= 3.0.x are already End of Life: https://github.com/espressif/esp-idf/blob/master/SUPPORT_POLICY.md#existing-releases. We do not provide support or bug fixes for these releases. We strongly recommend upgrading to a more recent release of ESP-IDF to take advantage of the latest features and bug fixes, including security bug fixes.

closed 2019-11-26 by igrr

Ritesh236, 2019-11-27:
@igrr ,

Thanks for providing details for same.

We have also tried to port into ESP32 IDF 3.2 IDF but still facing compilation related issue into that IDF. So, Ultimately, We need to move ESP32 IDF 4.0 which is still not stable release or we can say that it is RC release.

So, Please provide your suggestions like how it can be ported into ESP32 IDF 3.2 IDF?

Regards,
Ritesh Prajapati