espressif/esp-idf #4403: Cant run examples/system/network_tests with new ESP_NETIF

nx518, 2019-11-26:
## Environment
- Module: ESP32-WROOM-32D
- IDF versionv  v4.1-dev-1086-g93a8603c5 (master)
- Build System:    CMake
- Compiler version 1.22.0-80-g6c4433a

## Problem Description
The network_tests example crashes repeatedly with the new ESP-NETIF architecture.
esp_netif_new(&config) seems to return NULL.
also README seems outdated (still with tcpip_adapter)

### Expected Behavior
runs like with tcpip_adapter
### Actual Behavior
constant reboot after failed assert
### Steps to reproduce
cd ~/esp/esp-idf/examples/system/network_tests
idf.py build flash monitor

## Debug Logs

```
E (390) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x3ffb6b34 (config or if_key is NULL or duplicate key)
assertion "netif" failed: file "../main/net_suite.c", line 106, function: app_main
```

david-cermak, 2019-11-26:
Thanks for reporting this issue and good to hear that this example code is used (besides the  internal network testing).
There is a configuration issue that causes the `esp_netif_new()` to fail to create the object and  returns NULL. We will fix it soon.

As a workaround to get the example running before the fix is available I might suggest adding the ` .if_key = "TEST",` to `netif_common_config` and ` .driver = (void*)"test"` to the overall netif `config`.

nx518, 2019-11-27:
Thanks for the quick reply, looking forward to the fix.

The workaround still gives me:
```
E (382) esp_netif_lwip: Cannot start esp_netif: Missing mandatory configuration:
esp_netif->driver_transmit: 0x0, esp_netif->driver_handle:0x74736574, esp_netif->lwip_input_fn: 0x400d4f48, esp_netif->lwip_init_fn:0x400d4ee0

0x400d4f48: testnetif_input at network_tests/build/../main/lwip_test_netif.c:100
0x400d4ee0: testnetif_init at network_tests/build/../main/lwip_test_netif.c:56

ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x40085f78
0x40085f78: _esp_error_check_failed at esp-idf/components/esp32/panic.c:739

file: "esp-idf/components/esp_netif/lwip/esp_netif_lwip.c" line 561
func: esp_netif_start_api
expression: esp_netif_config_sanity_check(esp_netif)
```

david-cermak, 2019-11-27:
Oh, sorry. Forgot that the I/O transmit has to be defined, too.
Here's a patch with the quick-fix 
[0001-netsuite-quick-fix-to-make-the-example-working-with-.patch.txt](https://github.com/espressif/esp-idf/files/3896353/0001-netsuite-quick-fix-to-make-the-example-working-with-.patch.txt)

Official fix coming soon.

nx518, 2019-11-27:
Works 😃  Thanks, David.
Now I can continue with my QCA7000-driver.

closed 2019-12-18 by espressif-bot

giorgio80, 2020-06-19:
> Works 😃 Thanks, David.
> Now I can continue with my QCA7000-driver.

Hello @nx518 , I'm trying to make same about qca7k . as you already have done it can I ask you some help on the topic?

thanks

nx518, 2020-06-22:
I got it working, but the project was stalled before reaching publishable state.
It is based on: [https://github.com/torvalds/linux/tree/master/drivers/net/ethernet/qualcomm](https://github.com/torvalds/linux/tree/master/drivers/net/ethernet/qualcomm
)

giorgio80, 2020-06-23:
@nx518  Can I ask you how u made spi working With Ethernet or , can ask if some code Can be shared ?

bbulkow, 2021-01-06:
I am having this same result on commit 39cbf2f7ed4810913031055065da08227020e8b4 (Wed Dec 30 2020) with Olimex POE board and standard example (having configured as per Olimex POE v4.0 example). Have the above changes been added to ESP-IDF in the subsequent year, or are changes still required to run ethernet?