espressif/esp-idf #4405: SIM800L stack smashing

Arthedian, 2019-11-26:
- Development Kit: ESP32-DevKitC
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.0-beta2-70-g0a03a55c1
- Build System: CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): (crosstool-NG esp32-2019r1) 8.2.0
- Operating System: Fedora 31
- Power Supply:  USB.

## Problem Description

I am using official PPPOS client example. I changed stack smashing to overall, and it stops working

### Expected Behavior

Run without problem and connect to the modem.

### Actual Behavior

``` (0) cpu_start: Starting scheduler on APP CPU.
I (481) uart: queue free spaces: 30

Stack smashing protect failure!

abort() was called at PC 0x400d42e4 on core 0
0x400d42e4: __stack_chk_fail at /home/arthedian/esp/esp-idf/components/esp_common/src/stack_check.c:36

ELF file SHA256: 937a41b4233f12612c54e70a42f1136fc6203c01f19f8c69ed53f1e4d2b8933f

Backtrace: 0x40086a6c:0x3ffbacc0 0x40086ee0:0x3ffbacf0 0x400d42e4:0x3ffbad20 0x400d8e5f:0x3ffbad40 0x400d75c6:0x3ffbad80 0x400d7679:0x3ffbadc0 0x400d7796:0x3ffbadf0 0x40089b92:0x3ffbae20
0x40086a6c: invoke_abort at /home/arthedian/esp/esp-idf/components/esp32/panic.c:155

0x40086ee0: abort at /home/arthedian/esp/esp-idf/components/esp32/panic.c:172

0x400d42e4: __stack_chk_fail at /home/arthedian/esp/esp-idf/components/esp_common/src/stack_check.c:36

0x400d8e5f: sim800_handle_cops at /home/arthedian/prace/BeiT/test/pppos_client/build/../components/modem/src/sim800.c:217

0x400d75c6: esp_dte_handle_line at /home/arthedian/prace/BeiT/test/pppos_client/build/../components/modem/src/esp_modem.c:84

0x400d7679: esp_handle_uart_pattern at /home/arthedian/prace/BeiT/test/pppos_client/build/../components/modem/src/esp_modem.c:117

0x400d7796: uart_event_task_entry at /home/arthedian/prace/BeiT/test/pppos_client/build/../components/modem/src/esp_modem.c:179

0x40089b92: vPortTaskWrapper at /home/arthedian/esp/esp-idf/components/freertos/port.c:143

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

```

### Steps to reproduce

1. Run idf.py menuconfig
2. Go to Compiler options ->Stack smashing protection mode. Set it to Overall.
3. Flash and run application

negativekelvin, 2019-11-26:
https://github.com/espressif/esp-idf/blob/93a8603c545fb8e54741d6685146e2f3b874378d/examples/protocols/pppos_client/main/Kconfig.projbuild#L80

Arthedian, 2019-11-27:
> https://github.com/espressif/esp-idf/blob/93a8603c545fb8e54741d6685146e2f3b874378d/examples/protocols/pppos_client/main/Kconfig.projbuild#L80

I didnt change this value.

Arthedian, 2019-11-27:
@negativekelvin  Even if I set it to 6000, than nothing change, because it is not stack overflow error right?

negativekelvin, 2019-11-27:
Yes, possibly this https://github.com/espressif/esp-idf/blob/44b97b7fe7e0f2612de467541b92e93abece5c87/examples/protocols/pppos_client/components/modem/src/sim800.c#L203-L205

Arthedian, 2019-11-27:
How to fix it?

Arthedian, 2019-11-27:
I change the length of the array on line 199 to 100. New error come up:

I (6303) pppos_example: Power down
I (6303) uart: queue free spaces: 30
E (6803) esp-modem: esp_modem_dte_send_cmd(215): process command timeout
E (6803) dce_service: esp_modem_dce_sync(48): send command failed
E (6803) sim800: sim800_init(460): sync failed
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d7264  PS      : 0x00060730  A0      : 0x800d2705  A1      : 0x3ffb6430  
0x400d7264: app_main at /home/arthedian/prace/BeiT/test/pppos_client/build/../main/pppos_client_main.c:203

A2      : 0x00000000  A3      : 0x400d8884  A4      : 0x3ffbd34c  A5      : 0x00000001  
0x400d8884: sim800_deinit at /home/arthedian/prace/BeiT/test/pppos_client/build/../components/modem/src/sim800.c:428

A6      : 0x3ffb64f0  A7      : 0x3ffb7ed0  A8      : 0x800d7262  A9      : 0x3ffb63f0  
A10     : 0x00000000  A11     : 0x82fe9097  A12     : 0x82fe9097  A13     : 0x82fe9097  
A14     : 0x00000000  A15     : 0x0001c200  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000007c  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

ELF file SHA256: 657a1919c377bb80aa992058ad127b6821ea2ba21678613a0660e40465242f2a

Backtrace: 0x400d7261:0x3ffb6430 0x400d2702:0x3ffb6510 0x4008a40e:0x3ffb6530
0x400d7261: app_main at /home/arthedian/prace/BeiT/test/pppos_client/build/../main/pppos_client_main.c:197

0x400d2702: main_task at /home/arthedian/esp/esp-idf/components/esp32/cpu_start.c:553

0x4008a40e: vPortTaskWrapper at /home/arthedian/esp/esp-idf/components/freertos/port.c:143

Rebooting...
ets Jun  8 2016 00:22:57

Arthedian, 2020-01-04:
@suda-morris or anyone, any idea?

suda-morris, 2020-05-26:
Hi @Arthedian from your log, esp32 sends "AT" and doesn't get correct response with a specific time. Maybe this time the module is not ready. It usually takes some time (dozens of seconds for SIM800L to restart again from last power off).

zhaochengwei, 2020-07-18:
Maybe you need to check your sim800l power, In the example ,no output power.
like this 
https://randomnerdtutorials.com/esp32-sim800l-publish-data-to-cloud/

make  out put 
```
#define GPIO_OUTPUT_IO_PWKEY    4
#define GPIO_OUTPUT_IO_RST    5
#define GPIO_OUTPUT_IO_POWER_ON    23
    gpio_set_level(GPIO_OUTPUT_IO_PWKEY,0);
    gpio_set_level(GPIO_OUTPUT_IO_RST,1);
    gpio_set_level(GPIO_OUTPUT_IO_POWER_ON,1);
```
> @suda-morris or anyone, any idea?

zhaochengwei, 2020-07-18:
> I change the length of the array on line 199 to 100. New error come up:
>
> ```
> I (6303) pppos_example: Power down
> I (6303) uart: queue free spaces: 30
> E (6803) esp-modem: esp_modem_dte_send_cmd(215): [...]
Maybe you need to check your sim800l power, In the example ,no output power.
like this
https://randomnerdtutorials.com/esp32-sim800l-publish-data-to-cloud/

make out put
```
#define GPIO_OUTPUT_IO_PWKEY    4
#define GPIO_OUTPUT_IO_RST    5
#define GPIO_OUTPUT_IO_POWER_ON    23
    gpio_set_level(GPIO_OUTPUT_IO_PWKEY,0);
    gpio_set_level(GPIO_OUTPUT_IO_RST,1);
    gpio_set_level(GPIO_OUTPUT_IO_POWER_ON,1);
```
solve the problem 
[image]

lhespress, 2020-12-11:
@Arthedian 

Please modify `sim800_handle_cops` as follows:

[image]

and modify `CONFIG_EXAMPLE_MODEM_UART_EVENT_TASK_STACK_SIZE` to **3072** or larger.

tom-borcin, 2021-05-07:
Hi @Arthedian,

we think the issue is fixed now and therefore we are closing it to keep our backlog manageable. If you don't agree with this decision, please let me know.

Thank you,

Tomas

closed 2021-05-07 by tom-borcin