espressif/esp-idf #4407: mbedtls_net_set_nonblock / mbedtls_net_set_block doesn't work

szmodz, 2019-11-26:
Latest master (93a8603c545fb8e54741d6685146e2f3b874378d) using 2019r2 toolchain.

mbedtls_net_set_nonblock / mbedtls_net_set_block doesn't work.

Steps to reproduce: try using mbedtls_net_set_nonblock / mbedtls_net_set_block

Apparently it's because HAVE_FCNTL was not defined when building the C library.

https://github.com/espressif/newlib-esp32/blob/esp_based_on_3_0_0/newlib/libc/syscalls/sysfcntl.c

szmodz, 2019-11-28:
I've hacked around the C library problem, and it *seems* to work, but perhaps there are other issues with nonblocking sockets and that's why fcntl was disabled in the first place?

vikramdattu, 2019-12-06:
@szmodz can you please share a short code snippet with comments highlighting the issue?

szmodz, 2019-12-06:
This code doesn't work at all with the 2019r2 toolchain:

https://github.com/espressif/esp-aws-iot/blob/0ba1fe942785b2efd553d73124afaef2f0327f41/port/network_mbedtls_wrapper.c#L208

It always fails with
> ESP_LOGE(TAG, "failed! net_set_(non)block() returned -0x%x", -ret);

If you need test cases, here they are (Espressif's own code):
https://github.com/espressif/esp-aws-iot/tree/master/examples

The problem also applies to esp-tls:
https://github.com/espressif/esp-idf/blob/93a8603c545fb8e54741d6685146e2f3b874378d/components/esp-tls/esp_tls.c#L191

Basically anything that uses esp-tls in nonblocking mode is broken with the gcc 8.2 based toolchains. It just goes unnoticed because the call is not checked for errors.

szmodz, 2019-12-11:
Any ACK on this? I think this problem needs to be fixed, as it affects multiple esp-idf components, but I can keep posting bug reports... ;-)

szmodz, 2019-12-11:
This is why it works with gcc 5:
https://github.com/espressif/esp-idf/blob/93a8603c545fb8e54741d6685146e2f3b874378d/components/vfs/vfs.c#L728

(attribute weak)

https://github.com/espressif/esp-idf/commit/48d9d507aaf50bbac023ac9fa888cb28c6bdf2c8

mahavirj, 2019-12-12:
@szmodz

> Any ACK on this? I think this problem needs to be fixed, as it affects multiple esp-idf components, but I can keep posting bug reports... ;-)

Thanks for reporting this issue. Fix has been merged for this internally, once ESP-IDF codebase sync with github, this issue will be marked as closed.

closed 2019-12-16 by espressif-bot