espressif/esp-idf #4410: ble_l2cap_tx crash on boundaries maybe

Francois-Belanger, 2019-11-27:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version:   v4.1-dev-1086-g93a8603c5
- Build System:         [CMake]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     [macOS]
- Power Supply:         [external 5V]

## Problem Description

Bluetooth Nimble crash in edge case.

### Steps to repropduce

1. Nimble peripheral to iPhone
2. Using the following function.  Seems random, but looks like an edge case.

```cpp

bool ble_send_SPP(char* string, int len)
{
    struct os_mbuf* om;
    om = ble_hs_mbuf_from_flat(string, len);
    if (!om) {
        return false;
    }
    return !ble_gattc_notify_custom(g_spp_conn_handle,
        g_bleuart_attr_read_handle,
        om);
}

```
## Debug Logs

```
LOG:I (154426) BLE: BLE_GAP_EVENT_NOTIFY_TX

LOG:Searching 4116590.wav, size on Disk 13998, expected size 13998, res 0
LOG:Searching 76Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400f16a5  PS      : 0x00060630  A0      : 0x800eed9d  A1      : 0x3ffe19f0  
0x400f16a5: ble_l2cap_tx at /esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_l2cap.c:417

A2      : 0x3ffc3e28  A3      : 0x3ffc2104  A4      : 0x3ffc49c0  A5      : 0x3ffe1a20  
A6      : 0x00000000  A7      : 0x00000003  A8      : 0x00000000  A9      : 0x000f423f  
A10     : 0x00000000  A11     : 0x0000001d  A12     : 0x0000000c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffe  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000014  LBEG    : 0x40093474  LEND    : 0x400934a2  LCOUNT  : 0x00000000  
0x40093474: memcpy at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:168

0x400934a2: memcpy at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:201

ELF file SHA256: 0c8e520a6fc174a7650853ccdb75e81e2e24cfca4563be74100834d01d93a9d3

Backtrace: 0x400f16a2:0x3ffe19f0 0x400eed9a:0x3ffe1a20 0x400eeb99:0x3ffe1a50 0x400ee885:0x3ffe1a90 0x400dcb71:0x3ffe1ab0 0x400dc723:0x3ffe1ad0 0x4009bcd9:0x3ffe1b00
0x400f16a2: ble_l2cap_tx at /esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_l2cap.c:417

0x400eed9a: ble_att_tx at /esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_cmd.c:76

0x400eeb99: ble_att_clt_tx_notify at /esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_clt.c:962

0x400ee885: ble_gattc_notify_custom at /esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:4178

0x400dcb71: ble_send_SPP at /build/../main/gatt_svr.c:141

0x400dc723: prvBleTask at build/../main/ble_Task.c:327

0x4009bcd9: vPortTaskWrapper at /esp/esp-idf/components/freertos/port.c:143

```

**Solution found:**
`
int ble_l2cap_tx(struct ble_hs_conn* conn, struct ble_l2cap_chan* chan, struct os_mbuf* txom)
{
    int rc;

    txom = ble_l2cap_prepend_hdr(txom, chan->dcid, OS_MBUF_PKTLEN(txom));
    if (txom == NULL) {
        return BLE_HS_ENOMEM;
    }

    rc = ble_hs_hci_acl_tx(conn, &txom);
    switch (rc) {
    case 0:
        /* Success. */
        return 0;

    case BLE_HS_EAGAIN:
        /* Controller could not accommodate full packet.  Enqueue remainder. */
        if (txom) {. ///// ADDED TO STOP CRASH !!!!
            STAILQ_INSERT_TAIL(&conn->bhc_tx_q, OS_MBUF_PKTHDR(txom), omp_next);
        }
        return 0;

    default:
        /* Error. */
        return rc;
    }
}
`

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/3897259/sdkconfig.zip)

- [ ] coredump (This provides stacks of tasks.)

dhrishi, 2019-11-28:
Hi @Francois-Belanger Thanks for reporting the issue as well as mentioning the solution.

Could you please give more details about what (and how much) data are you passing to the function `ble_send_SPP` ? Also, if this issue occurs in the very first iteration? We would want to try this out.

Francois-Belanger, 2019-11-28:
Hi, I'm passing between 1 and 128 bytes per transfert. mtu is set at 256.  But when passing too much data it was crashing here.  So I made a new function that reduced it to max 128, solved most of the crash.  It you try to send large amount without this.  It crash a lot.

Here is the code:
`
                if (bleConnectedToHost) {
                    int byteToSends = strlen(ble_Mess);
                    int byteSent = 0;
                    // ets_printf("bleTask_send_SPP_Fast %s", ble_Mess);
                    while (byteToSends - byteSent > 0) {
                        if (xSemaphoreTake(fsmSemaphore_tx, portMAX_DELAY) == pdTRUE) {
                            int len = 0;
                            if (byteToSends - byteSent > 128) {
                                len = 128;
                            } else {
                                len = byteToSends - byteSent;
                            }
                            if (ble_send_SPP(&ble_Mess[byteSent], len)) {
                                byteSent += len;
                            } else {
                                xSemaphoreGive(fsmSemaphore_tx);
                            }
                        }
                    }
                }
`

Thanks

dhrishi, 2019-11-29:
@Francois-Belanger Thanks for the details. I have a question from the code above. You seem to give the semaphore only when ble_send_SPP returns false (it fails).So, how does this send data continuously (assuming it is binary semaphore) ? Won't it get blocked on the semaphore after the first successful send?

On a different note, as the fix you have provided is in the submodule (different repository) would you mind raising a PR to the original upstream repository https://github.com/apache/mynewt-nimble? I can then pull in the change in esp-nimble repository.

Francois-Belanger, 2019-11-29:
Hi, this is the missing code not pasted:

In Gap Event callback:
`    case BLE_GAP_EVENT_NOTIFY_TX:
        ESP_LOGI(TAG, "BLE_GAP_EVENT_NOTIFY_TX\n");
        xSemaphoreGive(fsmSemaphore_tx);
        break;
`

For the PR, I will do later today Thanks.

dhrishi, 2019-11-29:
@Francois-Belanger  Ohh okay, that makes sense.

> For the PR, I will do later today Thanks.

Great! Thanks.

dhrishi, 2019-12-17:
@Francois-Belanger We have recently made some changes related to this. Can you please pull in the latest Github code and try it out?

Also, I was not able to reproduce your problem using the inputs above. If the issue is not resolved, would you mind sharing the application file that you are using?

Francois-Belanger, 2019-12-17:
Hi Pull the IDF or the BLE Stack ?

dhrishi, 2019-12-18:
@Francois-Belanger  Pull the IDF and then update the NimBLE submodule `git submodule update --init`

Alvin1Zhang, 2020-01-07:
@Francois-Belanger Thanks for reporting. Would you please help share if any updates for this issue? Thanks.

Francois-Belanger, 2020-01-07:
Hi, will try soon.  Was on another project.

bbinet, 2020-02-10:
Hi @Francois-Belanger, I came across this issue while looking for a nimble-compatible BLE SPP implementation.
It seems you have actually implemented it already: would you agree to share the code you've written with the community. It would be great for example to have a dedicated BLE SPP implementation available in the [nimble examples](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble).

Alvin1Zhang, 2020-02-11:
@Francois-Belanger  Thanks for reporting. Do you have time to try? Would you please help share if any updates or feedback? Thanks.

Francois-Belanger, 2020-02-11:
I'm sorry, will do so tomorrow.

Francois-Belanger, 2020-02-11:
> Hi @Francois-Belanger, I came across this issue while looking for a nimble-compatible BLE SPP implementation.
> It seems you have actually implemented it already: would you agree to share the code [...]

Actually, I can't share everything inside my implementation, but I could show the important parts :-)

Alvin1Zhang, 2020-02-11:
@Francois-Belanger  Thanks.

bbinet, 2020-02-11:
> Actually, I can't share everything inside my implementation, but I could show the important parts :-)

That would be great, thank you.

Francois-Belanger, 2020-02-11:
@Alvin1Zhang , just downloaded the latest IDF (v4.1-dev-2224-gbcd2358eb) bug seem fixed :-)

Alvin1Zhang, 2020-02-12:
@Francois-Belanger Thanks for sharing the updates. Feel free to reopen if the issue still happens. Thanks.

closed 2020-02-12 by Alvin1Zhang