espressif/esp-idf #4411: esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x3ffc8b00

rebeliousconformist, 2019-11-27:
## Environment

- Development Kit:      [ESP32-Wrover-Kite]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
 (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     [Linux]
- Power Supply:         [external 5V Battery]

## Problem Description
I am starting the ESP32 in AP mode with Station scanning.
Then i am selecting a Station SSID and entering its password
After which im rebooting the entire wifi stack as shown in the code.
Earlier it functioned. After the esp_netif_init() insertion im getting a reboot(its configured to reboot upon fail)

After changing the tcpip_Adapter_init() to esp_netif_init() i am getting the following behaviour

### Expected Behavior

<pre>I (521303) connectingtostation: starting connect
I (521303) deinit: 1------------------------
I (523313) wifi: station: f0:d5:bf:fb:9f:f0 leave, AID = 1, bss_flags is 134259, bss:0x3ffbaf18
I (523313) wifi: new:&lt;1,0&gt;, old:&lt;1,1&gt;, ap:&lt;1,1&gt;, sta:&lt;0,0&gt;, prof:1
I (523413) wifi: flush txq
I (523413) wifi: stop sw txq
I (523413) wifi: lmac stop hw txq
I (523413) deinit: 2------------------------
I (525413) deinit: 3------------------------
I (525413) wifi: Deinit lldesc rx mblock:10
I (527423) deinit: 4------------------------
I (529433) wifi: wifi driver task: 3ffc0030, prio:23, stack:3584, core=0
I (529433) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (529433) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (529463) wifi: wifi firmware version: a44d1c6
I (529463) wifi: config NVS flash: enabled
I (529463) wifi: config nano formating: disabled
I (529463) wifi: Init dynamic tx buffer num: 32
I (529463) wifi: Init data frame dynamic rx buffer num: 32
I (529473) wifi: Init management frame dynamic rx buffer num: 32
I (529483) wifi: Init management short buffer num: 32
I (529483) wifi: Init static rx buffer size: 1600
I (529483) wifi: Init static rx buffer num: 10
I (529493) wifi: Init dynamic rx buffer num: 32
I (529493) deinit: ssid is pemwireless
I (529503) deinit: password is pemwireless52072
I (529633) wifi: mode : sta (bc:dd:c2:c9:0a:60) + softAP (bc:dd:c2:c9:0a:61)
I (529643) wifi: Total power save buffer number: 16
I (529643) deinit: wifi connected starting MQTT
I (529643) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
11. I (529653) MQTT: Other event id:7
W (529653) httpd_txrx: httpd_sock_err: error in send : 113
W (529663) httpd_txrx: httpd_sock_err: error in recv : 113
I (529763) wifi: ap channel adjust o:1,1 n:6,1
I (529763) wifi: new:&lt;6,1&gt;, old:&lt;1,0&gt;, ap:&lt;6,1&gt;, sta:&lt;6,1&gt;, prof:1
I (530343) wifi: state: init -&gt; auth (b0)
I (530363) wifi: state: auth -&gt; assoc (0)
I (530363) wifi: state: assoc -&gt; run (10)
I (530383) wifi: connected with pemwireless, aid = 2, channel 6, 40U, bssid = f8:1a:67:a5:fd:64, security type6
I (530393) wifi: pm start, type: 1

I (530453) wifi: AP&apos;s beacon interval = 102400 us, DTIM period = 1
I (531243) esp_netif_handlers: sta ip: 192.168.0.100, mask: 255.255.255.0, gw: 192.168.0.1
I (535833) wifi: new:&lt;6,1&gt;, old:&lt;6,1&gt;, ap:&lt;6,1&gt;, sta:&lt;6,1&gt;, prof:6
I (535833) wifi: station: f0:d5:bf:fb:9f:f0 join, AID=1, bgn, 40U
I (535873) ESP Event Handler: station f0:d5:bf:fb:9f:f0 join, AID=1
E (559663) MQTT_CLIENT: Error transport connect
10. I (559663) MQTT: MQTT_EVENT_ERROR
5. I (559663) MQTT: MQTT_EVENT_DISCONNECTED
11. I (574663) MQTT: Other event id:7
I (574673) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
I (574683) MQTT: MQTT_EVENT_CONNECTED
1. I (574683) MQTT: sent publish successful, msg_id=36647
2. I (574693) MQTT: sent subscribe successful, msg_id=28149
3. I (574693) MQTT: sent subscribe successful, msg_id=36941
4. I (574703) MQTT: sent unsubscribe successful, msg_id=62782
8. I (574703) MQTT: MQTT_EVENT_PUBLISHED, msg_id=36647
6. I (574713) MQTT: MQTT_EVENT_SUBSCRIBED, msg_id=28149
I (574713) MQTT: sent publish successful, msg_id=0
6. I (574723) MQTT: MQTT_EVENT_SUBSCRIBED, msg_id=36941
I (574723) MQTT: sent publish successful, msg_id=0
7. I (574733) MQTT: MQTT_EVENT_UNSUBSCRIBED, msg_id=62782
9. I (574743) MQTT: MQTT_EVENT_DATA
TOPIC=/topic/qos0
DATA=data
9. I (574743) MQTT: MQTT_EVENT_DATA
TOPIC=/topic/qos0
DATA=data
I (689173) wifi: new:&lt;6,1&gt;, old:&lt;6,1&gt;, ap:&lt;6,1&gt;, sta:&lt;6,0&gt;, prof:6
I (690193) wifi: new:&lt;6,1&gt;, old:&lt;6,1&gt;, ap:&lt;6,1&gt;, sta:&lt;6,1&gt;, prof:6
I (798233) wifi: new:&lt;6,1&gt;, old:&lt;6,1&gt;, ap:&lt;6,1&gt;, sta:&lt;6,0&gt;, prof:6
I (800173) wifi: new:&lt;6,1&gt;, old:&lt;6,1&gt;, ap:&lt;6,1&gt;, sta:&lt;6,1&gt;, prof:6
</pre>

### Actual Behavior
<pre>I (31963) connectingtostation: starting connect
I (31973) deinit: 1------------------------
I (33973) wifi: station: f0:d5:bf:fb:9f:f0 leave, AID = 1, bss_flags is 134259, bss:0x3ffbabf0
I (33973) wifi: new:&lt;1,0&gt;, old:&lt;1,1&gt;, ap:&lt;1,1&gt;, sta:&lt;0,0&gt;, prof:1
I (34003) wifi: flush txq
I (34003) wifi: stop sw txq
I (34003) wifi: lmac stop hw txq
I (34003) deinit: 2------------------------
I (36003) deinit: 3------------------------
I (36003) wifi: Deinit lldesc rx mblock:10
I (38013) deinit: 4------------------------
E (40013) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x3ffc8b00 (config or if_key is)
assertion &quot;netif&quot; failed: file &quot;/home/pemubuntu/espressif/esp-idf/components/esp_wifi/src/wifi_default.c&quot;, linp
abort() was called at PC 0x400d6bf3 on core 0

ELF file SHA256: 740df468a3cab8b73a3c87029b0942602d3a3c1d229ce3e15f2b562ccc579035

Backtrace: 0x4008f8f5:0x3ffc8a90 0x4008fc75:0x3ffc8ab0 0x400d6bf3:0x3ffc8ad0 0x400f791d:0x3ffc8b00 0x400d429d:0

Rebooting...
ets Jun  8 2016 00:22:57
</pre>

### Steps to repropduce

1. step1
Start the ESP32 in AP mode
enter password
ESP32 reboots the wifi stacks and now connects to the selection accesspoint as well as starts it personal accesspoint
2. code fails.

###Code which works properly

```cpp
<script src="https://gist.github.com/rebeliousconformist/b9f05de13691d8ffa9ef6cdd0023c45a.js"></script>
```cpp

### Code to reproduce this issue

```cpp
<script src="https://gist.github.com/rebeliousconformist/7352d013cd7036fc7e44b06a2dc20d43.js"></script>
```cpp
``` 

// the code should be wrapped in the ```cpp tag so that it will be displayed better.

david-cermak, 2019-11-28:
Hi @rebeliousconformist 

Thanks for reporting this issue and sharing both versions of the code.
The problem in this case is creating the default AP multiple times. Both these methods to create default wifi interfaces: `esp_netif_create_default_wifi_ap()` and `esp_netif_create_default_wifi_sta()` return a pointer to the corresponding esp_netif object. This is not very clear from the examples, as the return value is effectively ignored since no deinit takes places.

To address the issue please store the pointer to the created network interface so that it could be destroyed later in the deinit phase:
* `esp_netif_t * my_ap = esp_netif_create_default_wifi_ap();` in https://gist.github.com/rebeliousconformist/7352d013cd7036fc7e44b06a2dc20d43#file-restartap2-c-L93
* `esp_netif_destroy(my_ap);` somewhere in [the deinit part](https://gist.github.com/rebeliousconformist/b9f05de13691d8ffa9ef6cdd0023c45a#file-restartap-c-L34)

The other alternative would be to call both create_ap and create_sta only once at startup (this is inline with tcpip_adapter behaviour, but not preferred though)

I'm sorry this is not (yet) properly documented.

rebeliousconformist, 2019-11-29:
works ! thanks !

closed 2019-11-29 by rebeliousconformist