espressif/esp-idf #4412: ESP_LOG/vprintf fail inside critical section. It`s right?

neorevx, 2019-11-27:
##  Environment

- Development Kit: Custom
- Module or chip used:  ESP32-WROOM-32D
- IDF version: v4.1-dev-1086-g93a8603c5
- Build System: CMake
- Compiler version: (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Windows
- Power Supply: Custom

## Problem Description

After pre-release v4.0/beta2, the ESP_LOG can't be called from critial section.
ESP_LOG call vprintf that use lock from newlib. Lock check xPortCanYield() (false in critical section) and then call abort!

https://github.com/espressif/esp-idf/blob/93a8603c545fb8e54741d6685146e2f3b874378d/components/newlib/locks.c#L140-L144

### Expected Behavior

I don't know if it's expected behavior for v4.1.

## Debug Logs

```
abort() was called at PC 0x400831e6 on core 0
0x400831e6: lock_acquire_generic at D:/OneDrive/ESP32-4/components/newlib/locks.c:143

0x4008bba5: invoke_abort at D:/OneDrive/ESP32-4/components/esp32/panic.c:157

0x4008bf95: abort at D:/OneDrive/ESP32-4/components/esp32/panic.c:174

0x400831e6: lock_acquire_generic at D:/OneDrive/ESP32-4/components/newlib/locks.c:143

0x40083309: _lock_acquire_recursive at D:/OneDrive/ESP32-4/components/newlib/locks.c:171

0x401369a2: _vfprintf_r at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:853 (discriminator 2)

0x401399d1: vprintf at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vprintf.c:34

0x400828d9: esp_log_write at D:/OneDrive/ESP32-4/components/log/log.c:219

0x400e72cd: gpio_config at D:/OneDrive/ESP32-4/components/driver/gpio.c:354 (discriminator 9)
```

igrr, 2019-11-27:
In the past it would work if the lock wasn't already taken by some other task. But if lock was taken, and log was called from a critical section, the task would attempt to block whole interrupts are disabled. This would lead to a lockup in FreeRTOS scheduler, and an interrupt WDT timeout. Now such cases are checked more strictly.

igrr, 2019-12-18:
@neorevx It's not clear if you meet this issue while using the CAN driver included in ESP-IDF, or a custom CAN driver? If this is with the IDF CAN driver, could you please post the complete backtrace? From the backtrace you have posted, `gpio_config` doesn't enter any critical sections, and the IDF CAN driver also doesn't call `gpio_config`. So I wonder where the critical section comes from.

neorevx, 2019-12-19:
@igrr I think it was not very clear. Excuse me. My point is that ESP_LOG has stopped working within critical regions. I asked only for codes where this could happen. (About stack, it's custom).
As an example, I talked about CAN. Following are the details:

This is the critical region of the can driver installer:
https://github.com/espressif/esp-idf/blob/30372f5a4ff2c0dfdaad95f544dc36bcdda30b75/components/driver/can.c#L699-L732

The function 'can_configure_gpio' is called in line 728.
https://github.com/espressif/esp-idf/blob/30372f5a4ff2c0dfdaad95f544dc36bcdda30b75/components/driver/can.c#L615-L641

The function 'gpio_set_pull_mode' is called from 'can_configure_gpio':
https://github.com/espressif/esp-idf/blob/30372f5a4ff2c0dfdaad95f544dc36bcdda30b75/components/driver/gpio.c#L225-L228

This function has some checks that can use ESP_LOG.
https://github.com/espressif/esp-idf/blob/30372f5a4ff2c0dfdaad95f544dc36bcdda30b75/components/driver/gpio.c#L33-L37

I just checked this case, but there may be other ESP_LOG within the other calls.

igrr, 2019-12-19:
Ah, thanks, this looks like a bug in the CAN driver then.

Previously ESP_LOG from a critical section could either succeed or fail, depending on whether some other task was holding the stdout lock or not. In recent versions it will always fail. We don't consider it valid to call ESP_LOG from a critical section, therefore this change in behavior is not considered a regression.

Of course, specific instances where ESP_LOG is called from a critical section will be fixed, such as in this case with the CAN driver.

neorevx, 2019-12-19:
I analyzed can.c and found some gpio checks in the install function. So this error will never happen the way I mentioned it. However, for future maintenance, I still suggest removing these calls from the critical region, as it links changes in gpio.c to a possible bug in can.c.
Another solution may be to remove esp_log from gpio.c.

neorevx, 2019-12-27:
There's a lot of ESP_LOG inside critical region of intr_alloc.c. Example:

https://github.com/espressif/esp-idf/blob/30372f5a4ff2c0dfdaad95f544dc36bcdda30b75/components/esp32/intr_alloc.c#L736-L781

igrr, 2019-12-27:
Thanks, the fix for these 2 ESP_LOGs in intr_alloc.c has been merged and will be available on Github soon. This issue will be automatically closed when that happens.

closed 2020-01-06 by espressif-bot