espressif/esp-idf #4413: Task watchdog got triggered in BLE functions

ildus, 2019-11-27:
Recently I have started experimenting with NimBLE and in after some time I started to get these errors:
```
E (233591) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (233591) task_wdt:  - IDLE0 (CPU 0)
E (233591) task_wdt: Tasks currently running:
E (233591) task_wdt: CPU 0: ble
E (233591) task_wdt: CPU 1: IDLE1
```
I have made some research and found out that this error could be fixed by putting `vTaskDelay` in these functions:

* `nimble_port_run`
* `ble_ll_task`

I'm not sure that's the right way to fix the problem so please let me know if there is more information needed. The problem occures on `master` branch, and I have not check older branches.

prasad-alatkar, 2019-11-28:
Hi @ildus Could you please let me know if you have made any changes to default `bleprph` example ? Just FYI, I have fixed an issue in NimBLE porting layer which is in internal pipeline that maybe relevant here, I will notify you as soon as it is available on github.

ildus, 2019-11-28:
I get this error on default `bleprph` too. Strange thing, when I just started everything was ok for a while.

prasad-alatkar, 2019-12-18:
@ildus fix for NimBLE porting layer is available on github master, commit id : 656005ef. Can you please check the issue with the latest master ?

Alvin1Zhang, 2020-01-07:
@ildus Thanks for reporting. Would you please help check whether your issue has been fixed with the latest master? Thanks.

ildus, 2020-01-07:
@Alvin1Zhang @prasad-alatkar sorry, will test today evening (about 20:00 UTC).

ildus, 2020-01-07:
Ok, just checked. The problem seems to be gone. Thank you.

albertmbt, 2020-03-11:
Hi there,

I still have this same issue with stable version 4.0. and working over the bleprph example with no modifications. (I actually tested also with the master branch getting same results).

At the first, I started working with no issues, but after a while it started with this error just after a subscribe event.

```
subscribe event; conn_handle=0 attr_handle=8 reason=1 prevn=0 curn=0 previ=0 curi=1

E (386094) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (386094) task_wdt:  - IDLE0 (CPU 0)
E (386094) task_wdt: Tasks currently running:
E (386094) task_wdt: CPU 0: ble
E (386094) task_wdt: CPU 1: IDLE1
```

I tried flashing it to a new module, and it is working properly for now, bit I think it's about time. In fact, there is no way that my first module can get it to work having the same code.

Maybe I lost something?

prasad-alatkar, 2020-03-12:
Hi @albertmbt Can you try  `erase_flash` on the module and flash again ? If you still face the issue, can you please share debug logs(`make menuconfig -->component config-->Log output-->Default log verbosity-->Debug`) ?

albertmbt, 2020-03-12:
Yep erasing memory solves the problem. I wasn't paying attention when I first got the error, probably it's a problem after flash then and not randomly on work as I thought.

prasad-alatkar, 2020-03-12:
Hi @albertmbt I understand, please provide debug logs if you get this issue again.

Alvin1Zhang, 2020-03-16:
@ildus  Thanks for reopening. Did you run into the issue again? Could you please help share any updates or details? Thanks.

Unsigus, 2020-03-17:
Hi, I also got into this problem. It happens when there are too many cccd's stored in NVS.
Here is the verbose debug log

`Encrypt change: status=0 handle=0 state=1
I (45857) bleprph: BLE_GAP_EVENT_ENC_CHANGE
I (45857) BLE_MISC: handle=0 our_ota_addr_type=0 our_ota_addr=3c39e7221b32
I (45857) BLE_MISC: our_id_addr_type=0 our_id_addr=3c39e7221b32
I (45867) BLE_MISC: peer_ota_addr_type=1 peer_ota_addr=61617c44f949
I (45877) BLE_MISC: peer_id_addr_type=1 peer_id_addr=61617c44f949
I (45877) BLE_MISC: conn_itvl=24 conn_latency=0 supervision_timeout=72 encrypted=1 authenticated=0 bonded=1
error persisting cccd; too many entries (8)
looking up our sec;
error persisting cccd; too many entries (8)
looking up our sec;
error persisting cccd; too many entries (8)
looking up our sec;
.... (infinite)`

Unsigus, 2020-03-18:
@albertmbt or @Alvin1Zhang 
The issue occours in the ble_store_write() in ble_store.c. The loop expects after calling ble_store_overflow_event(), that there is room available. Somehow this isn't the case.

I'm too unfamiliar with the IDF to further investigate the issue, I hope you can solve it. Thanks!

dhrishi, 2020-03-18:
@albertmbt  @Unsigus, @prasad-alatkar is looking into this and will update after there is progress. Thanks!

prasad-alatkar, 2020-03-18:
Hi @Unsigus this is actually not the same issue.
For your case, I am afraid but you have to increase CCCDs from menuconfig (`make menuconfig --> Component config--> Bluetooth--> NimBLE Options--> Maximum number of CCC descriptors to save across reboots`).  

> The issue occours in the ble_store_write() in ble_store.c. The loop expects after calling ble_store_overflow_event(), that there is room available. Somehow this isn't the case.

Yes, NimBLE stack makes space by deleting oldest peer `ble_gap_unpair_oldest_peer`. However for your case if application needs more than 8 CCDs per device then the application should configure it beforehand. 

Please let me know if you find any other issue. Please help to close the issue if you find the response satisfactory.

Unsigus, 2020-03-18:
Hi @prasad-alatkar, Thank you for your time, and will tweak my settings.
But don't you agree that the app needs to respond differently (like an assert) than getting into an infinite loop (thus a watchdog timeout)? That can save a lot of time when others run into this problem.

prasad-alatkar, 2020-03-20:
@Unsigus Could you please update if your issue was resolved with tweaking settings from menuconfig  ? 

> But don't you agree that the app needs to respond differently (like an assert) than getting into an infinite loop (thus a watchdog timeout)? That can save a lot of time when others run into this problem.

I do agree that there should be better way to handle this. However it is application developer's responsibility as well to configure minimum required settings. Having said that I am also thinking about proper handling of this particulate situation, will update you accordingly. 

Apart from that just to maintain history of the issue clean, the issue @Unsigus has faced is different from the original reported issue which is infrequent and `make erase_flash` happened to resolve it. @albertmbt @ildus DId you guys get a chance to reproduce the issue with DEBUG logs on ?

Unsigus, 2020-03-20:
@prasad-alatkar, from my point of view the issue is the same. I also got this message every x seconds, while the app stops responding to everything else:
```
E (386094) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (386094) task_wdt:  - IDLE0 (CPU 0)
E (386094) task_wdt: Tasks currently running:
E (386094) task_wdt: CPU 0: ble
E (386094) task_wdt: CPU 1: IDLE1
```

Using `make erase_flash` did solve that for some time. When the cccd NVS reaches is maximum (Don't now how that happens, my application doesn't need more than 8 cccd's per device. But I think it did not clear the records correctly when I was developing with trial and errors, clearing nvs did solve that)

With that information I've reproduced the issue that I think @albertmbt has. With the DEBUG log on this was printed directly when turning on notifications:
> 
> Here is the verbose debug log from the bleprph example: 
> ```
> rxed att command: find info req; conn=0 start_handle=0x0009 end_handle=0x0009
>...
> error persisting cccd; too many entries (8)
> looking up our sec;
> ...
> ```

Changing `Maximum number of CCC descriptors to save across reboots` will solve the issue, but that's just a workaround in my opinion (the wrongly stored cccd's are still stored and will never be removed)

If this still not the same issue in your opinion, please inform me and I will leave you guys alone.

albertmbt, 2020-03-20:
Hi guys, i've been busy on other project those days and i haven't been able to keep looking at it. i expect to have time during the next week. In the poor test's that i've made those days i haven't been able to reproduce the issue again. i'll keep updating.

ildus, 2020-03-20:
@prasad-alatkar I just used old bluetooth stack for my device, and for now I'm not able to debug. But I remember in my case that the issue also was fixed after erasing the flash. But I agree with @Unsigus that is a bug, since it happens without any extra changes on example application, for example I didn't even know that there are some cccd's that are stored to flash. At least it should be logged, and there should be proper workaround.

prasad-alatkar, 2020-03-20:
Hi @Unsigus Could you please share the entire debug log ?

Unsigus, 2020-03-20:
Here you go
```
I (29) boot: ESP-IDF v4.2-dev-701-g0ae960f2f-dirty 2nd stage bootloader
I (29) boot: compile time 14:03:03
I (30) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0e898 ( 59544) map
I (136) esp_image: segment 1: paddr=0x0001e8c0 vaddr=0x3ffbdb60 size=0x01758 (  5976) load
I (139) esp_image: segment 2: paddr=0x00020020 vaddr=0x400d0020 size=0x55820 (350240) map
0x400d0020: _stext at ??:?

I (276) esp_image: segment 3: paddr=0x00075848 vaddr=0x3ffbf2b8 size=0x01d24 (  7460) load
I (280) esp_image: segment 4: paddr=0x00077574 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at ~/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1817

I (284) esp_image: segment 5: paddr=0x00077980 vaddr=0x40080404 size=0x137e4 ( 79844) load
I (337) boot: Loaded app from partition at offset 0x10000
I (337) boot: Disabling RNG early entropy source...
I (338) cpu_start: Pro cpu up.
I (341) cpu_start: Application information:
I (346) cpu_start: Project name:     bleprph
I (351) cpu_start: App version:      v4.2-dev-701-g0ae960f2f-dirty
I (358) cpu_start: Compile time:     Mar 20 2020 14:02:56
I (364) cpu_start: ELF file SHA256:  807a1f7926ac86cb...
I (370) cpu_start: ESP-IDF:          v4.2-dev-701-g0ae960f2f-dirty
I (377) cpu_start: Starting app cpu, entry point is 0x400812fc
0x400812fc: call_start_cpu1 at ~/esp/esp-idf/components/esp32/cpu_start.c:273

I (0) cpu_start: App cpu up.
D (387) memory_layout: Checking 11 reserved memory ranges:
D (393) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (399) memory_layout: Reserved memory range 0x3ffae6e0 - 0x3ffaff10
D (405) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb6388
D (412) memory_layout: Reserved memory range 0x3ffb8000 - 0x3ffb9a20
D (418) memory_layout: Reserved memory range 0x3ffbdb28 - 0x3ffbdb5c
D (425) memory_layout: Reserved memory range 0x3ffbdb60 - 0x3ffc3bd0
D (431) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (438) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (444) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (450) memory_layout: Reserved memory range 0x40078000 - 0x40080000
0x40080000: _WindowOverflow4 at ~esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1817

D (457) memory_layout: Reserved memory range 0x40080000 - 0x40093be8
0x40080000: _WindowOverflow4 at ~/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1817

D (463) memory_layout: Building list of available memory regions:
D (469) memory_layout: Available memory region 0x3ffaff10 - 0x3ffb0000
D (476) memory_layout: Available memory region 0x3ffb6388 - 0x3ffb8000
D (483) memory_layout: Available memory region 0x3ffb9a20 - 0x3ffbdb28
D (489) memory_layout: Available memory region 0x3ffbdb5c - 0x3ffbdb60
D (496) memory_layout: Available memory region 0x3ffc3bd0 - 0x3ffc4000
D (502) memory_layout: Available memory region 0x3ffc4000 - 0x3ffc6000
D (509) memory_layout: Available memory region 0x3ffc6000 - 0x3ffc8000
D (516) memory_layout: Available memory region 0x3ffc8000 - 0x3ffca000
D (522) memory_layout: Available memory region 0x3ffca000 - 0x3ffcc000
D (529) memory_layout: Available memory region 0x3ffcc000 - 0x3ffce000
D (535) memory_layout: Available memory region 0x3ffce000 - 0x3ffd0000
D (542) memory_layout: Available memory region 0x3ffd0000 - 0x3ffd2000
D (549) memory_layout: Available memory region 0x3ffd2000 - 0x3ffd4000
[... 414 lines cut ...]
looking up our sec;
error persisting cccd; too many entries (4)
looking up our sec;
error persisting cccd; too many entries (4)
looking up our sec;
error persisting cccd; too many entries (4)
looking up our sec;
error persisting cccd; too many entries (4)
looking up our sec;
error persisting cccd; too many entries (4)
looking up our sec;
error persisting cccd; too many entries (4)
looking up our sec;
error persisting cccd; too many entries (4)
looking up our sec;
error persisting cccd; too many entries (4)
looking up our sec;
error persisting cccd; too many entries (4)
looking up our sec;
error persisting cccd; too many entries (4)
looking up our sec;
[-> will continue into infinity)

Done
```

prasad-alatkar, 2020-03-31:
Hi @Unsigus Thank you for reporting this issue, I am attaching a temporary patch here to be applied at `$IDF_PATH/components/bt/host/nimble/nimble`. Could you please test and let me know if it resolves your issue ? 
patch:
[NimBLE_CCCD_overflow_loop_fix.zip](https://github.com/espressif/esp-idf/files/4411057/NimBLE_CCCD_overflow_loop_fix.zip)

Unsigus, 2020-04-01:
Thanks @prasad-alatkar. Sadly I was unable to reproduce the issue before and after the patch. I updated the ESP-IDF to v:latest, no issue there. @albertmbt, do you have time to look into this?

I keep trying and get back to this when I know more.

prasad-alatkar, 2020-04-01:
@Unsigus Try to reduce number  of CCCDs from menuconfig to reproduce this issue, `make menuconfig --> Component config → Bluetooth → NimBLE Options --> Maximum number of CCC descriptors to save across reboots `. 

Apart from this just want to correct my earlier comment 

> if application needs more than 8 CCDs per device ...

The CCCDs configured are total CCCDs and not per device.

closed 2020-05-22 by espressif-bot

prasad-alatkar, 2020-05-22:
Hi @Unsigus The fix for this issue has been merged in mainstream `esp-idf`(Ref. commit: 2ef48900).

ZweiEuro, 2021-02-23:
Hi
I know I am a little late to the issue, but I get the same behavior after incorporating:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_server
into my project.

I ma not using Nimble though, so should I open a new issue for this ? 
The error itself seems to be the same. 

```
E (20458) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (20458) task_wdt:  - IDLE0 (CPU 0)
E (20458) task_wdt: Tasks currently running:
E (20458) task_wdt: CPU 0: btController
E (20458) task_wdt: CPU 1: IDLE1
E (20458) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x401383FB:0x3FFBE8C0 0x40082E71:0x3FFBE8E0 0x40185A7D:0x3FFB5E90 0x4018602D:0x3FFB5EB0 0x4008FED5:0x3FFB5EE0 

E (20458) task_wdt: Print CPU 1 backtrace

Backtrace:0x40081DC3:0x3FFBEEC0 0x40082E71:0x3FFBEEE0 0x4000BFED:0x3FFBC870 0x40090135:0x3FFBC880 0x4013869F:0x3FFBC8A0 0x401386AB:0x3FFBC8D0 0x40138761:0x3FFBC8F0 0x40090B0D:0x3FFBC910 0x4008FED5:0x3FFBC930 

E (25458) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (25458) task_wdt:  - IDLE0 (CPU 0)
E (25458) task_wdt: Tasks currently running:
E (25458) task_wdt: CPU 0: btController
E (25458) task_wdt: CPU 1: IDLE1
E (25458) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x401383FB:0x3FFBE8C0 0x40082E71:0x3FFBE8E0 0x40185A7D:0x3FFB5E90 0x4018602D:0x3FFB5EB0 0x4008FED5:0x3FFB5EE0 

E (25458) task_wdt: Print CPU 1 backtrace

Backtrace:0x40081DC3:0x3FFBEEC0 0x40082E71:0x3FFBEEE0 0x4000BFED:0x3FFBC870 0x40090135:0x3FFBC880 0x4013869F:0x3FFBC8A0 0x401386AB:0x3FFBC8D0 0x40138761:0x3FFBC8F0 0x40090B0D:0x3FFBC910 0x4008FED5:0x3FFBC930 

E (30458) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (30458) task_wdt:  - IDLE0 (CPU 0)
E (30458) task_wdt: Tasks currently running:
E (30458) task_wdt: CPU 0: btController
E (30458) task_wdt: CPU 1: IDLE1
E (30458) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x401383FB:0x3FFBE8C0 0x40082E71:0x3FFBE8E0 0x40185A7D:0x3FFB5E90 0x4018602D:0x3FFB5EB0 0x4008FED5:0x3FFB5EE0 

E (30458) task_wdt: Print CPU 1 backtrace

Backtrace:0x40081DC3:0x3FFBEEC0 0x40082E71:0x3FFBEEE0 0x4000BFED:0x3FFBC870 0x40090135:0x3FFBC880 0x4013869F:0x3FFBC8A0 0x401386AB:0x3FFBC8D0 0x40138761:0x3FFBC8F0 0x40090B0D:0x3FFBC910 0x4008FED5:0x3FFBC930 
```

hsteinhaus, 2025-04-30:
Another crash with BLE tracker enabled without any BLE devices in range:
```
[07:28:15][D][debug:311]: EFuse MAC: 2C:BC:BB:82:96:30
[07:28:15][D][api:103]: Accepted 10.22.20.37t due to task watchdog
[07:28:15][D][api.connection:1801]: Home Assistant 2025.4.4 (10.22.20.37): Connected successfully
[07:28:43][I][safe_mode:041]: Boot seems successful; resetting boot]loo: Saving 1 preferences to flash...BT, Cores:2 Revision:301|
[07:28:43][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failedakeup: Unknown'
[07:31:35]:0x00,q_drv:0x00,d_drv:0x00,cs0configsip: 0, SPIWP:0xeeg
[07:31:35]clk_drviowime:r:064]: 'Reset Reason': Sending state 'Reset due to task watchdog'
[07:31:35]E (234395) task_wdt:  - loopTask (CPU 1)
[07:31:35]E (234395) task_wdt: Tasks currently running:Address    Size      
[07:31:35]E (234395) task_wdt: CPU 0: IDLE0   0        0x00009000 0x00002000
[07:31:35]E (234395) task_wdt: CPU 1: IDLE1   1        0x0000B000 0x00001000
[07:31:35]E (234395) task_wdt: Aborting.00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[07:31:35]mode:DIO, clock div:11         0    17       0x001D0000 0x001C0000
[07:31:35]load:0x3fff0030,len:7112       1    2        0x00390000 0x0006D000
[07:31:35]load:0x40078000,len:15864 (34246) wifi:a
[07:31:35]load:0x40080400,len:4
[07:31:35]load:0x40080404,len:3892
[07:31:35]entry 0x4008064c
```

My BLE-related configuration:
```
esp32_ble_tracker:
#  max_connections: 3
  scan_parameters:
    interval: 10ms
    window: 10ms
    active: false

victron_ble:
  - id: MySmartShunt
    mac_address: ${smart_shunt_mac_address}
    bindkey: ${smart_shunt_encryption_key}

sensor:
  # MySmartShunt
  - platform: victron_ble
    victron_ble_id: MySmartShunt
    name: "State of charge"
    type: STATE_OF_CHARGE
    id: soc
    web_server:
      sorting_group_id: bat_data
```

As soon as I remove BLE from config, the device is running rock-solid. The time to crash with BLE enabled varies hevily - from below one minute until more than 30 minutes. Mostly without 2 or 3 minutes.