espressif/esp-idf #4415: EWifi can not connect to Android hot spot

Reiner1210, 2019-11-28:
Hello,

With the new IDF WiFi can no longer connect to my Android hot spot.
I tried it with you station example. here the vebose outpu:

<code>
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7132
load:0x40078000,len:14824
load:0x40080400,len:4468
entry 0x400806ec
I (73) boot: Chip Revision: 1
I (73) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot: ESP-IDF v4.1-dev-1086-g93a8603c5 2nd stage bootloader
I (39) boot: compile time 11:05:00
I (39) boot: Enabling RNG early entropy source...
I (45) boot: SPI Speed      : 40MHz
I (49) boot: SPI Mode       : DIO
I (53) boot: SPI Flash Size : 4MB
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1ac70 (109680) map
I (150) esp_image: segment 1: paddr=0x0002ac98 vaddr=0x3ffb0000 size=0x03378 ( 13176) load
I (156) esp_image: segment 2: paddr=0x0002e018 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/reiner/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

I (157) esp_image: segment 3: paddr=0x0002e420 vaddr=0x40080400 size=0x01bf8 (  7160) load
I (169) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x6af8c (438156) map
0x400d0020: _stext at ??:?

I (331) esp_image: segment 5: paddr=0x0009afb4 vaddr=0x40081ff8 size=0x130a4 ( 77988) load
0x40081ff8: _xt_user_exit at /home/reiner/esp/esp-idf/components/freertos/xtensa_vectors.S:790

I (377) boot: Loaded app from partition at offset 0x10000
I (377) boot: Disabling RNG early entropy source...
I (378) cpu_start: Pro cpu up.
I (381) cpu_start: Application information:
I (386) cpu_start: Project name:     wifi_station
I (391) cpu_start: App version:      v4.1-dev-1086-g93a8603c5
I (398) cpu_start: Compile time:     Nov 28 2019 11:04:59
I (404) cpu_start: ELF file SHA256:  b04f36b3758455e6...
I (410) cpu_start: ESP-IDF:          v4.1-dev-1086-g93a8603c5
I (416) cpu_start: Starting app cpu, entry point is 0x40081198
0x40081198: call_start_cpu1 at /home/reiner/esp/esp-idf/components/esp32/cpu_start.c:276

I (0) cpu_start: App cpu up.
V (427) memory_layout: reserved range is 0x3f41ac68 - 0x3f41ac90
D (432) memory_layout: Checking 7 reserved memory ranges:
D (438) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (444) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb9248
D (451) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (457) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (464) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (470) memory_layout: Reserved memory range 0x40078000 - 0x40080000
0x40080000: _WindowOverflow4 at /home/reiner/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

[... 1006 lines cut ...]
D (25990) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (26000) event: running post WIFI_EVENT:5 with handler 0x400d2bbc on loop 0x3ffbe73c
0x400d2bbc: event_handler at /home/reiner/esp/esp-idf/examples/wifi/getting_started/station/main/station_example_main.c:44

V (26010) esp_adapter: thread sem get: sem=0x3ffc5b0c
I (26010) wifi station: retry to connect to the AP
I (26020) wifi station: connect to the AP fail
D (28060) event: running post WIFI_EVENT:5 with handler 0x400eaccc on loop 0x3ffbe73c
0x400eaccc: wifi_default_action_sta_disconnected at /home/reiner/esp/esp-idf/components/esp_wifi/src/wifi_default.c:103

D (28060) esp_netif_handlers: esp_netif action disconnected with netif0x3ffbf66c from event_id=5
D (28070) esp_netif_lwip: check: remote, if=0x3ffbf66c fn=0x400ea85c
0x400ea85c: esp_netif_down_api at /home/reiner/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1035

D (28080) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffbf66c
D (28080) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffbf66c
D (28090) esp_netif_lwip: if0x3ffbf66c start ip lost tmr: no need start because netif=0x3ffbf6e0 interval=120 ip=0
V (28100) esp_netif_objects: esp_netif_next_unsafe 0x0
V (28100) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffbf66c
V (28110) esp_netif_objects: esp_netif_next_unsafe 0x3ffbf66c
D (28120) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (28120) event: running post WIFI_EVENT:5 with handler 0x400d2bbc on loop 0x3ffbe73c
0x400d2bbc: event_handler at /home/reiner/esp/esp-idf/examples/wifi/getting_started/station/main/station_example_main.c:44

</code>

negativekelvin, 2019-11-29:
> With the new IDF WiFi 

Specify a commit

liuzfesp, 2019-12-31:
Hi @Reiner1210 does this issue still exists on your side?

Alvin1Zhang, 2019-12-31:
@Reiner1210 Thanks for reporting. @negativekelvin Thanks for the notes. Feel free to reopen if the issue still happens. Thanks.

closed 2019-12-31 by Alvin1Zhang

HaydnLiao, 2019-12-31:
I got this issue in v4.0-dev. In v4.0-bate2 it don't happen.

Reiner1210, 2020-01-01:
I confirm that it seems to work on
v4.1-dev-1572-g30372f5a4