espressif/esp-idf #4417: Issue with Finishing ESP-IDF Installation on Windows 7 64-bit

Stoi723, 2019-11-28:
Installation on Windows 7 64-bit never end when running :

Running command: powershell -ExecutionPolicy ByPass -File "C:\Users\s\.espressif\dist\tools_WD_excl.ps1" -AddExclPath "C:\Users\s\.espressif\*.exe"

[image]

squonk11, 2019-12-17:
I had the same problem. I simply solved it by not ticking the box for setting the Windows Defender exceptions. This can be done in the last dialog box before statring the installation.

igrr, 2019-12-18:
Please try the latest version of the installer (v2.2), it should fix the issue.

Alvin1Zhang, 2020-01-07:
@Stoi723 Thanks for reporting. Would you please help check whether your issue has been fixed by trying the latest version of the installer (v2.2)? Thanks.

pacucha42, 2020-01-07:
This issue is solved by the latest fix for https://github.com/espressif/esp-idf/issues/4225 (to be released soon). It's quite likely you don't have Windows Defender cmdlets installed (can be checked by running **Get-Command -Module Defender**  from PS cmdline, see the linked issue), the latest fix does the detection for you and disables whole WD function.
Sorry for inconvenience

Alvin1Zhang, 2020-05-27:
@Stoi723 Thanks for reporting. Would you please help if any updates for the issue? Has your issue been resolved? Thanks.

georgik, 2021-04-13:
@squonk11 We updated ESP-IDF Tools Windows Installer. The checkbox was moved to the last screen of the installation process because it also requires elevation of privileges. Having the checkbox on the last screen helps to avoid the problem with the timeout of UAC. 

We also added an Antivirus check and the checkbox is only available when Windows Defender is detected.

Please, test the latest installer: https://dl.espressif.com/dl/esp-idf/

closed 2021-04-16 by Alvin1Zhang