espressif/esp-idf #4419: Signed App Images Not Working with ESP-IDF v3.3

TonyTrupp, 2019-11-29:
## Environment

- Development Kit:  NA, happens during build, not during/after flashing 
- Kit version (for WroverKit/PicoKit/DevKitC): NA
- Module or chip used:  ESP32-WROOM-32D
- IDF version: v3.3
- Build System:   CMake
- Compiler version: 5.2.0
- Operating System: macOS 
- Power Supply:  NA, happens during build, not during/after flashing 

## Problem Description

The build won't complete with Secure Features - > Require Signed App Images & Verify app signature on update enabled with ESP-IDF 3.3. I need to use 3.3 because Arduino-ESP32 doesn't support any later versions.   

This appears to be similar to the issue described here: 
https://github.com/espressif/esp-idf/issues/3864  

And fixed within this commit within 4.1: 
https://github.com/espressif/esp-idf/commit/75488f18060553d7853e3c93cb6264d112b8eb07

Since this is an important feature, and Arduino-ESP32 users don't have the ability to update to the latest ESP-IDF version, I'm hoping this could be included as part of an 3.3.1 patch release.  

### Expected Behavior 

The signed build should complete without errors. 
 

### Actual Behavior

The build won't complete with Secure Features - > Require Signed App Images & Verify app signature on update enabled.  It builds fine with those features turned off. 

Here's the error I'm getting: 
undefined reference to `_binary_signature_verification_key_bin_end'
esp-idf/bootloader_support/libbootloader_support.a(secure_boot_signatures.c.obj):(.literal.esp_secure_boot_verify_signature_block+0x4): undefined reference to `_binary_signature_verification_key_bin_start'

### Steps to repropduce

1. Turn on Secure Features - > Require Signed App Images & Verify app signature
2. idf.py fullclean 
3. idf.py build

### Code to reproduce this issue

I've created a stripped down example project that demonstrates this issue here:  
https://github.com/TonyTrupp/esp-idf-3.3-signed-builds-bug

## Debug Logs

```
Python requirements from /Users/tonytrupp/esp/esp-idf/requirements.txt are satisfied.
Running ninja in directory /Users/tonytrupp/esp/SignedBuildsBug/build
Executing "ninja all"...
[1/6] Performing build step for 'bootloader'
ninja: no work to do.
[2/4] Linking CXX executable SignedBuildsBug.elf
FAILED: SignedBuildsBug.elf 
: && /Users/tonytrupp/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls  -nostdlib CMakeFiles/SignedBuildsBug.elf.dir/dummy_main_src.c.obj  -o SignedBuildsBug.elf  -Wl,--cref -Wl,--Map=SignedBuildsBug.map -Wl,--gc-sections -Wl,--start-group esp-idf/soc/libsoc.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/xtensa-debug-module/libxtensa-debug-module.a esp-idf/app_trace/libapp_trace.a esp-idf/freertos/libfreertos.a esp-idf/vfs/libvfs.a esp-idf/newlib/libnewlib.a esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/driver/libdriver.a esp-idf/esp_event/libesp_event.a esp-idf/ethernet/libethernet.a esp-idf/mbedtls/libmbedtls.a esp-idf/micro-ecc/libmicro-ecc.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/app_update/libapp_update.a esp-idf/spi_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/lwip/liblwip.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/pthread/libpthread.a esp-idf/smartconfig_ack/libsmartconfig_ack.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/espcoredump/libespcoredump.a esp-idf/esp32/libesp32.a esp-idf/cxx/libcxx.a esp-idf/console/libconsole.a esp-idf/mdns/libmdns.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/sdmmc/libsdmmc.a esp-idf/fatfs/libfatfs.a esp-idf/spiffs/libspiffs.a esp-idf/openssl/libopenssl.a esp-idf/arduino-esp32/libarduino-esp32.a esp-idf/asio/libasio.a esp-idf/jsmn/libjsmn.a esp-idf/coap/libcoap.a esp-idf/nghttp/libnghttp.a esp-idf/esp-tls/libesp-tls.a esp-idf/esp_adc_cal/libesp_adc_cal.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/esp_https_ota/libesp_https_ota.a esp-idf/esp_https_server/libesp_https_server.a esp-idf/expat/libexpat.a esp-idf/freemodbus/libfreemodbus.a esp-idf/json/libjson.a esp-idf/libsodium/liblibsodium.a esp-idf/mqtt/libmqtt.a esp-idf/protobuf-c/libprotobuf-c.a esp-idf/protocomm/libprotocomm.a esp-idf/ulp/libulp.a esp-idf/unity/libunity.a esp-idf/wifi_provisioning/libwifi_provisioning.a esp-idf/main/libmain.a -lgcov esp-idf/mbedtls/mbedtls/library/libmbedtls.a esp-idf/mbedtls/mbedtls/library/libmbedx509.a esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a esp-idf/soc/libsoc.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/freertos/libfreertos.a -Wl,--undefined=uxTopUsedPriority esp-idf/newlib/libnewlib.a -L /Users/tonytrupp/esp/esp-idf/components/newlib/lib -lc -lm esp-idf/esp32/libesp32.a -L /Users/tonytrupp/esp/esp-idf/components/esp32/lib -lcoexist -lcore -lespnow -lmesh -lnet80211 -lphy -lpp -lrtc -lsmartconfig -lwpa2 -lwpa -lwps -L /Users/tonytrupp/esp/SignedBuildsBug/build/esp-idf/esp32 -T esp32_out.ld -T /Users/tonytrupp/esp/SignedBuildsBug/build/esp-idf/esp32/esp32.project.ld -L /Users/tonytrupp/esp/esp-idf/components/esp32/ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld /Users/tonytrupp/esp/esp-idf/components/esp32/libhal.a -lgcc -u call_user_start_cpu0 -u ld_include_panic_highint_hdl esp-idf/cxx/libcxx.a -lstdc++ -u __cxa_guard_dummy -u esp_app_desc && :
esp-idf/bootloader_support/libbootloader_support.a(secure_boot_signatures.c.obj):**(.literal.esp_secure_boot_verify_signature_block+0x0): undefined reference to `_binary_signature_verification_key_bin_end'
esp-idf/bootloader_support/libbootloader_support.a(secure_boot_signatures.c.obj):(.literal.esp_secure_boot_verify_signature_block+0x4): undefined reference to `_binary_signature_verification_key_bin_start'**
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

## Other items if possible

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3903938/sdkconfig.txt)

TonyTrupp, 2019-11-29:
@projectgus I created another repo that's even simpler to demonstrate this problem.  This version does not include the arduino-esp32 library, so we can rule out that it's not a bug introduced by that repo. It breaks as soon as esp_ota_set_boot_partition() is included.  
https://github.com/TonyTrupp/esp-idf-3.3-signed-builds-bug-without-arduino-esp32

chegewara, 2019-11-29:
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#minimal-component-cmakelists

You need to add `REQUIRES` libraries that you want to use.

And this for certificates:
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#embedding-binary-data

TonyTrupp, 2019-11-29:
@chegewara Thanks for helping me out, but I'm not sure what libraries I need to explicitly add to REQUIRES. Can you please provide an example of what I need to add? 

I tried adding that pem like this: 
set(COMPONENT_EMBED_TXTFILES ../secure_boot_signing_key.pem) 
...as described in the 3.3 documentation: 
https://docs.espressif.com/projects/esp-idf/en/v3.3/api-guides/build-system-cmake.html#embedding-binary-data

But I'm still seeing the same issue after doing that.  Are you able to get this working on that git repo that I provided? 

I'm also not seeing references to either of these requirements within the secure boot documentation. If these are requirements, perhaps it should be added to the instructions there?  
https://docs.espressif.com/projects/esp-idf/en/v3.3/security/secure-boot.html#signed-app-verify-howto

chegewara, 2019-11-30:
I found solution to make it build but it is far from perfect, probably its only workaround.
In `main/CmakeLists.txt` add this line `set(COMPONENT_EMBED_FILES "../build/bootloader/signature_verification_key.bin")`.

The problem with this solution is that i had to comment it out to build bootloader first `idf.py bootloader` to create this file. Then uncomment it and `idf.py build`.
I hope this helps you to build project and maybe @projectgus can help with it better.

TonyTrupp, 2019-12-01:
Well that's some progress I guess.  I don't understand the esp-idf internals that well, but my assumption is that it may require a change to the bootloader_support cmakelists, similar to what projectgus did here: 
https://github.com/espressif/esp-idf/commit/75488f18060553d7853e3c93cb6264d112b8eb07 

If that's it, hopefully he can come up with a similar solution that's compatible with esp-idf 3.3 🤞

projectgus, 2019-12-02:
> If that's it, hopefully he can come up with a similar solution that's compatible with esp-idf 3.3 crossed_fingers

Hi @TonyTrupp ,

I think you're right that a similar fix can be applied to fix this exact problem on IDF v3.3. However I'm not sure that Secure Boot will work with just this change.

The reason is that we don't actually support Secure Boot or Flash Encryption in the CMake build system on ESP-IDF V3.3. CMake is a Preview in this version.

There's a short list of unsupported features at the top of the Get Started guide:
https://docs.espressif.com/projects/esp-idf/en/stable/get-started-cmake/index.html

Full CMake support in a stable release will start with the ESP-IDF V4.0 stable release.

Is moving back to the default GNU Make build system an option for you?

Angus

TonyTrupp, 2019-12-03:
Thanks Angus, that's helpful. I missed that message about encryption not being supported with CMake within esp-idf 3.3.  It'll be a bit of refactoring to get it to work with Make, but at least I have a path forward now. Much appreciated.

closed 2019-12-03 by TonyTrupp

projectgus, 2019-12-03:
@TonyTrupp Good to hear. This is partially our fault for leaving these options visible in KConfig when CMake is in use, will update the next v3.3.x release to hide these when CMake is being used.

chegewara, 2019-12-03:
@projectgus Sorry for interruption, but maybe leave it with info it is not working straight (is not supported), since with some workaround it eventually can be used?
Hiding this option with KConfig it wont be possible to use it at all.

projectgus, 2019-12-03:
@chegewara If someone wants to modify the IDF source to make it work then they can also revert the commit that disables the Kconfig items (it will be linked to this GitHub issue once merged).

By the way, if anyone has a use case for why they need Secure Boot support with CMake in v3.3.x, then please let us know and we'll discuss backporting the support from v4.0.

shangl, 2019-12-10:
@projectgus Do you already know when a stable release of v4.0 will be available. We ported our framework to use CMake but we also need the secure boot / flash encryption feature for our software.

martinius96, 2019-12-15:
@projectgus Hello there, Is it possible to sign app image and verify it on update/via bootloader in ESP-IDF v3.3.1? Or solution will be only at 4.0+ version?

projectgus, 2019-12-15:
@shangl Release Candidate (RC) testing is in progress now, so we're hoping to release the final version in January.

@martinius96 This should work in both v3.3 and v3.3.1 if you use the `GNU Make` build system as described in the Getting Started Guide.  This issue relates to using the CMake build system to build (ie running `cmake` or `idf.py` as the build tool), only.

If you're using CMake build system, this feature will be available in v4.0.

If you're using GNU Make build system and it's not working, please open a new issue with more details about what is not working.