espressif/esp-idf #4421: some questions about esp-tls

xcguang, 2019-11-29:
https://github.com/espressif/esp-idf/blob/93a8603c545fb8e54741d6685146e2f3b874378d/components/esp-tls/esp_tls.h#L252

Hi,
the struct is 
```
typedef struct esp_tls {
#ifdef CONFIG_ESP_TLS_USING_MBEDTLS
    mbedtls_ssl_context ssl;                                                    /*!< TLS/SSL context */
 
    mbedtls_entropy_context entropy;                                            /*!< mbedTLS entropy context structure */
 
    mbedtls_ctr_drbg_context ctr_drbg;                                          /*!< mbedTLS ctr drbg context structure.
                                                                                     CTR_DRBG is deterministic random 
                                                                                     bit generation based on AES-256 */
 
    mbedtls_ssl_config conf;                                                    /*!< TLS/SSL configuration to be shared 
                                                                                     between mbedtls_ssl_context 
                                                                                     structures */
 
    mbedtls_net_context server_fd;                                              /*!< mbedTLS wrapper type for sockets */
 
    mbedtls_x509_crt cacert;                                                    /*!< Container for the X.509 CA certificate */

    mbedtls_x509_crt *cacert_ptr;                                               /*!< Pointer to the cacert being used. */

    mbedtls_x509_crt clientcert;                                                /*!< Container for the X.509 client certificate */

    mbedtls_pk_context clientkey;                                               /*!< Container for the private key of the client
                                                                                     certificate */
#ifdef CONFIG_ESP_TLS_SERVER
    mbedtls_x509_crt servercert;                                                /*!< Container for the X.509 server certificate */

    mbedtls_pk_context serverkey;                                               /*!< Container for the private key of the server
                                                                                   certificate */
#endif
#elif CONFIG_ESP_TLS_USING_WOLFSSL
    void *priv_ctx;
    void *priv_ssl;
#endif
    int sockfd;                                                                 /*!< Underlying socket file descriptor. */
 
    ssize_t (*read)(struct esp_tls  *tls, char *data, size_t datalen);          /*!< Callback function for reading data from TLS/SSL
                                                                                     connection. */
 
    ssize_t (*write)(struct esp_tls *tls, const char *data, size_t datalen);    /*!< Callback function for writing data to TLS/SSL
                                                                                     connection. */

    esp_tls_conn_state_t  conn_state;                                           /*!< ESP-TLS Connection state */

    fd_set rset;                                                                /*!< read file descriptors */

    fd_set wset;                                                                /*!< write file descriptors */

    bool is_tls;                                                                /*!< indicates connection type (TLS or NON-TLS) */

    esp_tls_role_t role;                                                        /*!< esp-tls role
                                                                                     - ESP_TLS_CLIENT
                                                                                     - ESP_TLS_SERVER */

    esp_tls_error_handle_t error_handle;                                        /*!< handle to error descriptor */

} esp_tls_t;
```
The macro `CONFIG_ESP_TLS_USING_WOLFSSL` and `CONFIG_ESP_TLS_USING_MBEDTLS` affect the size of the structure, can it be avoided?  When it's used in lib, it limits the project to wolfssl or mbedtls, otherwise, the exception occurs, which is not what I want. If so, OpenSSL APIs can achieve the same effect.

Alvin1Zhang, 2019-11-29:
@xcguang Thanks for reporting.

AdityaHPatwardhan, 2019-12-02:
@xcguang , esp-tls component has two underlying stacks , mbedtls and wolfssl , the basic rule is that they should be independent of each other.  what you request will be a major change at the component level which would require renaming lot of calls and variables. Another simpler solution might be to put all of it in union and keep it same for both However it makes wolfssl dependent on mbedtls and also it increases size of the structure very much which is a unreasonable penalty for wolfssl users, However I do not understand the motive behind putting  esp-tls in a library format and using it afterwards , can you please elaborate on your need ? , Thank you !

xcguang, 2019-12-03:
In my project, I need provide the libs, and the customer may port codes from other project to carry out secondary development based on mbedtls or wolfssl.  But in my libs, I need call esp-tls APIs, so I need use `esp_tls_t`, which will lead to the issue I said before. 
If ESP_TLS works the same as OpenSSL APIs, they have the same limitation, maybe I prefer to use OpenSSL. Should we make an independent middle layer about esp-tls, which does not care of the stack in the low layer? In this way, customers may prefer to use it instead of OpenSSL APIs, and it will more easy to use esp_tls.

AdityaHPatwardhan, 2019-12-04:
@xcguang ,Actually esp-tls is already an abstraction layer , so adding one more layer would be difficult to maintain.I guess for this to work we will have to put pointers to all the structures and take stack specific structure in its appropriate position ,is that what you ask for ? ( as a solution )

negativekelvin, 2019-12-04:
Can you provide 2 versions of libs and select one for linking based on CONFIG_ESP_TLS_USING_

xcguang, 2019-12-05:
@AdityaHPatwardhan I think the pointer is a solution, but the application should not  care of what the pointer points, maybe it is only for esp-tls to use.

@negativekelvin The multiple libs is also a solution, but it is not a good one, because we have provided some different libs for debug. If there is no choice, we have to do it.

AdityaHPatwardhan, 2019-12-06:
@xcguang , 
>  I think the pointer is a solution, but the application should not care of what the pointer points, maybe it is only for esp-tls to use.

Even with the pointer approach , You will not be able to compile both the tls-stack in one library ( as the stacks are alternative to each other ) , the source files for both the stacks are different which are mapped to esp-tls in our case. I think the reasonable solution here might be this -> 

> Can you provide 2 versions of libs and select one for linking based on CONFIG_ESP_TLS_USING_
If you have any alternate solution in mind please raise the PR.

xcguang, 2019-12-09:
It seems these variable wrapped in the macro are not used for application, are these two methods feasible？

1. the variable wrapped in the macro are defined as a pointer, the pointer points the wolfssl or mbedtls structure,
```
typedef struct esp_tls {
    void* priv;        // it depends on mbedtls or wolfssl, which not for application
    int sockfd;                                                                 /*!< Underlying socket file descriptor. */
 
    ssize_t (*read)(struct esp_tls  *tls, char *data, size_t datalen);          /*!< Callback function for reading data from TLS/SSL
                                                                                     connection. */
 
    ssize_t (*write)(struct esp_tls *tls, const char *data, size_t datalen);    /*!< Callback function for writing data to TLS/SSL
                                                                                     connection. */

    esp_tls_conn_state_t  conn_state;                                           /*!< ESP-TLS Connection state */

    fd_set rset;                                                                /*!< read file descriptors */

    fd_set wset;                                                                /*!< write file descriptors */

    bool is_tls;                                                                /*!< indicates connection type (TLS or NON-TLS) */

    esp_tls_role_t role;                                                        /*!< esp-tls role
                                                                                     - ESP_TLS_CLIENT
                                                                                     - ESP_TLS_SERVER */

    esp_tls_error_handle_t error_handle;                                        /*!< handle to error descriptor */

} esp_tls_t;
```

2. move the variable wrapped in the macro to the end of the structure
```
typedef struct esp_tls {
    int sockfd;                                                                 /*!< Underlying socket file descriptor. */
 
    ssize_t (*read)(struct esp_tls  *tls, char *data, size_t datalen);          /*!< Callback function for reading data from TLS/SSL
                                                                                     connection. */
 
    ssize_t (*write)(struct esp_tls *tls, const char *data, size_t datalen);    /*!< Callback function for writing data to TLS/SSL
                                                                                     connection. */

    esp_tls_conn_state_t  conn_state;                                           /*!< ESP-TLS Connection state */

    fd_set rset;                                                                /*!< read file descriptors */

    fd_set wset;                                                                /*!< write file descriptors */

    bool is_tls;                                                                /*!< indicates connection type (TLS or NON-TLS) */

    esp_tls_role_t role;                                                        /*!< esp-tls role
                                                                                     - ESP_TLS_CLIENT
                                                                                     - ESP_TLS_SERVER */

    esp_tls_error_handle_t error_handle;                                        /*!< handle to error descriptor */

   // the following variable are not used for application
  #ifdef CONFIG_ESP_TLS_USING_MBEDTLS
    mbedtls_ssl_context ssl;                                                    /*!< TLS/SSL context */
 
    mbedtls_entropy_context entropy;                                            /*!< mbedTLS entropy context structure */
 
    mbedtls_ctr_drbg_context ctr_drbg;                                          /*!< mbedTLS ctr drbg context structure.
                                                                                     CTR_DRBG is deterministic random 
                                                                                     bit generation based on AES-256 */
 
    mbedtls_ssl_config conf;                                                    /*!< TLS/SSL configuration to be shared 
                                                                                     between mbedtls_ssl_context 
                                                                                     structures */
 
    mbedtls_net_context server_fd;                                              /*!< mbedTLS wrapper type for sockets */
 
    mbedtls_x509_crt cacert;                                                    /*!< Container for the X.509 CA certificate */

    mbedtls_x509_crt *cacert_ptr;                                               /*!< Pointer to the cacert being used. */

    mbedtls_x509_crt clientcert;                                                /*!< Container for the X.509 client certificate */

    mbedtls_pk_context clientkey;                                               /*!< Container for the private key of the client
                                                                                     certificate */
#ifdef CONFIG_ESP_TLS_SERVER
    mbedtls_x509_crt servercert;                                                /*!< Container for the X.509 server certificate */

    mbedtls_pk_context serverkey;                                               /*!< Container for the private key of the server
                                                                                   certificate */
#endif
#elif CONFIG_ESP_TLS_USING_WOLFSSL
    void *priv_ctx;
    void *priv_ssl;
#endif
} esp_tls_t;
```

AdityaHPatwardhan, 2019-12-16:
@xcguang I did not understand how moving the pointers to the end of the structure would solve the problem , can you please help raise a PR , which would compile successfully ,( What I think is providing two libs is the only options. as the internal source code for both stacks is different which is selected  at compile time with the `ESP_TLS_USING_###`flag, How can you have two implementations of same functions in `esp-tls` ? )

ustccw, 2019-12-25:
@Alvin1Zhang Maybe I have fully understood @xcguang  problem.
Please let me clarify it again.

the struct of `esp_tls_t` is public for developer, user could use some variables(sockfd or others) in `struct esp_tls_t`,
such at the following code:
```
esp_tls_t *tls = esp_tls_conn_new(...);  // establish link
// the following code will lead to an wrong assignment of xxxfd, because sockfd has a different offset in struct esp_tls_t while using between wolfssl and mbedtls
// then, it will lead to fatal crash 
int xxxfd = tls->sockfd;

// manage link by xxxfd
FD_SET(xxxfd, &readfds);
FD_SET(xxxfd, &expfds);
...
FD_ISSET(xxxfd, &readfds);
ioctl(xxxfd, ...);
...
```

**So We expect:**
- the same variables(in mbedtls and wolfssl) can move to the beginning of `struct esp_tls_t`.
- the different variables(in mbedtls and wolfssl) can get to by more esp-tls APIs, if necessary.

In that way, `esp-tls` would be independent, whatever using mbedtls or wolfssl.

ustccw, 2019-12-25:
@AdityaHPatwardhan @Alvin1Zhang 
At present, we also encounter this `esp-tls` problem, when using sockfd in `struct esp_tls_t`.

As the last reply said, my esp-tls code works well in mbedtls way. But will crash if switch esp-tls to wolfssl. because of an wrong assignment of xxxfd.

Expect your reply.

mahavirj, 2020-01-22:
@xcguang This should be addressed with https://github.com/espressif/esp-idf/commit/c7f44a301dce4e9bb5064893cdea58fa656eb748 (New API has been added to retrieve `sockfd` from esp-tls session handle). Please feel free to reopen if you face any issues.

closed 2020-01-22 by mahavirj