espressif/esp-idf #4422: Error in i2c driver

bitmakk, 2019-11-29:
## Environment

- Development Kit:      [Custom]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version master, v4.1-dev-1086-g93a8603c5
- Build System:         [CMake]
- Compiler version  xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     [Windows]
- Power Supply:         [external 5V]

## Problem Description

I have an i2c touchpad, from which I read data, when interrupt arrives on standard GPIO.
The ESP is configured in master mode, and works a few seconds. Then i get the attached crash.
I debugged the given lines, and got that the p_i2c->rx_ring_buf will be NULL pointer before line 514 in driver/i2c.c after one use.
The declaration says: "RingbufHandle_t rx_ring_buf;     /*!< rx ringbuffer handler of slave mode */"
So this attribute should not be used in Master mode I think.

### Expected Behavior
The i2c doesn't crash, the driver doesn't uses the slave's buffer in Master mode
### Actual Behavior
The driver uses the rx_ring_buf attribute of i2c_obj_t, so it crashes
### Steps to repropduce

step 1: install i2c driver with proper config
step2: subscribe to interrupts
step3: send interrupts with the touchpad (maximum 1-2 ms time between them) 

### Code to reproduce this issue

https://gist.github.com/bitmakk/2d07ee3777d05245ff8b7cb7fe154887

## Debug Logs

```
C:/ESP/esp-idf/components/esp_ringbuf/ringbuf.c:1060 (xRingbufferSendFromISR)- assert failed!
abort() was called at PC 0x4008724b on core 1
0x4008724b: xRingbufferSendFromISR at C:/ESP/esp-idf/components/esp_ringbuf/ringbuf.c:1060 (discriminator 1)

ELF file SHA256: 382754ed3c31928e057af64c0c94e987103e6f8affddc27813b6aa336bf27260

Backtrace: 0x40087ef5:0x3ffb0c00 0x40088281:0x3ffb0c20 0x4008724b:0x3ffb0c40 0x40084501:0x3ffb0c60 0x400828d9:0x3ffb0c90 0x40100df7:0x3ffccca0 0x400d2cf7:0x3ffcccc0 0x4008ca95:0x3ffccce0 0x4008ae61:0x3ffccd00
0x40087ef5: invoke_abort at C:/ESP/esp-idf/components/esp32/panic.c:157

0x40088281: abort at C:/ESP/esp-idf/components/esp32/panic.c:174

0x4008724b: xRingbufferSendFromISR at C:/ESP/esp-idf/components/esp_ringbuf/ringbuf.c:1060 (discriminator 1)

0x40084501: i2c_isr_handler_default at C:/ESP/esp-idf/components/driver/i2c.c:514

0x400828d9: _xt_lowint1 at C:/ESP/esp-idf/components/freertos/xtensa_vectors.S:1153

0x40100df7: esp_pm_impl_waiti at C:/ESP/esp-idf/components/esp32/pm_esp32.c:484

0x400d2cf7: esp_vApplicationIdleHook at C:/ESP/esp-idf/components/esp_common/src/freertos_hooks.c:63

0x4008ca95: prvIdleTask at C:/ESP/esp-idf/components/freertos/tasks.c:3381 (discriminator 1)

0x4008ae61: vPortTaskWrapper at C:/ESP/esp-idf/components/freertos/port.c:143
```

## Other items if possible
None, it contains sensitive data, but all my bo
- [X ] sdkconfig file 
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3905408/sdkconfig.txt)

- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

stickbreaker, 2019-12-04:
@bitmakk I think you are seeing an overrun condition.  The interrupt that is triggering in line 514 of `0x40084501: i2c_isr_handler_default at C:/ESP/esp-idf/components/driver/i2c.c:514` is triggering because the I2C RX_FIFO is overflowing.  The Driver only empties RX_FiFO when a Master READ command completes.  If the RX_FIFO overflows because of an error, the driver treats the interrupt as if it is in slave mode.

Can you log failing I2C transactions instead of just returning false
```c++
	if(ret != ESP_OK)
		return false;
```
And you should call 
```c++
	i2c_cmd_link_delete(cmd);
```
before you exit the function.

Chuck.

bitmakk, 2019-12-04:
I tried it, even when crashes, the i2c transactions return ESP_OK.

mws-rmain, 2019-12-11:
I am experiencing the same problem, and get the same exact error. 
```
C:/ESP/esp-idf/components/esp_ringbuf/ringbuf.c:1060 (xRingbufferSendFromISR)- assert failed!
abort() was called at PC 0x4008724b on core 1
0x4008724b: xRingbufferSendFromISR at C:/ESP/esp-idf/components/esp_ringbuf/ringbuf.c:1060 (discriminator 1)
```
In my case:
- the error occurs after a random number of sucessfull reads (anywhere from several to tens);
- the error only occurs if reading more than 11 bytes (you appear to be transfering 15);
- I was syncing to latest git: switching to branch 'release/v4.0' resolved the issue.
Hopefully the above will help you, and help to resolve this issue.

koobest, 2019-12-11:
Hi,
We have found a bug in i2c driver,
in `i2c_hw_fsm_reset()` we do not need to enable the `I2C_RXFIFO_FULL` interrupt because we are in master mode. Please remove the `I2C_RXFIFO_FULL` interrupt enable mask in `i2c_hw_fsm_reset()` and retry you code. If this is the root cause , we will  fix it soon.

thanks !!

mws-rmain, 2019-12-11:
Thanks.
Seems like that fixed it for me.
Thanks for the quick response!

bitmakk, 2019-12-16:
It fixed the problem!
You may commit the fix.

bitmakk, 2019-12-16:
fixed in https://github.com/espressif/esp-idf/commit/aac935ec81324578e29606a70b4708a308863382, I close for now, if the problem is about to reappear, it can be reopened

closed 2019-12-16 by bitmakk