espressif/esp-idf #4423: ESP32 IO_MUX GPIO_MATRIX

Paullake85, 2019-11-29:
Hello,

I'm new in ESP32 programming. Reading di esp_idf documentation is seems that through IO_MUX and/or GPIO MATRIX most of the peripheals available can be assigned to any GPIO pin, but I cannot understand how to do that.  Should I write a specific inizialization funtion for the pin assigment or should I use the menuconfig? Is there an example?

Thank you in advance,

Paul

negativekelvin, 2019-11-30:
Please ask general questions on the forum at esp32.com.

Each peripheral has an API to set the pins.

https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/

Alvin1Zhang, 2019-12-02:
@negativekelvin Thanks for the notes. @Paullake85 Thanks for reporting, for how questions or third party question, please help submit a post on our forum at esp32.com, thanks.

closed 2019-12-02 by Alvin1Zhang