espressif/esp-idf #4428: HTTP Parser fails when @ sign in user name

Reiner1210, 2019-12-02:
I think the http_parser fails when the following is given:
<code>http://user:pass@xxxx.com:8080/mydata/version.txt</code>
When now the <code>user</code> and/or <code>pass</code> field contains a @ sign the parser works not as expected.

Alvin1Zhang, 2019-12-03:
@Reiner1210 Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

negativekelvin, 2019-12-03:
If the username or password contains a reserved character it should be percent encoded

ESP-Marius, 2019-12-03:
Yeah, it's not a valid URL (see RFC 3986). If you want to by-pass the parser you could add it to the config struct, e.g.:

`esp_http_client_config_t config = {
    .url = "http://httpbin.org/basic-auth/user/passwd",
    .username = "us@er",
    .password = "pas@swd",
    .auth_type = HTTP_AUTH_TYPE_BASIC,
};`

Reiner1210, 2019-12-03:
Ok. I used it to make a call to no-ip DNS service. They use the email Adress as username. 
I have a own parser so I avoid the parser and use esp_tls_conn_new_sync directly.

Alvin1Zhang, 2019-12-04:
@ESP-Marius Thanks for the efforts. @Reiner1210 Thanks for reporting, feel free to reopen, thanks.

closed 2019-12-04 by Alvin1Zhang