espressif/esp-idf #4430: release 2019r2 doesn't compile => error: implicit declaration of function 'ESP_LOG_LEVEL'; did you mean 'ESP_LOGE'

vortex314, 2019-12-03:
MESSAGE: Entering directory `/Users/lieven/workspace/esp32-streams'
/bin/sh -c 'make DRIVE'
----------Building project:[ esp32-streams - Debug ]----------
Python requirements from /Users/lieven/esp/esp-idf/requirements.txt are satisfied.
GENCONFIG
App "esp32-streams" version: a32e94a-dirty
Toolchain path: /Users/lieven/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp-2019r2
Compiler version: 8.2.0
Python requirements from /Users/lieven/esp/esp-idf/requirements.txt are satisfied.
Building partitions from /Users/lieven/esp/esp-idf/components/partition_table/partitions_singleapp.csv...
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/bootloader_common.o
CC build/bootloader/bootloader_support/src/bootloader_efuse_esp32.o
CC build/bootloader/bootloader_support/src/bootloader_flash.o
CC build/bootloader/bootloader_support/src/bootloader_flash_config_esp32.o
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/bootloader/bootloader_support/src/bootloader_random.o
CC build/bootloader/bootloader_support/src/bootloader_utility.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o
CC build/bootloader/bootloader_support/src/esp32/bootloader_esp32.o
CC build/bootloader/bootloader_support/src/esp32/bootloader_sha.o
CC build/bootloader/bootloader_support/src/esp32/flash_encrypt.o
CC build/bootloader/bootloader_support/src/esp32/secure_boot.o
CC build/bootloader/bootloader_support/src/esp32/secure_boot_signatures.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/spi_flash/esp32/spi_flash_rom_patch.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/micro-ecc/micro-ecc/uECC.o
AR build/bootloader/micro-ecc/libmicro-ecc.a
CC build/bootloader/soc/esp32/cpu_util.o
CC build/bootloader/soc/esp32/emac_hal.o
CC build/bootloader/soc/esp32/gpio_periph.o
CC build/bootloader/soc/esp32/i2s_periph.o
CC build/bootloader/soc/esp32/interrupts.o
CC build/bootloader/soc/esp32/ledc_periph.o
CC build/bootloader/soc/esp32/rtc_clk.o
CC build/bootloader/soc/esp32/rtc_clk_init.o
CC build/bootloader/soc/esp32/rtc_init.o
CC build/bootloader/soc/esp32/rtc_periph.o
CC build/bootloader/soc/esp32/rtc_pm.o
CC build/bootloader/soc/esp32/rtc_sleep.o
CC build/bootloader/soc/esp32/rtc_time.o
CC build/bootloader/soc/esp32/rtc_wdt.o
CC build/bootloader/soc/esp32/sdio_slave_periph.o
CC build/bootloader/soc/esp32/sdmmc_periph.o
CC build/bootloader/soc/esp32/soc_memory_layout.o
CC build/bootloader/soc/esp32/spi_periph.o
CC build/bootloader/soc/src/lldesc.o
CC build/bootloader/soc/src/memory_layout_utils.o
CC build/bootloader/soc/src/soc_include_legacy_warn.o
CC build/bootloader/soc/src/hal/spi_flash_hal.o
CC build/bootloader/soc/src/hal/spi_flash_hal_iram.o
CC build/bootloader/soc/src/hal/spi_hal.o
CC build/bootloader/soc/src/hal/spi_hal_iram.o
CC build/bootloader/soc/src/hal/spi_slave_hal.o
CC build/bootloader/soc/src/hal/spi_slave_hal_iram.o
AR build/bootloader/soc/libsoc.a
CC build/bootloader/main/bootloader_start.o
AR build/bootloader/main/libmain.a
CC build/bootloader/efuse/esp32/esp_efuse_table.o
CC build/bootloader/efuse/src/esp_efuse_api.o
CC build/bootloader/efuse/src/esp_efuse_fields.o
CC build/bootloader/efuse/src/esp_efuse_utility.o
CC build/bootloader/efuse/src/esp32/esp_efuse_api.o
CC build/bootloader/efuse/src/esp32/esp_efuse_fields.o
CC build/bootloader/efuse/src/esp32/esp_efuse_utility.o
AR build/bootloader/efuse/libefuse.a
LD build/bootloader/bootloader.elf
esptool.py v2.9-dev

CC build/Common/jsmn.o
CC build/Common/printf.o
CXX build/Common/Base64.o
CXX build/Common/BufferedByteStream.o
CXX build/Common/Bytes.o
CXX build/Common/CborQueue.o
CXX build/Common/CircBuf.o
CXX build/Common/Config.o
CXX build/Common/EventSource.o
CXX build/Common/Json.o
CXX build/Common/LinkedList.o
CXX build/Common/Log.o
CXX build/Common/Msgpack.o
CXX build/Common/Node.o
CXX build/Common/PIC32_stubs.o
CXX build/Common/Register.o
CXX build/Common/RtosQueue.o
CXX build/Common/STM32_stubs.o
CXX build/Common/Sema.o
CXX build/Common/Str.o
CXX build/Common/Strpack.o
CXX build/Common/Sys.o
CXX build/Common/Uid.o
CXX build/Common/Xdr.o
CXX build/Common/XdrQueue.o
AR build/Common/libCommon.a
CC build/app_trace/app_trace.o
CC build/app_trace/app_trace_util.o
CC build/app_trace/heap_trace_tohost.o
CC build/app_trace/host_file_io.o
CC build/app_trace/gcov/gcov_rtio.o
AR build/app_trace/libapp_trace.a
App "esp32-streams" version: a32e94a-dirty
CC build/app_update/esp_app_desc.o
CC build/app_update/esp_ota_ops.o
In file included from /Users/lieven/esp/esp-idf/components/app_update/esp_ota_ops.c:37:
/Users/lieven/esp/esp-idf/components/app_update/esp_ota_ops.c: In function 'read_otadata':
/Users/lieven/esp/esp-idf/components/log/include/esp_log.h:341:41: error: implicit declaration of function 'ESP_LOG_LEVEL'; did you mean 'ESP_LOGE'? [-Werror=implicit-function-declaration]
         if ( LOG_LOCAL_LEVEL >= level ) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
                                         ^~~~~~~~~~~~~
/Users/lieven/esp/esp-idf/components/log/include/esp_log.h:285:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGE( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR,   tag, format, ##__VA_ARGS__)
                                      ^~~~~~~~~~~~~~~~~~~
/Users/lieven/esp/esp-idf/components/app_update/esp_ota_ops.c:80:9: note: in expansion of macro 'ESP_LOGE'
         ESP_LOGE(TAG, "not found otadata");
         ^~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [esp_ota_ops.o] Error 1
make[1]: *** [component-app_update-build] Error 2
make: *** [DRIVE] Error 2
====3 errors, 9 warnings, total time: 00:00:14 seconds====

igrr, 2019-12-16:
Hi @vortex314, could you please add the information requested in the issue template? In particular, the ESP-IDF commit ID, code to reproduce the issue, whether you can reproduce the same issue using any of the ESP-IDF examples?

vortex314, 2019-12-16:
Hi Ivan,

thanks for inquiring. Great stuff you guys are building !
I had the issue on my MacBook Pro, since then I re-installed esp-idf and compiler. The problem has gone. So I will close the issue.

vortex314, 2019-12-16:
problem gone after re-installs.

closed 2019-12-16 by vortex314