espressif/esp-idf #4432: Feature request: allow to pass HSM-backed client key to esp-tls

wouterdevinck, 2019-12-03:
## Environment

- Development Kit: custom board
- Module or chip used:  ESP32-WROOM-32 + ATECC608A
- IDF version: v3.3
- Build System: Make
- Compiler version: 1.22.0-80-g6c4433a-5.2.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

When authenticating against an IOT framework (Azure, Amazon, etc.) using certificates, it is desirable to store the private key of the device in a secure element or HSM rather than in flash. Even if the flash is encrypted and secure boot is enabled, keys kept in flash (and loaded in memory) could leak through memory safety bugs in firmware. Secure elements generate and keep the private key in hardware and never give it up. Such secure elements then offer an API to perform cryptographic operations, such as signing (e.g. ECDSA) in the secure element rather than giving back the private key to perform these operations in software. An example of such a secure element is the ATECC608A.

Working with such a secure element in mbedtls requires passing a specific *mbedtls_pk_context* with a reference to the key in the secure element, rather than passing the content of the private key as a byte array. The esp-tls component does not currently expose this functionality.

This quick-and-dirty patch enables passing the *mbedtls_pk_context* struct through esp-tls to mbedtls:
https://github.com/wouterdevinck/lamp/blob/c5ced581a5/make/patches/0001-esp-tls.v3.3.patch

### Expected Behavior

Allow to pass a specific client PK context to the underlying TLS library (mbedtls in v3.3; mbedtls or wolfssl in v4.0). 

### Actual Behavior

The esp-tls API only allows passing a byte array and length with the contents of the PK.

### Steps to reproduce

n/a

### Code to reproduce this issue

Starting point without secure element:

https://github.com/espressif/esp-azure

Azure IOT "TLSIO" implementation using the modified esp-tls (see lines 225 and 239):

https://github.com/wouterdevinck/lamp/blob/c5ced581a/src/mcu/components/azureiot/port/tlsio_esp_atca.c

Azure IOT "HSM" implementation that does not return the private key (line 26):

https://github.com/wouterdevinck/lamp/blob/c5ced581a/src/mcu/components/azureiot/port/hsm.c

Modifications to esp-tls:

https://github.com/wouterdevinck/lamp/blob/c5ced581a5/make/patches/0001-esp-tls.v3.3.patch

Mbedtls configuration:

https://github.com/wouterdevinck/lamp/blob/c5ced581a5/src/mcu/components/mbedtls-mods/Makefile.projbuild

Mbedtls alternate implementations using the ATECC608A, done in Microchip's "cryptoauthlib" library, included as a submodule:

https://github.com/wouterdevinck/lamp/blob/c5ced581a5/src/mcu/components/cryptoauthlib/component.mk#L10

HAL for cryptoauthlib to talk with the secure element over I2C:

https://github.com/wouterdevinck/lamp/tree/c5ced581a5/src/mcu/components/cryptoauthlib_hal

## Debug Logs

n/a

## Other items if possible

n/a

fromeijn, 2020-02-05:
Good work Wouter, thanks for sharing! I would like this feature as well! Apart from encryption and secure boot, i think it's still a good idea to use a HSM.

fromeijn, 2020-02-17:
Hey @wouterdevinck how did you debug TLS, can't find what is causing a problem.
What I got now:
```
I (18436) tlsio: atca_mbedtls_ecdh_slot_cb
I (18516) tlsio: atca_mbedtls_ecdh_ioprot_cb
E (19136) esp-tls: mbedtls_ssl_handshake returned -0x7280
```
#define MBEDTLS_ERR_SSL_CONN_EOF                          -0x7280  /**< The connection indicated an EOF. */

In Azure I cant find any logging of authentication (that failed)

fromeijn, 2020-02-18:
found the problem, I used secp256k1 instead of secp256r1 (called prime256v1 in openssl). I'm using certificates/keys generated with `openssl` and load the private key on the HSM. Used `Wireshark` and `[ ] Enable mbedTLS debugging` in `make menuconfig` for debugging, didn't found any debugging tools on the Azure side

wouterdevinck, 2020-02-21:
@fromeijn Glad to read you got it working! This can indeed be pain in the ** to debug, since DPS tends to just close the connection without explanation if anything is wrong. The mbedTLS debugging option saved me a few times as well :)

closed 2020-05-22 by espressif-bot

wouterdevinck, 2020-05-24:
@AdityaHPatwardhan @mahavirj Cool to see quite a bit of work in this area on master. I would be willing to give this a try to see if it covers my use case without having to patch esp-idf, but a bit more info on how you approached things would be much appreciated.

AdityaHPatwardhan, 2020-05-24:
Hi @wouterdevinck, below is some general information about how `ATECC608A` is used for TLS,
* The support for `ATECC608A` has been added for `esp-tls` when used with `mbedtls`  as its internal ssl/tls stack. 
1) Firstly, user has to enable secure_element with menuconfig, refer [link](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/secure_element/atecc608_ecdsa#configuration) for more details
2) After making required configurations, user has to set `use_secure_element` to `true` in `esp_tls_cfg_t`. for details on how to use `ATECC608A` with `esp-tls` refer [link](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_tls.html#atecc608a-secure-element-with-esp-tls).
3) The above configurations make `mbedtls` to set pki context with help of hardware key from slot 0 of `ATECC608A` by making use of  some `ATECC608A` specific API provided by [esp-cryptoauthlib](https://github.com/espressif/esp-cryptoauthlib). you can take a look at the function [here](https://github.com/espressif/esp-idf/blob/7d75213674b4572f90c68162ad6fe9b16dae65ad/components/esp-tls/esp_tls_mbedtls.c#L606), which initialises `ATECC608A` interface and sets pki context.
4) When above configurations are enabled these two `mbedtls` specific flags are set
 :- `MBEDTLS_ECDSA_SIGN_ALT` and `MBEDTLS_ECDSA_VERIFY_ALT`,
with these flags set `mbedtls` finds for alternate implementation of `ecdsa_sign/verify` and that alternate implementation which signs and verifies the data with help of hardware key residing in `secure_element` is provided in [esp-cryptoauthlib](https://github.com/espressif/esp-cryptoauthlib), you can take a look at it [here](https://github.com/espressif/esp-cryptoauthlib/blob/master/cryptoauthlib/lib/mbedtls/atca_mbedtls_ecdsa.c). Thus ECDSA sign/verify with help of hardware key takes place.

> Note: [esp-cryptoauthlib](https://github.com/espressif/esp-cryptoauthlib) is a trimmed fork of original [cryptoauthlib](https://github.com/MicrochipTech/cryptoauthlib)

I hope this helps to solve your query,  please let me know if you want to know more about any point.

SalmanMalik7183, 2023-06-22:
Hi @wouterdevinck, 
were you able to connect the esp to Azure using the certificates from the secure element?
I am using atecc608b with the esp32c3.  I tried following the step provided by @AdityaHPatwardhan but they are performing the sign/verify operation using the secure element while I need to connect the device to the Azure cloud using the certificates from hsm. If you can help in anyway that would be great. Thanks