espressif/esp-idf #4434: BLE-Mesh with ESP CAM AI Thinker

Schaggo, 2019-12-03:
Hi, 
I'm fairly new to ESP and everything programming.

I'm working on a synchronously triggered camera array containing 60 ESP Cams. 
Currently I trigger image capture via a wired button. Things work nicely so far, but I was wondering if I could utilize BLE-Mesh to trigger all cams simultaneously.

Is it possible?

chegewara, 2019-12-03:
Yes, if you assign all device to a group address.

Schaggo, 2019-12-03:
> Yes, if you assign all device to a group address. But there may be small delay between devices response.

Thanks for the reply! 
I just had a look at the ble mesh examples and I’m a little overwhelmed.

I flashed a server and a client module - so far I can turn the Server led on/off. I think I made a mistake somewhere - when using the nrfMesh app I can’t control anything — using the Silicon Labs App I am able to turn the Server led on/off. I guess I’ll redo the tutorial once more tomorrow.

My real problem is this:
I do not quite get how I implement my own functions.

My guess is that I have to set up „models“ and somehow advertise them to the provisioning service. 

I’m kind of stumbling through the dark here. 
If you have any time could you eli5 me what to do or what do look up?

Cheers!

chegewara, 2019-12-03:
Try to assign group address in nRF mesh app. Just remember that group address is starting from 0xc000.

Alvin1Zhang, 2020-07-30:
Thanks for reporting, feel free to reopen with more details. Thanks.

closed 2020-07-30 by Alvin1Zhang