espressif/esp-idf #4436: Install.ps1 fails installing pip requirements.txt

ryanwinter, 2019-12-04:
## Environment

- Development Kit:     none
- Kit version (for WroverKit/PicoKit/DevKitC): NA
- Module or chip used:  ESP32
- IDF version (run ``git describe --tags`` to find it): v4.1-dev-1086-g93a8603c5
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.2.0
- Operating System:     Windows
- Power Supply:         USB

## Problem Description
Running install.ps1 from a powershell, failes to install all the pip requirements
It fails with the error:
`ERROR: Could not install packages due to an EnvironmentError: [Errno 22] Invalid argument: '\\\\${IDF_PATH}\\tools\\kconfig_new\\esp-windows-curses'`

### Expected Behavior
All pip requirements installed correctly

### Actual Behavior
Pip requirements fail to install because of malformed requirements.txt

### Steps to repropduce

1. following the setup guide for the esp-idf dev environment
2. From a powershell, launch `install.ps1`

dobairoland, 2019-12-04:
Thanks for your report. We have already fixed this and will be pushed to Github soon.

For future references, commit 9fa3974447287efa9accea4746a90c62eefe0927 should fix this.

ryanwinter, 2019-12-04:
Thanks dobairoland,

I can't appear to find that commit, is there a PR associated with this?

dobairoland, 2019-12-09:
No, the fix is not yet on Github. You can subscribe to #4341 and you will be notified when it gets published. You can switch back to v4.0 until then which doesn't have this problem.

dobairoland, 2019-12-16:
The fix is in the master branch. Please reopen the issue if it didn't help.

closed 2019-12-16 by dobairoland