espressif/esp-idf #4438: A2DP sink example cannot control volume with iPhone or MacBook

cavonlee, 2019-12-04:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      none
- Kit version: N/A
- Module or chip used:  ESP32-WROOM-32
- IDF version:  v4.1-dev-1086-g93a8603c5
- Build System: CMake
- Compiler version: 8.2.0
- Operating System:  macOS
- Power Supply:  USB

## Problem Description

Hi, I was going to make a bluetooth speaker with my ESP32 WROOM module, with a MAX98357 module, and I tried the a2dp sink example. I can successfully connect it with my iPhone, MacBook, or Android phone. But something weird happened.

When connecting to iPhone or MacBook, and try to change the volume, there print's out `Volume is set by remote controller 37%`, but the volume never changes, except for 0%, there'll be no sound. And the `volume_change_simulation` prints out `Volume is set locally to: `, still never changes the volume, except for 0%. 

Then I connect my Android phone, and changes the volume, there's no ` Volume is set by remote controller 37%` printing, but the actual volume does.

I've searched all over the internet and issue, and gone through all their attempts, but no luck.

I also notice there's an error after connecting to iPhone/MacBook
```
E (4862) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
```
As iPhone/MacBook controls the music with AVRC, I think maybe it's related.

### Expected Behavior

Change volume on iPhone/Mac, the sound from Speaker should changes respectively.

### Actual Behavior

Volume change not working, keeping the same volume

### Steps to repropduce

1. Compile and flash the a2dp sink example
2. Connect bluetooth with iPhone/MacBook
3. Play a music or video
4. Change the volume with iPhone/MacBook

### Code to reproduce this issue

Didn't changes the example

[... 93 lines cut ...]
I (14332) BT_LOG: bta_av_link_role_ok hndl:x41 role:1 conn_audio:x1 bits:1 features:x824b

W (14332) BT_APPL: new conn_srvc id:19, app_id:1
I (14342) BT_AV: A2DP audio state: Started
I (15152) RCTG: Volume is set locally to: 70%
I (15182) RCTG: AVRC register event notification: 13, param: 0x0
I (16702) BT_AV: Audio packet count 100
I (18492) RCTG: AVRC set absolute volume: 67%
I (18492) RCTG: Volume is set by remote controller 67%

I (18502) RCTG: AVRC set absolute volume: 65%
I (18502) RCTG: Volume is set by remote controller 65%

I (18512) RCTG: AVRC set absolute volume: 62%
I (18512) RCTG: Volume is set by remote controller 62%

I (18542) RCTG: AVRC set absolute volume: 60%
I (18542) RCTG: Volume is set by remote controller 60%

I (18562) RCTG: AVRC set absolute volume: 57%
I (18562) RCTG: Volume is set by remote controller 57%

I (18642) RCTG: AVRC set absolute volume: 54%
I (18642) RCTG: Volume is set by remote controller 54%
```

avsheth, 2019-12-05:
Hi @cavonlee What version of MacOS are you using? This issue was seen with earlier versions of MacOS and was found working okay post Catalina.

cavonlee, 2019-12-05:
Hi, @avsheth I'm running macOS Catalina 10.15, and my iPhone 8 is on 13.1.3, should I update my iPhone?

avsheth, 2019-12-05:
can you try updating both?

cavonlee, 2019-12-05:
@avsheth I've just  tried updating both, but no luck. But I found something. On my MacBook Music App there's an individual volume control in APP, works. Change the volume and everything. But, that did not triggers the Volume set remotely debug. Nothing prints out, but still works. While changing the system volume, triggers the debug, but not the actual volume.

I think it might be the problem of the AVRC part. And as I mentioned, every time I connect it, there's an error says `E (12913) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0`. I check the source code, but still don't know how to deal with it. Should I ignore it?

avsheth, 2019-12-06:
Hi @cavonlee 
I will try to reproduce this at my end and let you know.

cavonlee, 2019-12-09:
@avsheth Thank you, and how's everything going?

cavonlee, 2019-12-11:
@avsheth It's been some days. And by the way, I notice that, the simulation volume change didn't work either. The `Volume is set locally to: XX%` prompt repeatedly, but also never changes the volume. I was wondering, is it because of the I2S device I use? the MAX98357? I bought a PCM5102. I will try it out and post the result here.

avsheth, 2019-12-11:
Hi @cavonlee 
Sorry I got occupied with something. I will get back by tomorrow.

cavonlee, 2019-12-12:
@avsheth Hi again, I just tried the PCM5102 just received, and still the same result, not changing the volume.

avsheth, 2019-12-13:
Hi @cavonlee 
Was PCM5102 initialised properly? Along with I2S initialisation, you'll have initialise codec as well (which is specific to the codec).

cavonlee, 2019-12-15:
@avsheth Sorry, I don't think I understand. I'm new with the IDF and I2S, and I don't see any initialization about the PCM5102 in the example, and don't know how to do that. Can you be more specific? or a link I could follow. 
Thank you

avsheth, 2019-12-16:
Hi @cavonlee 
Please check esp-adf example [here](https://github.com/espressif/esp-adf/blob/master/examples/get-started/pipeline_a2dp_sink_and_hfp/main/a2dp_sink_and_hfp_example.c)
You can trace audio_board_init() in this file and see how it initialises lyrat codec (not PCM5102).
You should be able to directly run this app as well on a lyrat board and listen to the audio.

cavonlee, 2019-12-17:
Hi @avsheth 
I trace all the way back audio_board_init(), and locate the initialization of the "es" family chip. And I don't see any initialization description in PCM5102's datasheet like the one in es8388. I'm really confused.
I bought the PCM5102 is because the example says it's built for PCM5102, so if it needs any codec initialization, it should have done it, right? And as it's "codec" is not been initialized, I shouldn't heard any sound, right?
And, another thing I found out, in `bt_app_av.c`, the `volume set by controller` function seems only changing the volume variable but not changing the actual volume. Cuz there seem to be no volume changing statement. Is is something we need to do by ourself? If is, how can i change the volume with software?
```
static void volume_set_by_controller(uint8_t volume)
{
    ESP_LOGI(BT_RC_TG_TAG, "Volume is set by remote controller %d%%\n", (uint32_t)volume * 100 / 0x7f);
    _lock_acquire(&s_volume_lock);
    s_volume = volume;
    _lock_release(&s_volume_lock);
}
```

riccardobrue, 2020-08-27:
I think I've fallen in this same issue, using Samsung smartphones (Galaxy S10) and iOS devices. While using Xiaomi Note 7 it works very well. I am using ESP-IDF 4.0. Any solution found? This is very critical.

PS: I've noticed that in the example there is a "simulation" task which is continuously changing the volume % "locally" (inside the ESP32?) How can I solve such issue? This is very critical to solve for me.

drewandre, 2020-08-27:
@riccardobrue did you disable the simulation task? It's been a while since I experience this issue, but I remember disabling that task. I thought it was pretty odd that the example came with that.

riccardobrue, 2020-08-28:
> @riccardobrue did you disable the simulation task? It's been a while since I experience this issue, but I remember disabling that task. I thought it was pretty odd that the example came with that.

Yes indeed, I disabled such task, so weird that there is such task enabled in a example as that. Still issues with audio +/- buttons with Samsung S10 and iOS devices.

bellatriks, 2020-11-24:
Hi all, 

Would you try the changes above and let me know if it helps with volume control, please?

punj, 2021-05-24:
Hello All!

Is there any solution to control volume from iPhone? Volume control is not working from iPhone yet

bellatriks, 2021-05-25:
> Hello All!
> 
> Is there any solution to control volume from iPhone? Volume control is not working from iPhone yet

Have you tried the https://github.com/espressif/esp-idf/pull/6150  ?

mlynch, 2021-06-16:
I also ran into this issue. Works fine in MacOS but not with my iPhone. In fact, weirdly the a2dp_sink local volume control keeps changing randomly (going up and down).

Anyways, I found this video on youtube where he claims to have fixed this issue himself. Not sure if it's similar to the changes in #6150 but I'm going to give it a try later and see how it works. His sample code is in the description as well.

https://youtu.be/_nCOluMHfGI?t=234

Wth-Esp, 2022-05-22:
Hi, all

Sorry for the late reply, and I think it's necessary to explain why iOS/Mac devices behave differently from Android devices.

First, via Classic Bluetooth, there are three ways to change the volume:
   - **Absolute Volume Control** (not every Bluetooth device support this feature)
   [image]
   - **Relative Volume PASS THROUGH **

Secondly, without the limitation of Classic Bluetooth, the Audio Source Device can also increase or decrease the amplitude of the audio signal to attain the volume change of the audio player or something.

iOS/Mac devices use the first method, and most Android devices use the second method. So, when you connect ESP32 to the iOS/Mac devices, ESP32 shall receive the AVRC control command while not seeing any AVRC callback print when connecting the Android devices.

When you see the AVRC callback print and make sure your device has initialized the i2S to handle the volume value. When you are connecting with an Android device, you may not need to do anything else.

Thanks
Tianhua Wei

Alvin1Zhang, 2022-06-07:
Thanks for reporting, feel free to reopen.

closed 2022-06-07 by Alvin1Zhang

hientv1999, 2023-06-27:
Hi sir, I'm working on PlatformIO and I cannot control the volume from my iPhone.There isn't even any callback being called when controlling the volume from the iPhone. I used Arduino IDE and I don't know how to enable those Verbose / Debug / Info logs. I tried set debug level to Verbose too but no other log rather than Serial print was shown. Can you please let me know what I did wrong?