espressif/esp-idf #4439: CMake target_link_libraries order problems

drony, 2019-12-04:
MySDK->MyLib->nvs_flash-> compilation success and i get libmylib.a from build

MyApp->libMyLib.a -> compilation error undefined reference to nvs_flash_init() etc.

`idf_component_register(SRCS Main.cpp
                    INCLUDE_DIRS REQUIRES nvs_flash MyLib log)

add_library(MyLib STATIC IMPORTED)

set_target_properties(MyLib PROPERTIES IMPORTED_LOCATION  ../../components/CustomStatic_LIB/lib MyLib.a)

target_link_libraries(${COMPONENT_LIB} PUBLIC  MyLib )`

**When I look at the build.ninja file** /build /build.ninja

generated this code but **libMyLib.a** end of the file

`  FLAGS = -mlongcalls -Wno-frame-address -g
  LINK_FLAGS = -nostdlib
  LINK_LIBRARIES = esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/driver/libdriver.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/app_update/libapp_update.a esp-idf/spi_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/esp_eth/libesp_eth.a esp-idf/lwip/liblwip.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_event/libesp_event.a esp-idf/pthread/libpthread.a esp-idf/espcoredump/libespcoredump.a esp-idf/esp32/libesp32.a esp-idf/xtensa/libxtensa.a esp-idf/esp_common/libesp_common.a esp-idf/esp_rom/libesp_rom.a esp-idf/soc/libsoc.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/freertos/libfreertos.a esp-idf/vfs/libvfs.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a esp-idf/app_trace/libapp_trace.a esp-idf/asio/libasio.a esp-idf/coap/libcoap.a esp-idf/console/libconsole.a esp-idf/nghttp/libnghttp.a esp-idf/esp-tls/libesp-tls.a esp-idf/esp_adc_cal/libesp_adc_cal.a esp-idf/esp_gdbstub/libesp_gdbstub.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/esp_https_ota/libesp_https_ota.a esp-idf/protobuf-c/libprotobuf-c.a esp-idf/protocomm/libprotocomm.a esp-idf/mdns/libmdns.a esp-idf/esp_local_ctrl/libesp_local_ctrl.a esp-idf/esp_websocket_client/libesp_websocket_client.a esp-idf/expat/libexpat.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/sdmmc/libsdmmc.a esp-idf/fatfs/libfatfs.a esp-idf/freemodbus/libfreemodbus.a esp-idf/jsmn/libjsmn.a esp-idf/json/libjson.a esp-idf/libsodium/liblibsodium.a esp-idf/mqtt/libmqtt.a esp-idf/openssl/libopenssl.a esp-idf/spiffs/libspiffs.a esp-idf/ulp/libulp.a esp-idf/unity/libunity.a esp-idf/wifi_provisioning/libwifi_provisioning.a esp-idf/CompanyName/libCompanyName.a esp-idf/main/libmain.a -Wl,--cref -Wl,--Map=C:/Users/xxxxx/source/embed/CompanyName.Embedded/Demo/EDKDemo/VisualGDB/Debug/EDKDemo.map esp-idf/esp_http_client/libesp_http_client.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/sdmmc/libsdmmc.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp-tls/libesp-tls.a esp-idf/protocomm/libprotocomm.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/nghttp/libnghttp.a esp-idf/protobuf-c/libprotobuf-c.a esp-idf/mdns/libmdns.a esp-idf/console/libconsole.a esp-idf/json/libjson.a esp-idf/CompanyName/libCompanyName.a esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/driver/libdriver.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/app_update/libapp_update.a esp-idf/spi_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/esp_eth/libesp_eth.a esp-idf/lwip/liblwip.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_event/libesp_event.a esp-idf/pthread/libpthread.a esp-idf/espcoredump/libespcoredump.a esp-idf/esp32/libesp32.a esp-idf/xtensa/libxtensa.a esp-idf/esp_common/libesp_common.a esp-idf/esp_rom/libesp_rom.a esp-idf/soc/libsoc.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/freertos/libfreertos.a esp-idf/vfs/libvfs.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a esp-idf/app_trace/libapp_trace.a esp-idf/mbedtls/mbedtls/library/libmbedtls.a esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a esp-idf/mbedtls/mbedtls/library/libmbedx509.a C:/SysGCC/esp32/esp-idf/v4.0/components/esp_wifi/lib_esp32/libcoexist.a C:/SysGCC/esp32/esp-idf/v4.0/components/esp_wifi/lib_esp32/libcore.a C:/SysGCC/esp32/esp-idf/v4.0/components/esp_wifi/lib_esp32/libespnow.a C:/SysGCC/esp32/esp-idf/v4.0/components/esp_wifi/lib_esp32/libmesh.a C:/SysGCC/esp32/esp-idf/v4.0/components/esp_wifi/lib_esp32/libnet80211.a C:/SysGCC/esp32/esp-idf/v4.0/components/esp_wifi/lib_esp32/libphy.a C:/SysGCC/esp32/esp-idf/v4.0/components/esp_wifi/lib_esp32/libpp.a C:/SysGCC/esp32/esp-idf/v4.0/components/esp_wifi/lib_esp32/librtc.a C:/SysGCC/esp32/esp-idf/v4.0/components/esp_wifi/lib_esp32/libsmartconfig.a esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/driver/libdriver.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/app_update/libapp_update.a esp-idf/spi_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/esp_eth/libesp_eth.a esp-idf/lwip/liblwip.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_event/libesp_event.a esp-idf/pthread/libpthread.a esp-idf/espcoredump/libespcoredump.a esp-idf/esp32/libesp32.a esp-idf/xtensa/libxtensa.a esp-idf/esp_common/libesp_common.a esp-idf/esp_rom/libesp_rom.a esp-idf/soc/libsoc.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/freertos/libfreertos.a esp-idf/vfs/libvfs.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a esp-idf/app_trace/libapp_trace.a esp-idf/mbedtls/mbedtls/library/libmbedtls.a esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a esp-idf/mbedtls/mbedtls/library/libmbedx509.a C:/SysGCC/esp32/esp-idf/v4.0/components/esp_wifi/lib_esp32/libcoexist.a C:/SysGCC/esp32/esp-idf/v4.0/components/esp_wifi/lib_esp32/libcore.a C:/SysGCC/esp32/esp-idf/v4.0/components/esp_wifi/lib_esp32/libespnow.a C:/SysGCC/esp32/esp-idf/v4.0/components/esp_wifi/lib_esp32/libmesh.a C:/SysGCC/esp32/esp-idf/v4.0/components/esp_wifi/lib_esp32/libnet80211.a C:/SysGCC/esp32/esp-idf/v4.0/components/esp_wifi/lib_esp32/libphy.a C:/SysGCC/esp32/esp-idf/v4.0/components/esp_wifi/lib_esp32/libpp.a C:/SysGCC/esp32/esp-idf/v4.0/components/esp_wifi/lib_esp32/librtc.a C:/SysGCC/esp32/esp-idf/v4.0/components/esp_wifi/lib_esp32/libsmartconfig.a -u esp_app_desc -L C:/SysGCC/esp32/esp-idf/v4.0/components/esp_wifi/lib_esp32 -u pthread_include_pthread_impl -u pthread_include_pthread_cond_impl -u pthread_include_pthread_local_storage_impl -L C:/Users/xxxxx/source/embed/CompanyName.Embedded/Demo/EDKDemo/VisualGDB/Debug/esp-idf/esp32 -T esp32_out.ld -u app_main -L C:/Users/xxxxx/source/embed/CompanyName.Embedded/Demo/EDKDemo/VisualGDB/Debug/esp-idf/esp32/ld -T esp32.project.ld -L C:/SysGCC/esp32/esp-idf/v4.0/components/esp32/ld -T esp32.peripherals.ld -u call_user_start_cpu0 -u ld_include_panic_highint_hdl C:/SysGCC/esp32/esp-idf/v4.0/components/xtensa/esp32/libhal.a -Wl,--gc-sections -L C:/SysGCC/esp32/esp-idf/v4.0/components/esp_rom/esp32/ld -T esp32.rom.ld -T esp32.rom.libgcc.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-data.ld -T esp32.rom.newlib-funcs.ld -Wl,--undefined=uxTopUsedPriority -u vfs_include_syscalls_impl esp-idf/newlib/libnewlib.a -u newlib_include_locks_impl -u newlib_include_heap_impl -u newlib_include_syscalls_impl -u newlib_include_pthread_impl -u __cxa_guard_dummy -lstdc++ esp-idf/pthread/libpthread.a -u __cxx_fatal_exception -fno-rtti -lgcov -lc -lm -lgcc **../../components/CompanyName/libMyLib.a**
  OBJECT_DIR = CMakeFiles\EDKDemo.elf.dir
  POST_BUILD = cd .
  PRE_LINK = cd .
  TARGET_COMPILE_PDB = CMakeFiles\EDKDemo.elf.dir\
  TARGET_FILE = EDKDemo.elf
  TARGET_PDB = EDKDemo.elf.pdb
  RSP_FILE = CMakeFiles\EDKDemo.elf.rsp`

but when I modified build.ninja file and myLib.a after libnvs_flash.a build and works

here is the changed file

`  FLAGS = -mlongcalls -Wno-frame-address -g
  LINK_FLAGS = -nostdlib
  LINK_LIBRARIES = esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/driver/libdriver.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/app_update/libapp_update.a esp-idf/spi_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/esp_eth/libesp_eth.a esp-idf/lwip/liblwip.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_event/libesp_event.a esp-idf/pthread/libpthread.a esp-idf/espcoredump/libespcoredump.a esp-idf/esp32/libesp32.a esp-idf/xtensa/libxtensa.a esp-idf/esp_common/libesp_common.a esp-idf/esp_rom/libe [... cut ...] _include_heap_impl -u newlib_include_syscalls_impl -u newlib_include_pthread_impl -u __cxa_guard_dummy -lstdc++ esp-idf/pthread/libpthread.a -u __cxx_fatal_exception -fno-rtti -lgcov -lc -lm -lgcc 
  OBJECT_DIR = CMakeFiles\EDKDemo.elf.dir
  POST_BUILD = cd .
  PRE_LINK = cd .
  TARGET_COMPILE_PDB = CMakeFiles\EDKDemo.elf.dir\
  TARGET_FILE = EDKDemo.elf
  TARGET_PDB = EDKDemo.elf.pdb
  RSP_FILE = CMakeFiles\EDKDemo.elf.rsp`

and the question how to insert libMylib.a to this using CMakeLists.txt

renzbagaporo, 2019-12-08:
Have you tried to establish dependency of the prebuilt library on the component:

```
set_property(TARGET MyLib APPEND INTERFACE_LINK_LIBRARIES idf::nvs_flash)
```

`idf::nvs_flash` is the library target created for component `nvs_flash`, since behind the scenes, IDF components are just CMake libraries created with `add_library`.

drony, 2019-12-08:
> Have you tried to establish dependency of the prebuilt library on the component:
> 
> ```
> set_property(TARGET MyLib APPEND PROPERTY INTERFACE_LINK_LIBRARIES idf::nvs_flash)
> ```
> 
> `idf::nvs_flash` is the library target created for component `nvs_flash`, since behind the scenes, IDF components are just CMake libraries created with `add_library`.

thanks a lot yesterday handle the problem with same way:)

closed 2019-12-08 by drony