espressif/esp-idf #4441: esp32-WROOM-32u flash read err, 1000

deedawg, 2019-12-04:
I have an esp32-WROOM-32u 16MB module reports `flash read err,1000` when the module is booted. The situation I have encountered reads to be slightly different than similar issues reported by other users. 

Other than the `flash read err, 1000` the module appears to be functioning correctly in that it can be successfully flashed with new firmware and it successfully runs that firmware, as far as I can determine based on the outputs in the serial monitor.  The error hasn’t caused any problems so far with the performance of the esp module so I have ignored it. However, I’m bothered as to why the error may be happening an I’d like to understand where it’s coming from. 

The esp module is being programmed using a programmer similar to this device. 
https://www.cnx-software.com/2017/09/18/needs-to-program-many-esp-wroom-32-esp-32s-modules-this-board-should-help/

As far as I can determine the programming module is correctly setting the strapping pins on the esp32 module.  There are no devices connected to any of the GPIO pins when the module is started, so nothing should be interfering with the connection to the integrated flash chip or GPIO12. I have used `esptool.py erase_flash` and then reflashed the esp32 with new firmware and this did not resolve `flash read err,1000` being reported upon restarting the esp module. 

I have 20 (never used) esp32-WROOM-32 modules that are demonstrating the same behaviour. It strikes me as unlikely that all of them would have a bad flash chips, which could explain the `flash read err,1000`. 

I’m not sure what else I can look at to resolve this. I am looking for some ideas.......Solutions to this issue i have read in other threads don’t seem to relevant in this set up. 

Outputs from `esptool.py` don’t show anything unusual, to my eye:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57
```

```
C:\ESP-IDF\tools>esptool.py flash_id
esptool.py v2.8
Found 1 serial ports
Serial port COM7
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: bc:dd:c2:ca:bd:54
Uploading stub...
Running stub...
Stub running...
Manufacturer: c8
Device: 4018
Detected flash size: 16MB
Hard resetting via RTS pin...
```

```
C:\ESP-IDF\tools>espefuse.py --port COM7 summary
espefuse.py v2.8
Connecting....
EFUSE_NAME             Description = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Config fuses:
XPD_SDIO_FORCE         Ignore MTDI pin (GPIO12) for VDD_SDIO on reset    = 0 R/W (0x0)
XPD_SDIO_REG           If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset   = 0 R/W (0x0)
XPD_SDIO_TIEH          If XPD_SDIO_FORCE & XPD_SDIO_REG, 1=3.3V 0=1.8V   = 0 R/W (0x0)
CLK8M_FREQ             8MHz clock freq override                          = 54 R/W (0x36)
SPI_PAD_CONFIG_CLK     Override SD_CLK pad (GPIO6/SPICLK)                = 0 R/W (0x0)
SPI_PAD_CONFIG_Q       Override SD_DATA_0 pad (GPIO7/SPIQ)               = 0 R/W (0x0)
SPI_PAD_CONFIG_D       Override SD_DATA_1 pad (GPIO8/SPID)               = 0 R/W (0x0)
SPI_PAD_CONFIG_HD      Override SD_DATA_2 pad (GPIO9/SPIHD)              = 0 R/W (0x0)
SPI_PAD_CONFIG_CS0     Override SD_CMD pad (GPIO11/SPICS0)               = 0 R/W (0x0)
DISABLE_SDIO_HOST      Disable SDIO host                                 = 0 R/W (0x0)
 
Efuse fuses:
WR_DIS                 Efuse write disable mask                          = 0 R/W (0x0)
RD_DIS                 Efuse read disablemask                            = 0 R/W (0x0)
CODING_SCHEME          Efuse variable block length scheme                = 0 R/W (0x0)
KEY_STATUS             Usage of efuse block 3 (reserved)                 = 0 R/W (0x0)
 
Security fuses:
FLASH_CRYPT_CNT        Flash encryption mode counter                     = 0 R/W (0x0)
FLASH_CRYPT_CONFIG     Flash encryption config (key tweak bits)          = 0 R/W (0x0)
CONSOLE_DEBUG_DISABLE  Disable ROM BASIC interpreter fallback            = 1 R/W (0x1)
ABS_DONE_0             secure boot enabled for bootloader                = 0 R/W (0x0)
ABS_DONE_1             secure boot abstract 1 locked                     = 0 R/W (0x0)
JTAG_DISABLE           Disable JTAG                                      = 0 R/W (0x0)
DISABLE_DL_ENCRYPT     Disable flash encryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_DECRYPT     Disable flash decryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_CACHE       Disable flash cache in UART bootloader            = 0 R/W (0x0)
BLK1                   Flash encryption key
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLK2                   Secure boot key
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLK3                   Variable Block 3
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
 
Identity fuses:
MAC                    Factory MAC Address
  = bc:dd:c2:ca:bd:54 (CRC 0xfe OK) R/W
CHIP_VER_REV1          Silicon Revision 1                                = 1 R/W (0x1)
CHIP_VER_REV2          Silicon Revision 2                                = 0 R/W (0x0)
CHIP_VERSION           Reserved for future chip versions                 = 2 R/W (0x2)
CHIP_PACKAGE           Chip package identifier                           = 1 R/W (0x1)
 
Calibration fuses:
BLK3_PART_RESERVE      BLOCK3 partially served for ADC calibration data  = 0 R/W (0x0)
ADC_VREF               Voltage reference calibration                     = 1121 R/W (0x3)
 
Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).
```

igrr, 2019-12-04:
This seems to be the issue described in section 3.8 of the [ESP32 errata](https://espressif.com/sites/default/files/documentation/eco_and_workarounds_for_bugs_in_esp32_en.pdf). It is related to the slow startup time of the SPI flash chip. Indeed, other than this message and slightly longer bootup time, it has no other effects.

deedawg, 2019-12-04:
> This seems to be the issue described in section 3.8 of the [ESP32 errata](https://espressif.com/sites/default/files/documentation/eco_and_workarounds_for_bugs_in_esp32_en.pdf). It is related to the [...]

I recall that being speculated as cause that was noted in the threads on this issue. So there is a remedy for this?  

I saw a recommendation to add a _delay_ to the firmware code but it wasn’t entirely clear to me what that meant (perhaps I misunderstood the recommendation, however).

igrr, 2019-12-04:
When this issue occurs on a power-on reset, there is no impact other than increased start up time, the error message, and incorrect reset reason reported by `esp_reset_reason` function. The only solution for this is to replace the flash chip, which may be impractical if you are using a module.

This issue also occurs after a deep sleep reset. However IDF already includes the workaround for this, delaying for some number of microseconds in deep sleep reset stub, before starting to boot from flash. Since the workaround is in place, you don't need to add any extra delay in your code.

Alvin1Zhang, 2020-05-27:
@deedawg Thanks for reporting. Would you please help share if any updates for the issue? Thanks.

mahavirj, 2021-07-21:
Closing this issue, please feel free to re-open if further explanation is required on this topic.

closed 2021-07-21 by mahavirj

varioscientist, 2023-01-14:
Im had same issue - some projects within wifi-bt functions fails with flash read error (but another started correctly). This cause solved by more accurate power supply to devboard via another microUSB  cable. 
While  checked power lines via oscilloscope i've saw abnormal power drop (and power LED had blinks a little).
This case was happend after some ESD stress  to ESP module within some tochpad function was fails .