espressif/esp-idf #4443: Just In Time Provisioning Issue (JITP) with AWS

monkeytronics, 2019-12-04:
Hey, 
I was using JITP with the Amazon FreeRTOS, but want to use ESP-IDF instead. It's just a much cleaner repo. 

So, I've fired up the Jump Start design. JITP was working fine with afr. The certificates requried are:

- keyCLIENT_PRIVATE_KEY_PEM = $DEVICE.key
- keyCLIENT_CERTIFICATE_PEM = $DEVICE.crt
- keyJITR_DEVICE_CERTIFICATE_AUTHORITY_PEM = $CA.ca.pem

Note that the CA.ca.pem file is **not** the Amazon authority server CA file that you download from Amazon. 

I've tried to replicate this in the Jumpstart project, but it fails with :

```
E (3222) aws_iot: failed! mbedtls_ssl_handshake returned -0x2700
E (3222) aws_iot:     Unable to verify the server's certificate.
E (3232) cloud: Error(-4) connecting to xxxxxxxx-ats.iot.ap-southeast-2.amazonaws.com:8883
```

I was really expecting this to work... What certificates are expected? Why is this different than when using amazon freeRTOS?

shahpiyushv, 2019-12-05:
@monkeytronics , I haven't used AFR myself, but for the esp-aws-iot, the CA certificate needs to be appended to the device certificate for JITP/JITR.

Eg.
$ cat $DEVICE.crt $CA.ca.pem > $DEVICE.$CA.crt

Then, use this combo certificate as the device certificate. Let me know if it works.

monkeytronics, 2019-12-05:
Hi @shahpiyushv. 

I've just tried: 

**device.key** is the private device key as before.
**device.cert** is now cat **device.cert && monkey.CA.pem**
And aws_root_ca_pem_start now points to AmazonRootCA1.pem

but I get the same error. Have I got the certificates right here?

closed 2019-12-05 by monkeytronics