espressif/esp-idf #4446: MCPWM: simultaneous rising/falling edge capture missing

paccerdk, 2019-12-05:
Currently, the MCPWM only supports capture on NEG / POS edge, but not on both/on change.

I've seen mention that this feature was being implemented, _posted by @koobest in issue https://github.com/espressif/esp-idf/issues/1907#issuecomment-436521941_ (the issue is also better explained and discussed here)
> "This feature will soon be merged into master."

I haven't been able to find any reference to this, anywhere else than this specific comment.

This seems like a rather essential functionality (e.g. measuring length of single pulses), and it seems its already been looked into implementing it (but then halted/forgotten?).

I was hoping someone could perhaps shed some light on what's going on here. Thanks!

wilhelmw201, 2020-07-22:
Hi,

`MCPWM_NEG_EDGE|MCPWM_POS_EDGE` will work :)

wilhelmw201, 2020-07-22:
Hi,

`MCPWM_POS_EDGE|MCPWM_NEG_EDGE` will work.

closed 2020-08-20 by projectgus