espressif/esp-idf #4448: Feature request: Allow runtime config mqtt protocol version

AxelLin, 2019-12-06:
The ESP-MQTT supports MQTT protocol 3.1 and MQTT protocol 3.1.1.
But it is decided at compile time by selecting MQTT_PROTOCOL_311 or not.
This is a mqtt *clinet* implemenation it should allow config using 3.1 or 3.1.1
at runtime because it's not possible to decide if the target mqtt broker support 3.1.1 or not.

Suggest providing API to allow config mqtt protocol version at runtime.

AxelLin, 2019-12-11:
The esp-mqtt is in another git tree (https://github.com/espressif/esp-mqtt).
So I'm wondering if I should move this issue from esp-idf to esp-mqtt or
just keep it here? comments?

david-cermak, 2019-12-17:
@AxelLin You can leave the issue here and it will be processed as IDF issue, so closed once the IDF references the esp-mqtt submodule supporting this feature.

I think it is a good idea to support runtime selection of mqtt protocol. Thank you for this  suggestion!

AxelLin, 2020-02-11:
Thank you for adding this feature.

closed 2020-02-11 by AxelLin