espressif/esp-idf #4449: ESP32最新SDK遇到了突然死机重启的问题

wangqianjiao, 2019-12-06:
## 1. 开发环境

项目应用场景介绍：空气净化器家电配网连云
ESP 芯片 ESP 模组名称：ESP32-WROVER-IB
硬件是否定制,PSRAM 大小：PSRAM 8M flash 也是8M
esP-idf 版本号：
```
sdk4.0到 commit ID 99fb9a3f7c28c5fa12b1bd4aa6fb7b622d841326
```

## 2. 问题描述

您好，
我们使用ESP32的最新适配了阿里云3.0.1的源码去调试设备联网时，遇到了突然死机的问题，没有什么征兆就是连接阿里云后机器放着运行了2天，就突然突然报了一个错误死机了，sdk更新到了sdk4.0到
```
commit ID 99fb9a3f7c28c5fa12b1bd4aa6fb7b622d841326
```
其他代码没有做任何变更
 

### 2.1 复现步骤

1. 配网完成；
2. 查看串口日志，观察到IP地址已获取成功，设备在线；
3. 放着不动等待中
4. 死机重启

### 2.2 复现问题的代码

无

## 3. 调试 Logs

assertion "sock->fd_used > 0" failed: file "/home/dong/airx_work/esp32_sdk/esp-idf_v4.0/components/lwip/lwip/src/api/sockets.c", line 389, function: done_socket
abort() was called at PC 0x40110bf3 on core 1

ELF file SHA256: d6db9525955b34aec907387dab070f32a3329f5b2cbf28a2f3c6274059beb7b2

Backtrace: 0x40091f78:0x3ffb85a0 0x40092385:0x3ffb85c0 0x40110bf3:0x3ffb85e0 0x401522b6:0x3ffb8610 0x40153d8d:0x3ffb8630 0x40129fdd:0x3ffb86d0 0x4017cb11:0x3ffb8730 0x4013f3e1:0x3ffb8750 0x4013ef72:0x3ffb8780 0x4013c636:0x3ffb87a0 0x4013cba5:0x3ffb87f0 0x4013de05:0x3ffb8870 0x40137f83:0x3ffb88b0 0x4010db85:0x3ffb88d0 0x4010dc6d:0x3ffb89e0

Rebooting...

## 4. 其他项
sdkconfig内容
```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_SDK_PYTHON="python"
CONFIG_SDK_MAKE_WARN_UNDEFINED_VARIABLES=y
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
CONFIG_BOOTLOADER_LOG_LEVEL_ERROR=y
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_INFO is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=1
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
[... 1105 lines cut ...]
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_IP_LOST_TIMER_INTERVAL=120
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
# End of deprecated options
```

liuzfesp, 2019-12-06:
Call @freakyxue to debug this issue.

freakyxue, 2019-12-06:
@wangqianjiao  Please provide the ELF file, thanks

wangqianjiao, 2019-12-06:
> @wangqianjiao Please provide the ELF file, thanks

您好，ELF发邮箱可以吗

freakyxue, 2019-12-06:
@wangqianjiao xueyunfei@espressif.com,elf, 文件 需要是发生crash 时对应的elf 文件 .谢谢

wangqianjiao, 2019-12-06:
> @wangqianjiao [xueyunfei@espressif.com](mailto:xueyunfei@espressif.com),elf, 文件 需要是发生crash 时对应的elf 文件 .谢谢

好的，是对应的elf文件，已发邮箱

wangqianjiao, 2019-12-06:
@freakyxue 您好，刚15:20分又发生了一次，一模一样死机重启，调试日志已更新

wangqianjiao, 2019-12-06:
> @freakyxue 您好，刚15:20分又发生了一次，一模一样死机重启，调试日志已更新

freakyxue, 2019-12-06:
@wangqianjiao 请将这次elf文件发邮箱 刚才elf 文件好像不对,没定位到问题

wangqianjiao, 2019-12-06:
> @wangqianjiao 请将这次elf文件发邮箱 刚才elf 文件好像不对,没定位到问题

好的，已发送

closed 2019-12-20 by wangqianjiao

dedvalson, 2020-04-14:
Can you tell me how you fixed this. I am having the same issue in my project.

freakyxue, 2020-04-15:
hi @dedvalson 
Can you provide a demo for me to reproduce this problem

dedvalson, 2020-04-15:
Hi.

Unfortunately it happens only every few days in a complex program. However, I think you might have fixed this in commit dd3b301750f19c9f15248ac32d366cd41c6b3225.

I can tell you that it happens in the select call in the following function:

`int connect_wait(int sockno, struct sockaddr *addr, size_t addrlen, struct timeval *timeout) {
    int res, opt;
    // get socket flags
    if ((opt = fcntl(sockno, F_GETFL, NULL)) < 0) {
        ESP_LOGE(LOG_TAG, "connect_wait: fcntl 1 failed, opt = %d", opt);
        return -1;
    }
    // set socket non-blocking
    if (fcntl(sockno, F_SETFL, opt | O_NONBLOCK) < 0) {
        ESP_LOGE(LOG_TAG, "connect_wait: fcntl 2 failed");
        return -1;
    }
    // try to connect
    if ((res = connect(sockno, addr, addrlen)) < 0) {
        if (errno == EINPROGRESS) {
            fd_set wait_set;
            // make file descriptor set with socket
            FD_ZERO(&wait_set);
            FD_SET(sockno, &wait_set);
            // wait for socket to be writable; return after given timeout
            res = select(sockno + 1, NULL, &wait_set, NULL, timeout);
        }
    }
    // connection was successful immediately
    else {
        res = 1;
    }

    // reset socket flags
    if (fcntl(sockno, F_SETFL, opt) < 0) {
        ESP_LOGE(LOG_TAG, "connect_wait: fcntl 3 failed");
        return -1;
    }

    // an error occured in connect or select
    if (res < 0) {
        ESP_LOGE(LOG_TAG, "connect_wait: failure res = %d", res);
        return -1;
    }
    // select timed out
    else if (res == 0) {
        ESP_LOGE(LOG_TAG, "connect_wait: timed out");
        errno = ETIMEDOUT;
        return 1;
    }
    // almost finished...
    else {
        socklen_t len = sizeof(opt);
        // check for errors in socket layer
        if (getsockopt(sockno, SOL_SOCKET, SO_ERROR, &opt, &len) < 0) {
            ESP_LOGE(LOG_TAG, "connect_wait: getsockopt failed");
            return -1;
        }
        // there was an error
        if (opt) {
            ESP_LOGE(LOG_TAG, "connect_wait: getsockopt failed opt = %d", opt);
            errno = opt;
            return -1;
        }
    }
    return 0;
}`

freakyxue, 2020-04-15:
hi @dedvalson 
Can I provide a demo of recurrence problem? I want to make sure that it's more convenient for me to check the recurrence problem. If I can't use this interface in case of recurrence problem, it's a waste of time

dedvalson, 2020-04-15:
I understand there isn't much you can do. 

I will keep working on it. 

I was hoping you might be able to tell me how this was resolved  back in December when it came up before.

freakyxue, 2020-04-15:
HI @dedvalson
I'm trying to reproduce this problem here. As long as the problem is repeated, it will be solved soon. Please provide more information to help me reproduce this bug. Thank you

dedvalson, 2020-04-15:
I am running ESP-IDF from release/v4.0 as of Feb 18, 2020. There have been a few checkins since then so I am going to pull down the current release/v4.0 and see if it helps. 

I have attached my SDKConfig file.

I am trying to reduce to a smaller app that can show the problem. 

The current app is polling WiFi outlets. There are multiple threads involved but each socket is only accessed from within a single thread. There are other threads opening other sockets to other devices. Each poll consists of Creating a new socket, connecting using the code above, sending one packet, receiving one packet and closing and destroying the socket. This happens thousands of times before a failure.  A socket is used only once for this process but multiple threads can have polls in progress.

I am also attempting to obtains a stack trace of the failure but so far it only fails when no trace is available.

If there is anything more I can provide just let me know.

Don

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/4481111/sdkconfig.zip)

Alvin1Zhang, 2020-04-16:
@dedvalson Thanks for keeping sharing the details, we will check, thanks.

freakyxue, 2020-04-28:
hi @dedvalson 
I think you can test on the latest LwIP branch. This crash problem may have been fixed. I look forward to your feedback. Thank you

chewhs00, 2020-09-01:
I have the same problem occurred once with esp-idf v4.0.1, but couldn't duplicate it. Need help on this.
[crash08232020.txt](https://github.com/espressif/esp-idf/files/5153915/crash08232020.txt)