espressif/esp-idf #4450: esp_vfs_spiffs_register gives an error if partition_labe is specified

nopnop2002, 2019-12-06:
My environment:

	freeRTOS version:V8.2.0
	NEWLIB version:3.0.0
	lwIP version:2-1-3-0
	ESP-IDF version:v4.1-dev-369-g4dac7c7df-dirty

I tried esp-idf/examples/storage/spiffs.
It work fine.

	I (71) boot: Chip Revision: 1
	I (72) boot_comm: mismatch chip revision, expect 1, found 0
	I (38) boot: ESP-IDF v4.1-dev-369-g4dac7c7df-dirty 2nd stage bootloader
	I (38) boot: compile time 13:04:57
	I (39) boot: Enabling RNG early entropy source...
	I (45) boot: SPI Speed      : 40MHz
	I (49) boot: SPI Mode       : DIO
	I (53) boot: SPI Flash Size : 4MB
	I (57) boot: Partition Table:
	I (60) boot: ## Label            Usage          Type ST Offset   Length
	I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
	I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
	I (83) boot:  2 factory          factory app      00 00 00010000 00100000
	I (90) boot:  3 storage          Unknown data     01 82 00110000 000f0000
	I (98) boot: End of partition table
	I (102) boot_comm: mismatch chip revision, expect 1, found 0
	I (108) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x06a74 ( 27252) map
	I (127) esp_image: segment 1: paddr=0x00016a9c vaddr=0x3ffb0000 size=0x02094 (  8340) load
	I (130) esp_image: segment 2: paddr=0x00018b38 vaddr=0x40080000 size=0x00400 (  1024) load
	0x40080000: _WindowOverflow4 at /home/robotics/esp-idf/components/freertos/xtensa_vectors.S:1778

	I (135) esp_image: segment 3: paddr=0x00018f40 vaddr=0x40080400 size=0x070d0 ( 28880) load
	I (156) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x1835c ( 99164) map
	0x400d0018: _stext at ??:?

	I (192) esp_image: segment 5: paddr=0x0003837c vaddr=0x400874d0 size=0x026b4 (  9908) load
	0x400874d0: xQueueGenericReset at /home/robotics/esp-idf/components/freertos/queue.c:272

	I (202) boot: Loaded app from partition at offset 0x10000
	I (202) boot: Disabling RNG early entropy source...
	I (203) cpu_start: Pro cpu up.
	I (207) cpu_start: Application information:
	I (212) cpu_start: Project name:     spiffs
	I (217) cpu_start: App version:      v4.1-dev-369-g4dac7c7df-dirty
	I (223) cpu_start: Compile time:     Dec  6 2019 13:05:01
	I (230) cpu_start: ELF file SHA256:  360f0f87545214dd...
	I (236) cpu_start: ESP-IDF:          v4.1-dev-369-g4dac7c7df-dirty
	I (242) cpu_start: Starting app cpu, entry point is 0x40080fdc
	0x40080fdc: call_start_cpu1 at /home/robotics/esp-idf/components/esp32/cpu_start.c:276

	I (0) cpu_start: App cpu up.
	I (253) heap_init: Initializing. RAM available for dynamic allocation:
	I (260) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
	I (266) heap_init: At 3FFB3088 len 0002CF78 (179 KiB): DRAM
	I (272) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
	I (278) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
	I (285) heap_init: At 40089B84 len 0001647C (89 KiB): IRAM
	I (291) cpu_start: Pro cpu start user code
	I (309) spi_flash: detected chip: generic
	I (310) spi_flash: flash io: dio
	I (310) cpu_start: Starting scheduler on PRO CPU.
	I (0) cpu_start: Starting scheduler on APP CPU.
	I (318) example: Initializing SPIFFS
	I (408) example: Partition size: total: 896321, used: 502
	I (408) example: Opening file
	I (508) example: File written
	I (508) example: Renaming file
	I (548) example: Reading file
	I (548) example: Read from file: 'Hello World!'
	I (548) example: SPIFFS unmounted

So i changed there:

    esp_vfs_spiffs_conf_t conf = {
      .base_path = "/spiffs",
      //.partition_label = NULL,
      .partition_label = "storage",
      .max_files = 5,
      .format_if_mount_failed = true
    };

    // Use settings defined above to initialize and mount SPIFFS filesystem.
    // Note: esp_vfs_spiffs_register is an all-in-one convenience function.
    esp_err_t ret = esp_vfs_spiffs_register(&conf);

esp_vfs_spiffs_register gives an error if partition_labe is specified

	I (71) boot: Chip Revision: 1
	I (72) boot_comm: mismatch chip revision, expect 1, found 0
	I (38) boot: ESP-IDF v4.1-dev-369-g4dac7c7df-dirty 2nd stage bootloader
	I (38) boot: compile time 13:04:57
	I (39) boot: Enabling RNG early entropy source...
	I (44) boot: SPI Speed      : 40MHz
	I (49) boot: SPI Mode       : DIO
	I (53) boot: SPI Flash Size : 4MB
	I (57) boot: Partition Table:
	I (60) boot: ## Label            Usage          Type ST Offset   Length
	I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
	I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
	I (82) boot:  2 factory          factory app      00 00 00010000 00100000
	I (90) boot:  3 storage          Unknown data     01 82 00110000 000f0000
	I (97) boot: End of partition table
	I (102) boot_comm: mismatch chip revision, expect 1, found 0
	I (108) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x06a7c ( 27260) map
	I (127) esp_image: segment 1: paddr=0x00016aa4 vaddr=0x3ffb0000 size=0x02094 (  8340) load
	I (130) esp_image: segment 2: paddr=0x00018b40 vaddr=0x40080000 size=0x00400 (  1024) load
	0x40080000: _WindowOverflow4 at /home/robotics/esp-idf/components/freertos/xtensa_vectors.S:1778

	I (135) esp_image: segment 3: paddr=0x00018f48 vaddr=0x40080400 size=0x070c8 ( 28872) load
	I (156) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x1835c ( 99164) map
	0x400d0018: _stext at ??:?

	I (192) esp_image: segment 5: paddr=0x0003837c vaddr=0x400874c8 size=0x026bc (  9916) load
	0x400874c8: xQueueGenericReset at /home/robotics/esp-idf/components/freertos/queue.c:268 (discriminator 2)

	I (202) boot: Loaded app from partition at offset 0x10000
	I (202) boot: Disabling RNG early entropy source...
	I (203) cpu_start: Pro cpu up.
	I (207) cpu_start: Application information:
	I (212) cpu_start: Project name:     spiffs
	I (217) cpu_start: App version:      v4.1-dev-369-g4dac7c7df-dirty
	I (223) cpu_start: Compile time:     Dec  6 2019 13:05:01
	I (230) cpu_start: ELF file SHA256:  6b6fe262b814aa02...
	I (235) cpu_start: ESP-IDF:          v4.1-dev-369-g4dac7c7df-dirty
	I (242) cpu_start: Starting app cpu, entry point is 0x40080fdc
	0x40080fdc: call_start_cpu1 at /home/robotics/esp-idf/components/esp32/cpu_start.c:276

	I (0) cpu_start: App cpu up.
	I (253) heap_init: Initializing. RAM available for dynamic allocation:
	I (260) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
	I (266) heap_init: At 3FFB3088 len 0002CF78 (179 KiB): DRAM
	I (272) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
	I (278) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
	I (285) heap_init: At 40089B84 len 0001647C (89 KiB): IRAM
	I (291) cpu_start: Pro cpu start user code
	I (309) spi_flash: detected chip: generic
	I (310) spi_flash: flash io: dio
	I (310) cpu_start: Starting scheduler on PRO CPU.
	I (0) cpu_start: Starting scheduler on APP CPU.
	I (318) example: Initializing SPIFFS
	E (408) example: Failed to get SPIFFS partition information (ESP_ERR_INVALID_STATE)
	I (408) example: Opening file
	I (508) example: File written
	I (508) example: Renaming file
	I (548) example: Reading file
	I (548) example: Read from file: 'Hello World!'
	I (548) example: SPIFFS unmounted

igrr, 2019-12-23:
@nopnop2002 Thanks for reporting this!

This seems to be a bug in the example; it should pass the same partition label as used in `esp_vfs_spiffs_register` to `esp_spiffs_info` and `esp_vfs_spiffs_unregister`, i.e.:

```c++
ret = esp_spiffs_info(conf.partition_label, &total, &used);
```
and
```c++
esp_vfs_spiffs_unregister(conf.partition_label);
```

This will be fixed soon (together with added clarification in the docs), in the meantime please make these changes locally.

nopnop2002, 2019-12-24:
Hello Iggr.
OK.
I will change locally.
Thank you for support.

closed 2020-01-06 by espressif-bot