espressif/esp-idf #4451: Can we use GATTS and GATTC at same time in ESP32.

ROYMURLI, 2019-12-06:
ESP_IDF = v3.3-beta1-136-g97eecfa1b
OS- windows (Msys32)

I want to know can we use GATTS and GATTC at same time in single ESP. Actually I am using two different ESP32, one as a server to connect through my Android app through BLE and second as a client to connect to a server to receive sensor data. Can I use one ESP32 as a client and server at a time instead of using two for above mentioned work. Please let me know if some example is present online and guide me to create one.

dhrishi, 2019-12-06:
@ROYMURLI I think you are looking for something like this: https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/coex/gattc_gatts_coex. You can check the README.md file present there and see if it helps.

On a different note, please use ESP32 forum to ask "How to" related questions.

chegewara, 2019-12-06:
GATTS and GATTC can work on 1 device at the same time.
@tobozo did it just recently with arduino on m5stack.

tobozo, 2019-12-06:
Actually the [BLECollector](https://github.com/tobozo/ESP32-BLECollector) runs on M5Stack, Odroid-Go and WROVER_KIT v4.2 :-)

Both GATTS and GATTC (time sharing and file sharing) are implemented in the this poject indeed but concurrent running is currently avoided because I believed this wasn't possible.

However, after reading @chegewara's comment I tried to run both a notification server and a file transfert client concurrently while doing a blescan, and I can confirm it's possible.

ROYMURLI, 2019-12-07:
Thanks for your suggestions @dhrishi @chegewara @tobozo . Will try IDF example. @tobozo I am not familiar with M5Stack much, but will go through your code. Is there any ".C" of your project so that I can take reference from that.

tobozo, 2019-12-07:
@ROYMURLI [there's a ".h"](https://github.com/tobozo/ESP32-BLECollector/blob/master/BLEFileSharing.h)

Alvin1Zhang, 2019-12-10:
@ROYMURLI Thanks for reporting. Feel free to reopen if the issue still happens or please help raise another ticket if new issue happens. Thanks.

ROYMURLI, 2019-12-18:
Hi Again,
I tried to implement the above IDF example of  https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/coex/gattc_gatts_coex in my code but there are some error which is mentioned below.
IDF- v4.1-dev-58-g02c7c3885
OS- Ubuntu
Compiler version - 5.2.0
Toolchain version - crosstool-ng-1.22.0-80-g6c4433a

app_main is given below:-

void app_main()
{
	esp_err_t ret;
	sprintf(device_status, "%s", "IDLE");

	printf("Dozee v2.2: [booting up...]\n");
	printf("Dozee FW verion : %s\n",curr_firmware_version);
	//initialize flash nvs...
	ret = nvs_flash_init();
	if (ret == ESP_ERR_NVS_NO_FREE_PAGES) {
		nvs_flash_erase();
		ret = nvs_flash_init();
	}

	system_init();

	//opening namespace storage...
	esp_err_t err;
	err = nvs_open("storage", NVS_READWRITE, &config); //ESP_ERROR_CHECK(err);
	if(err != ESP_OK)
	{
		printf("Dozee v2.1: [could not open NVS namespace...]\n");
	}

	peripheral_init();
	esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);

	//check if the device is configured or not...
	err = nvs_get_str(config, "configured", NULL , &value_len);
	if(err != ESP_OK)	//device has no configuration settings, set one up...
	{
		printf("Dozee v2.1: [device is not configured, enter configuration...]\n");

		xTaskCreatePinnedToCore(&taskConfigureAndTest, "taskConfigureAndTest", 16384, NULL, 5, NULL, 0);
		xTaskCreatePinnedToCore(&taskUnconfiguredUI, "taskUnconfiguredUI", 2048, NULL, 5, NULL, 1);
	}
	else
	{
		//initialize wifi...
		tcpip_adapter_init();	
		wifi_event_group = xEventGroupCreate();
		esp_event_loop_init(wifi_event_handler, NULL);
		wifi_init_config_t wifi_init_config = WIFI_INIT_CONFIG_DEFAULT();
		esp_wifi_init(&wifi_init_config);
		esp_wifi_set_storage(WIFI_STORAGE_RAM);
		esp_wifi_set_mode(WIFI_MODE_STA);
		esp_wifi_set_auto_connect(false);
		esp_wifi_start();
		vTaskDelay(DELAY_IN_MS(500));	

		nvs_get_str(config, "configured", is_configured, &value_len);
		nvs_get_str(config, "device_id", NULL, &value_len); 
		nvs_get_str(config, "device_id", device_id, &value_len); 
		nvs_get_str(config, "device_name", NULL, &value_len); 
		nvs_get_str(config, "device_name", device_name, &value_len);
		nvs_get_u32(config, "enable_debug", &enable_debug);

		if(strcmp(is_configured, "YES") == 0)
		{
			if(enable_debug) printf("Dozee v2.1: [device is paired...<%s>,<%s>]\n", device_id, device_name);

			tcpip_adapter_set_hostname(TCPIP_ADAPTER_IF_STA, device_name);

			//start upload manager task...
			xTaskCreatePinnedToCore(&taskUploadManager, "taskUploadManager", 25576, NULL, 3, &hdl_upload_manager, 0);

			#ifdef ENABLE_TOUCH
			//start touch sensor task...
			xTaskCreatePinnedToCore(&taskTouchSensor, "taskTouchSensor", 2048, NULL, 2, NULL, 0);
			#endif

			
    		//start recording manager task...
			xTaskCreatePinnedToCore(&taskRecordingManager, "taskRecordingManager", 10192, NULL, 4, &hdl_recording_manager, 1);
			xTaskCreatePinnedToCore(&TaskSupervisor, "SupervisorTask", 4096, NULL, 2, &hdl_supervisor, 1);
		}
		else
		{	
			//device is not paired...
			if(enable_debug) printf("Dozee v2.1: [device is not paired...<%s>,<%s>]\n", device_id, device_name);
			memset(device_status, 0, 16);
			sprintf(device_status, "%s", "UNPAIRED");
			xTaskCreatePinnedToCore(&taskUnpairedUI, "taskUnpairedUI", 4096, NULL, 1, NULL, 0);
		}
	}

	//initialize ble...
	//#ifdef ENABLE_BLE
	
	printf("gatts services initiated \n");
	esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
	ret = esp_bt_controller_init(&bt_cfg);
	if (ret) {
		if(enable_debug) ESP_LOGE(GATTS_TAG, "%s initialize controller failed\n", __func__);
		return;
	}

	ret = esp_bt_controller_enable(ESP_BT_MODE_BLE);
	if (ret) {
		if(enable_debug) ESP_LOGE(GATTS_TAG, "%s enable controller failed\n", __func__);
		return;
	}

	ret = esp_bluedroid_init();
	if (ret) {
		if(enable_debug) ESP_LOGE(GATTS_TAG, "%s init bluetooth failed\n", __func__);
		return;
	}

	ret = esp_bluedroid_enable();
	if (ret) {
		if(enable_debug) ESP_LOGE(GATTS_TAG, "%s enable bluetooth failed\n", __func__);
		return;
	}

	ret = esp_ble_gatts_register_callback(gatts_event_handler);
	if (ret){
		if(enable_debug) ESP_LOGE(GATTS_TAG, "gatts register error, error code = %x", ret);
		return;
	}

	ret = esp_ble_gap_register_callback(gap_event_handler);
	if (ret){
		if(enable_debug) ESP_LOGE(GATTS_TAG, "gap register error, error code = %x", ret);
		return;
	}

	ret = esp_ble_gatts_app_register(PROFILE_CONFIG_APP_ID);
	if (ret){
		if(enable_debug) ESP_LOGE(GATTS_TAG, "gatts app register error, error code = %x", ret);
		return;
	}

	    // gattc regisrter
	//nvs_get_str(config, "configured", is_configured, &value_len);
	if(strcmp(is_configured, "YES") == 0)
	{
	printf("gattc access granted \n");
    ret = esp_ble_gattc_register_callback(esp_gattc_cb);
    if(ret) {
        ESP_LOGE(COEX_TAG, "%s gattc register failed, error code = %x\n", __func__, ret);
        return;
    }

    ret = esp_ble_gattc_app_register(GATTC_PROFILE_C_APP_ID);
    if (ret) {
        ESP_LOGE(COEX_TAG, "%s gattc app register failed, error code = %x\n", __func__, ret);
    }
	}

	esp_err_t local_mtu_ret = esp_ble_gatt_set_local_mtu(500);
	if (local_mtu_ret){
		if(enable_debug) ESP_LOGE(GATTS_TAG, "set local  MTU failed, error code = %x", local_mtu_ret);
	}
	
	//#endif

	return;
}

Problems-
1- When the BLE is advertising (GATTS) I am able to connect only first time after program flashing after that if I disconnect or restart the ESP it advertise the device name but I am unable to connect with my app/mobile.
2- In case of "is_configure = YES" the client (GATTC) is not initiated. As not a single line of GATTC callback function print statement is printed.

SDK config of my project is attached here.
[sdkconfig (copy).txt](https://github.com/espressif/esp-idf/files/3978360/sdkconfig.copy.txt)

ROYMURLI, 2019-12-19:
Hi Again,
I tried to implement the above IDF example of https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/coex/gattc_gatts_coex in my code but there are some error which is mentioned below.
IDF- v4.1-dev-58-g02c7c3885
OS- Ubuntu
Compiler version - 5.2.0
Toolchain version - crosstool-ng-1.22.0-80-g6c4433a

app_main is given below:-

void app_main()
{
esp_err_t ret;
sprintf(device_status, "%s", "IDLE");
printf("Dozee v2.2: [booting up...]\n");
printf("Dozee FW verion : %s\n",curr_firmware_version);
//initialize flash nvs...
ret = nvs_flash_init();
if (ret == ESP_ERR_NVS_NO_FREE_PAGES) {
nvs_flash_erase();
ret = nvs_flash_init();
}

system_init();
//opening namespace storage...
esp_err_t err;
err = nvs_open("storage", NVS_READWRITE, &config); //ESP_ERROR_CHECK(err);
if(err != ESP_OK)
{
printf("Dozee v2.1: [could not open NVS namespace...]\n");
}

peripheral_init();
esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);

//check if the device is configured or not...
err = nvs_get_str(config, "configured", NULL , &value_len);
if(err != ESP_OK) //device has no configuration settings, set one up...
{
printf("Dozee v2.1: [device is not configured, enter configuration...]\n");
xTaskCreatePinnedToCore(&taskConfigureAndTest, "taskConfigureAndTest", 16384, NULL, 5, NULL, 0);
xTaskCreatePinnedToCore(&taskUnconfiguredUI, "taskUnconfiguredUI", 2048, NULL, 5, NULL, 1);
}
else
{
//initialize wifi...
tcpip_adapter_init();
wifi_event_group = xEventGroupCreate();
esp_event_loop_init(wifi_event_handler, NULL);
wifi_init_config_t wifi_init_config = WIFI_INIT_CONFIG_DEFAULT();
esp_wifi_init(&wifi_init_config);
esp_wifi_set_storage(WIFI_STORAGE_RAM);
esp_wifi_set_mode(WIFI_MODE_STA);
esp_wifi_set_auto_connect(false);
esp_wifi_start();
vTaskDelay(DELAY_IN_MS(500));

nvs_get_str(config, "configured", is_configured, &value_len);
nvs_get_str(config, "device_id", NULL, &value_len); 
nvs_get_str(config, "device_id", device_id, &value_len); 
nvs_get_str(config, "device_name", NULL, &value_len); 
nvs_get_str(config, "device_name", device_name, &value_len);
nvs_get_u32(config, "enable_debug", &enable_debug);

if(strcmp(is_configured, "YES") == 0)
{
	if(enable_debug) printf("Dozee v2.1: [device is paired...<%s>,<%s>]\n", device_id, device_name);

	tcpip_adapter_set_hostname(TCPIP_ADAPTER_IF_STA, device_name);

	//start upload manager task...
	xTaskCreatePinnedToCore(&taskUploadManager, "taskUploadManager", 25576, NULL, 3, &hdl_upload_manager, 0);

	#ifdef ENABLE_TOUCH
	//start touch sensor task...
	xTaskCreatePinnedToCore(&taskTouchSensor, "taskTouchSensor", 2048, NULL, 2, NULL, 0);
	#endif

	
	//start recording manager task...
	xTaskCreatePinnedToCore(&taskRecordingManager, "taskRecordingManager", 10192, NULL, 4, &hdl_recording_manager, 1);
	xTaskCreatePinnedToCore(&TaskSupervisor, "SupervisorTask", 4096, NULL, 2, &hdl_supervisor, 1);
}
else
{	
	//device is not paired...
	if(enable_debug) printf("Dozee v2.1: [device is not paired...<%s>,<%s>]\n", device_id, device_name);
	memset(device_status, 0, 16);
	sprintf(device_status, "%s", "UNPAIRED");
	xTaskCreatePinnedToCore(&taskUnpairedUI, "taskUnpairedUI", 4096, NULL, 1, NULL, 0);
}

}

//initialize ble...
//#ifdef ENABLE_BLE

printf("gatts services initiated \n");
esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
ret = esp_bt_controller_init(&bt_cfg);
if (ret) {
if(enable_debug) ESP_LOGE(GATTS_TAG, "%s initialize controller failed\n", func);
return;
}

ret = esp_bt_controller_enable(ESP_BT_MODE_BLE);
if (ret) {
if(enable_debug) ESP_LOGE(GATTS_TAG, "%s enable controller failed\n", func);
return;
}

ret = esp_bluedroid_init();
if (ret) {
if(enable_debug) ESP_LOGE(GATTS_TAG, "%s init bluetooth failed\n", func);
return;
}

ret = esp_bluedroid_enable();
if (ret) {
if(enable_debug) ESP_LOGE(GATTS_TAG, "%s enable bluetooth failed\n", func);
return;
}

ret = esp_ble_gatts_register_callback(gatts_event_handler);
if (ret){
if(enable_debug) ESP_LOGE(GATTS_TAG, "gatts register error, error code = %x", ret);
return;
}

ret = esp_ble_gap_register_callback(gap_event_handler);
if (ret){
if(enable_debug) ESP_LOGE(GATTS_TAG, "gap register error, error code = %x", ret);
return;
}

ret = esp_ble_gatts_app_register(PROFILE_CONFIG_APP_ID);
if (ret){
if(enable_debug) ESP_LOGE(GATTS_TAG, "gatts app register error, error code = %x", ret);
return;
}

// gattc regisrter

//nvs_get_str(config, "configured", is_configured, &value_len);
if(strcmp(is_configured, "YES") == 0)
{
printf("gattc access granted \n");
ret = esp_ble_gattc_register_callback(esp_gattc_cb);
if(ret) {
ESP_LOGE(COEX_TAG, "%s gattc register failed, error code = %x\n", func, ret);
return;
}

ret = esp_ble_gattc_app_register(GATTC_PROFILE_C_APP_ID);
if (ret) {
ESP_LOGE(COEX_TAG, "%s gattc app register failed, error code = %x\n", func, ret);
}
}

esp_err_t local_mtu_ret = esp_ble_gatt_set_local_mtu(500);
if (local_mtu_ret){
if(enable_debug) ESP_LOGE(GATTS_TAG, "set local MTU failed, error code = %x", local_mtu_ret);
}

//#endif

return;

Problems-
1- When the BLE is advertising (GATTS) I am able to connect only first time after program flashing after that if I disconnect or restart the ESP it advertise the device name but I am unable to connect with my app/mobile.
2- In case of "is_configure = YES" the client (GATTC) is not initiated. As not a single line of GATTC callback function print statement is printed.

SDK config of my project is attached here.
[sdkconfig (copy).txt](https://github.com/espressif/esp-idf/files/3981819/sdkconfig.copy.txt)

tobozo, 2019-12-19:
my eyes bleed trying to read the code you pasted, can you edit your post and enclose the source code with ``` so it is properly highlighted ?

gengyuchao, 2019-12-19:
Your sdkconfig may have been modified a lot. Can you provide a project that works and can reproduces the problem?

ROYMURLI, 2019-12-19:
@tobozo  sorry for inconvenience. I marked it properly. Please check. 
@gengyuchao sorry, I can't give my whole project alternatively I can provide the details of each task personally to give more insight of that. 
Before including GATTC, My project is completely based on GATTS , WIFI , PCA 9685, I2C, SPI, UART for communication with other MCU and sensor to get data which is working fine. I want to remove the UART part by including GATTC in ESP. Let me know if you need anything else.

chegewara, 2019-12-19:
> Before including GATTC, My project is completely based on GATTS , WIFI , PCA 9685, I2C, SPI, UART for communication with other MCU and sensor to get data which is working fine.

Probably you should start with this info. Do you track free heap in app, especially in place your app dont work as expected?

Using WIFI + GATTC + GATTS usually leads to heap shortage and you are using I2C and SPI in addition.

ROYMURLI, 2019-12-19:
@chegewara  Yes, I track free heap and it worked as I am not using GATTS always and many task are in hold while I am using GATTS. I maintain my xPortGetFreeHeapSize() always greater than 24000, 95% of time it is 33000+ and in some cases only it come down nearly 24000-26000. In the same way I2C and SPI  dependent task are not always on running. I can share some of my work about task and its dependencies if needed.

gengyuchao, 2019-12-19:
@tobozo Thank you for saving my eyes.
@ROYMURLI Could you provide some log near the position that the gattc may execute? And I think you can start with the gattc_gatts_coex. It is best to ensure that all functions in the examples are included in your project.

ROYMURLI, 2019-12-21:
Hi, I tried implementing above example by including it in my project. The server part (GATTS) is working fine but GATTC part is not working. When I connect it through my mobile the GATTC part is also automatically executed and ESP_GATTC_CONNECT_EVT is called and further more GATTC CB is called along with GATTS. The logs screenshot and sdkconfig is attached here.
[image]

chegewara, 2019-12-21:
In logs you have disconnect reason 19(0x13):
`0x13 Remote User Terminated Connection`

ROYMURLI, 2019-12-25:
Hi again,
I have implemented the gattc_gatts_coex.c example with my project and GATTC and GATTS connection are working fine. In GATTS I am able to read and write the commands and values but in case of GATTC I am able to write but not able to get values through "ESP_GATTC_NOTIFY_EVT". I am receiving data through "ESP_GATTC_NOTIFY_EVT" in gattc_gatts_coex.c example.
Below is the screenshot of logs of receiving data using "gattc_gatts_coex.c". But in case of my project I am unable to receive any data from server through ESP_GATTC_NOTIFY_EVT. logs and sdkconfig of my example is attached below.

gengyuchao, 2019-12-25:
Do you use a different UUID from the gattc_gatts_coex demo to do notify?

ROYMURLI, 2019-12-25:
No, I didn't use different UUID from the gattc_gatts_coex demo to notify, I used the same one in both and YES my characteristic for this service support notify. The edited code of gattc_gatts_coex is attached below and my project declaration of NOTIFY UUID is mentioned below:-
```
//BLE service parameters
  #define GATTS_TAG "DOZEE"
  #define GATTS_SERVICE_UUID_TEST_F   0x00FF
  #define GATTS_CHAR_UUID_TEST_F      0xFF01
  #define GATTS_DESCR_UUID_TEST_F     0x6666
  #define GATTS_NUM_HANDLE_TEST_F     4
  #define TEST_MANUFACTURER_DATA_LEN  17
  #define GATTS_DEMO_CHAR_VAL_LEN_MAX 0x40

  #define PROFILE_NUM 1
  #define PROFILE_CONFIG_APP_ID 0

//Advertising configuration flags
  static uint8_t adv_config_done = 0;
  #define adv_config_flag      (1 << 0)
  #define scan_rsp_config_flag (1 << 1)

//gatts and gattc
  #define PREPARE_BUF_MAX_SIZE        1024
  #define GATTC_PROFILE_NUM           1
  #define GATTC_PROFILE_C_APP_ID      0
  #define GATTC_PROF  ILE_C_APP_ID      0
  #define REMOTE_SERVICE_UUID         0xFFE0
  #define REMOTE_NOTIFY_CHAR_UUID     0xFFE1
  #define INVALID_HANDLE              0
  #define COEX_TAG                    "GATTC_GATTS_COEX"
  #define NOTIFY_ENABLE               0x0001 
  #define INDICATE_ENABLE             0x0002
  #define NOTIFY_INDICATE_DISABLE     0x0000 

  static const char remote_device_name[] = "HC-42";
```
below is the edited example gattc_gatts_coex.c which is working fine.
[gattc_gatts_coex_edited.zip](https://github.com/espressif/esp-idf/files/4000510/gattc_gatts_coex_edited.zip)

gengyuchao, 2019-12-25:
If I understand correctly, your server is HC-42 and the client is ESP32. Notifications are sent from the server to the client. Using the modified gattc_gatts_coex example will work, but you will not receive a notification in your project. Is that so?

ROYMURLI, 2019-12-25:
Absolutely!

gengyuchao, 2019-12-26:
Maybe you can try to change this code in ESP_GATTC_REG_FOR_NOTIFY_EVT from
```
            uint16_t count = 0;
            uint16_t notify_en = 1;
```
to
```
            int count = 0;
            int notify_en = 1;
```
Because there is a known issue that may cause notify enabled fail.

ROYMURLI, 2019-12-26:
Tried with above modification but in unable to write anything to server. Logs are attached below. 
[logs_dozee1.txt](https://github.com/espressif/esp-idf/files/4002292/logs_dozee1.txt)

tobozo, 2019-12-26:
@ROYMURLI does it still occur if you use two separate characteristics (notify / write) to talk to the BLE server ?

ROYMURLI, 2019-12-26:
Yes, I am using two different UUIDs for notify and write as you can see in my this reply. I included all the functions and CB in my projects from gattc_gatts_coex example then also it is not receiving data from server whereas gattc_gatts_coex example works perfect.

> No, I didn't use different UUID from the gattc_gatts_coex demo to notify, I used the same one in both and YES my characteristic for this service support notify. The edited code of gattc_gatts_coex is attached below and my project declaration of NOTIFY UUID is mentioned below:-
> 
> ```
> //BLE service parameters
>   #define GATTS_TAG "DOZEE"
>   #define GATTS_SERVICE_UUID_TEST_F   0x00FF
>   #define GATTS_CHAR_UUID_TEST_F      0xFF01
>   #define GATTS_DESCR_UUID_TEST_F     0x6666
>   #define GATTS_NUM_HANDLE_TEST_F     4
>   #define TEST_MANUFACTURER_DATA_LEN  17
>   #define GATTS_DEMO_CHAR_VAL_LEN_MAX 0x40
> 
>   #define PROFILE_NUM 1
>   #define PROFILE_CONFIG_APP_ID 0
> 
> //Advertising configuration flags
>   static uint8_t adv_config_done = 0;
>   #define adv_config_flag      (1 << 0)
>   #define scan_rsp_config_flag (1 << 1)
> 
> //gattc
>   #define PREPARE_BUF_MAX_SIZE        1024
>   #define GATTC_PROFILE_NUM           1
>   #define GATTC_PROFILE_C_APP_ID      0
>   #define GATTC_PROF  ILE_C_APP_ID      0
>   #define REMOTE_SERVICE_UUID         0xFFE0
>   #define REMOTE_NOTIFY_CHAR_UUID     0xFFE1
>   #define INVALID_HANDLE              0
>   #define COEX_TAG                    "GATTC_GATTS_COEX"
>   #define NOTIFY_ENABLE               0x0001 
>   #define INDICATE_ENABLE             0x0002
>   #define NOTIFY_INDICATE_DISABLE     0x0000 
> 
>   static const char remote_device_name[] = "HC-42";
> ```
> 
> below is the edited example gattc_gatts_coex.c which is working fine.
> [gattc_gatts_coex_edited.zip](https://github.com/espressif/esp-idf/files/4000510/gattc_gatts_coex_edited.zip)

ROYMURLI, 2019-12-28:
Any solution? @chegewara  @tobozo

tobozo, 2019-12-28:
@ROYMURLI there's no canned solution I can think about.

I'd eventually try random things e.g. start the wifi/ble/bluetooth in different orders to see what it gives, also test on builds with/without psram just to make sure it fails on both.

chegewara, 2019-12-28:
@ROYMURLI Ive been working a lot with BLE in past 2 years and i have few procedures to find the problems, not always works, but most cases it does.
1) test peripheral with nRF connect app on android smartphone,
2) duplicate peripheral with nRF connect, both advertising and GATT attributes, to see logs when esp32 is connected and trying to interact like read/write characteristics and descriptors, also you can see if peripheral is configured properly,
3) in esp32 code add GAP and GATT callbacks to print all events (i am working with Kolban's library, in your case just LOG all events)

This is start to see where the problem can be.

gengyuchao, 2020-01-07:
@ROYMURLI Is your problem solved with the solution in #4367 ? Please let me know if you have any other questions. : )

ROYMURLI, 2020-01-07:
> If I understand correctly, your server is HC-42 and the client is ESP32. Notifications are sent from the server to the client. Using the modified gattc_gatts_coex example will work, but you will not receive a notification in your project. Is that so?

Still facing this issue. Trying various alterations. Will inform you soon. I can give you my code personally if possible to debug.

Alvin1Zhang, 2020-04-26:
@ROYMURLI Thanks for reporting, would you please help share if any updates for the issue? Thanks.

ROYMURLI, 2020-04-26:
Its solved, SDK for that is attached here. We can close this issue.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/4535171/sdkconfig.txt)

closed 2020-04-26 by ROYMURLI