espressif/esp-idf #4454: Lan8710a ok in v3 but unstable in v4, corrupt heap errors

will-emmerson, 2019-12-06:
## Environment
- Development Kit:      Olimex-POE-ISO rev B, Olimex Gateway rev E
- Module or chip used:  ESP32-WROOM-32
- IDF version: v3.3 or master
- Build System:         CMake
- Operating System:     Linux
- Power Supply:         USB

## Problem Description
I have been trying to debug a strange problem with ethernet, where I intermittently get corrupt heap errors such as the ones below when I try to do OTA update or download a file, using any v4:

```
I (11359) esp_https_ota: Starting OTA...
I (11359) esp_https_ota: Writing to partition subtype 16 at offset 0x120000
CORRUPT HEAP: multi_heap.c:432 detected at 0x3ffc7010
abort() was called at PC 0x400879e7 on core 0
0x400879e7: multi_heap_assert at /home/willemmerson/esp/esp-idf/components/heap/multi_heap_platform.h:59
 (inlined by) multi_heap_malloc_impl at /home/willemmerson/esp/esp-idf/components/heap/multi_heap.c:432

ELF file SHA256: 3e9953136fce01fad133518a7498c10d864b1e00b2410f31fbe5025711540b49

Backtrace: 0x40085475:0x3ffbed30 0x4008586d:0x3ffbed50 0x400879e7:0x3ffbed70 0x40087d95:0x3ffbed90 0x4008224d:0x3ffbedb0 0x4008227d:0x3ffbedd0 0x4008ad7d:0x3ffbedf0 0x400e9475:0x3ffbee10 0x400e94eb:0x3ffbee30 0x400e955c:0x3ffbee50 0x400e9f59:0x3ffbee70 0x400ea131:0x3ffbee90 0x400f9a0a:0x3ffbeec0 0x40126199:0x3ffbeee0 0x400d94d1:0x3ffbef00 0x40127ad6:0x3ffbef20 0x40087f3d:0x3ffbef50
0x40085475: invoke_abort at /home/willemmerson/esp/esp-idf/components/esp32/panic.c:157

0x4008586d: abort at /home/willemmerson/esp/esp-idf/components/esp32/panic.c:174

0x400879e7: multi_heap_assert at /home/willemmerson/esp/esp-idf/components/heap/multi_heap_platform.h:59
 (inlined by) multi_heap_malloc_impl at /home/willemmerson/esp/esp-idf/components/heap/multi_heap.c:432

0x40087d95: multi_heap_malloc at /home/willemmerson/esp/esp-idf/components/heap/multi_heap_poisoning.c:191

0x4008224d: heap_caps_malloc at /home/willemmerson/esp/esp-idf/components/heap/heap_caps.c:115

0x4008227d: heap_caps_malloc_default at /home/willemmerson/esp/esp-idf/components/heap/heap_caps.c:144

0x4008ad7d: malloc at /home/willemmerson/esp/esp-idf/components/newlib/heap.c:32

0x400e9475: mem_malloc at /home/willemmerson/esp/esp-idf/components/lwip/lwip/src/core/mem.c:237

0x400e94eb: do_memp_malloc_pool at /home/willemmerson/esp/esp-idf/components/lwip/lwip/src/core/memp.c:254

0x400e955c: memp_malloc at /home/willemmerson/esp/esp-idf/components/lwip/lwip/src/core/memp.c:350 (discriminator 2)

0x400e9f59: pbuf_alloc_reference at /home/willemmerson/esp/esp-idf/components/lwip/lwip/src/core/pbuf.c:336

0x400ea131: pbuf_alloc at /home/willemmerson/esp/esp-idf/components/lwip/lwip/src/core/pbuf.c:237

0x400f9a0a: ethernetif_input at /home/willemmerson/esp/esp-idf/components/lwip/port/esp32/netif/ethernetif.c:166

0x40126199: esp_netif_receive at /home/willemmerson/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:660

0x400d94d1: eth_stack_input at /home/willemmerson/esp/esp-idf/components/esp_eth/src/esp_eth.c:89

0x40127ad6: emac_esp32_rx_task at /home/willemmerson/esp/esp-idf/components/esp_eth/src/esp_eth_mac_esp32.c:255

0x40087f3d: vPortTaskWrapper at /home/willemmerson/esp/esp-idf/components/freertos/port.c:143

```

Or this one:

```
CORRUPT HEAP: Bad tail at 0x3ffccf0a. Expected 0xbaad5678 got 0xf9df9a30
abort() was called at PC 0x40082b4e on core 0
0x40082b4e: lock_acquire_generic at /home/willemmerson/esp/esp-idf/components/newlib/locks.c:143

ELF file SHA256: d91a330eb1636750deae43532c4496f7d0ead8dd4cf6c06973b17037c18b16c0

Backtrace: 0x4008fb09:0x3ffbe710 0x4008ff01:0x3ffbe730 0x40082b4e:0x3ffbe750 0x40082c71:0x3ffbe780 0x40150add:0x3ffbe7a0 0x4014c649:0x3ffbea60 0x4014c5d1:0x3ffbeab0 0x40091dd3:0x3ffbeae0 0x4008246e:0x3ffbeb00 0x400958f5:0x3ffbeb20 0x4012f0a9:0x3ffbeb40 0x4011f683:0x3ffbeb60 0x40123d88:0x3ffbeb80 0x40129216:0x3ffbebb0 0x4012ea7a:0x3ffbebd0 0x4011d269:0x3ffbebf0 0x4011d2e8:0x3ffbec10 0x400925d1:0x3ffbec40
0x4008fb09: invoke_abort at /home/willemmerson/esp/esp-idf/components/esp32/panic.c:157
[... 125 lines cut ...]
CONFIG_ETH_RMII_CLK_OUTPUT=y
# CONFIG_ETH_RMII_CLK_OUTPUT_GPIO0 is not set
CONFIG_ETH_RMII_CLK_OUT_GPIO=17
CONFIG_ETH_SMI_MDC_GPIO=23
CONFIG_ETH_SMI_MDIO_GPIO=18
# CONFIG_ETH_PHY_USE_RST is not set
CONFIG_ETH_DMA_BUFFER_SIZE=512
CONFIG_ETH_DMA_RX_BUFFER_NUM=10
CONFIG_ETH_DMA_TX_BUFFER_NUM=10
# CONFIG_ETH_USE_SPI_ETHERNET is not set
# CONFIG_ETH_USE_OPENETH is not set

CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_PARTITION_TABLE_TWO_OTA=y            
CONFIG_ESP32_PANIC_PRINT_HALT=y     
CONFIG_HEAP_POISONING_COMPREHENSIVE=y       
CONFIG_ESP32_ENABLE_COREDUMP_TO_FLASH=y    
CONFIG_FREERTOS_USE_TRACE_FACILITY=y
CONFIG_FREERTOS_USE_STATS_FORMATTING_FUNCTIONS=y
```

Using `phy_device_power_enable_via_gpio` was the only way I could get it working but I don't think this is the issue because this code isn't even needed on Olimex Gateway yet it still has the same problem.

### Code to reproduce this issue
https://gist.github.com/will-emmerson/557ce34c456358997ae3c1ff240311f2

szmodz, 2019-12-06:
probably related:
https://github.com/espressif/esp-idf/issues/4406

suda-morris, 2019-12-09:
@will-emmerson 
Thanks for reporting this issue and your analysis, let's firstly enlarge `ETH_DMA_BUFFER_SIZE` to a larger size (e.g. 1500)

will-emmerson, 2019-12-09:
Thanks, that appears to have fixed it.

closed 2019-12-16 by espressif-bot

benpeoples, 2020-07-21:
> @will-emmerson
> Thanks for reporting this issue and your analysis, let's firstly enlarge `ETH_DMA_BUFFER_SIZE` to a larger size (e.g. 1524)

Hello!  So this just fixed my similar issue.   Can we look at pushing this into the documentation?

bbulkow, 2021-01-06:
Note: this problem still exists in ESP-IDF latest as of 1/6/21. The shipping default for the DMA buffer is still 512. Would like to know if this is a known issue only with certain Ethernet PY or just with the LAN8710a, or whether there was an underlying driver issue resolved and this config change is no longer needed.