espressif/esp-idf #4455: Bug: MQTT connect to local Broker failure

rebeliousconformist, 2019-12-06:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Custom Development 16 MB Flash
- Module or chip used:  [ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [|Linux|]
- Power Supply:         [|external 5V| |]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

Broker is 
<pre>docker run -it -p 192.168.0.102:1883:1883 -p 9001:9001  eclipse-mosquitto
</pre>

Ive also tried rabbitmq as a broker . 
The same issue occurs there as well

Does not connect and hangs in loop. It connects successfully to mqtt://mqtt.eclipse.org
### Actual Behavior
It connects to MQTT broker and starts sending data.

### Steps to repropduce

1. program as shown in https://gist.github.com/rebeliousconformist/5e5bf1b08c1a5afd638bb290c75a1667
2. It is the same program from the MQTT/tcp examples

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

https://gist.github.com/rebeliousconformist/5e5bf1b08c1a5afd638bb290c75a1667

## Debug Logs

```
<pre>rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7216
ho 0 tail 12 room 4
load:0x40078000,len:14640
load:0x40080400,len:4512
entry 0x400806fc
I (77) boot: Chip Revision: 1
I (77) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (42) boot: ESP-IDF v4.1-dev-1086-g93a8603c5-dirty 2nd stage bootloader
I (42) boot: compile time 16:57:29
I (43) boot: Enabling RNG early entropy source...
I (49) boot: SPI Speed      : 40MHz
I (53) boot: SPI Mode       : DIO
I (57) boot: SPI Flash Size : 16MB
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2474c (149324) map
I (167) esp_image: segment 1: paddr=0x00034774 vaddr=0x3ffb0000 size=0x033d4 ( 13268) load
I (172) esp_image: segment 2: paddr=0x00037b50 vaddr=0x40080000 size=0x00400 (  1024) load
I (174) esp_image: segment 3: paddr=0x00037f58 vaddr=0x40080400 size=0x080c0 ( 32960) load
I (196) esp_image: segment 4: paddr=0x00040020 vaddr=0x400d0020 size=0x8b360 (570208) map
I (397) esp_image: segment 5: paddr=0x000cb388 vaddr=0x400884c0 size=0x0c918 ( 51480) load
I (430) boot: Loaded app from partition at offset 0x10000
I (430) boot: Disabling RNG early entropy source...
I (431) cpu_start: Pro cpu up.
I (435) cpu_start: Application information:
I (439) cpu_start: Project name:     mqtt_tcp
I (444) cpu_start: App version:      1
I (449) cpu_start: Compile time:     Dec  6 2019 16:57:29
I (455) cpu_start: ELF file SHA256:  3a48fb9020a82998...
I (461) cpu_start: ESP-IDF:          v4.1-dev-1086-g93a8603c5-dirty
I (468) cpu_start: Starting app cpu, entry point is 0x40081150
I (0) cpu_start: App cpu up.
I (478) heap_init: Initializing. RAM available for dynamic allocation:
I (485) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (491) heap_init: At 3FFB95C0 len 00026A40 (154 KiB): DRAM
I (497) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (504) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (510) heap_init: At 40094DD8 len 0000B228 (44 KiB): IRAM
I (516) cpu_start: Pro cpu start user code
I (535) spi_flash: detected chip: gd
I (535) spi_flash: flash io: dio
I (536) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (543) MQTT_EXAMPLE: [APP] Startup..
I (543) MQTT_EXAMPLE: [APP] Free memory: 271784 bytes
I (553) MQTT_EXAMPLE: [APP] IDF version: v4.1-dev-1086-g93a8603c5-dirty
I (603) wifi: wifi driver task: 3ffc1238, prio:23, stack:3584, core=0
I (603) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (603) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (633) wifi: wifi firmware version: a44d1c6
I (633) wifi: config NVS flash: enabled
I (633) wifi: config nano formating: disabled
I (633) wifi: Init dynamic tx buffer num: 32
I (633) wifi: Init data frame dynamic rx buffer num: 32
I (643) wifi: Init management frame dynamic rx buffer num: 32
I (643) wifi: Init management short buffer num: 32
I (653) wifi: Init static rx buffer size: 1600
I (653) wifi: Init static rx buffer num: 10
I (663) wifi: Init dynamic rx buffer num: 32
I (663) example_connect: Connecting to pemwireless...
I (763) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (763) wifi: mode : sta (bc:dd:c2:c9:0a:60)
E (763) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x3ffbf820 (config or if_key is N)
I (773) example_connect: Waiting for IP
I (1863) wifi: new:&lt;10,2&gt;, old:&lt;1,0&gt;, ap:&lt;255,255&gt;, sta:&lt;10,2&gt;, prof:1
I (2403) wifi: state: init -&gt; auth (b0)
I (2413) wifi: state: auth -&gt; assoc (0)
I (2413) wifi: state: assoc -&gt; run (10)
I (2993) wifi: connected with pemwireless, aid = 2, channel 10, 40D, bssid = 30:b5:c2:eb:87:d2, security type:1
I (2993) wifi: pm start, type: 1

I (3023) wifi: AP&apos;s beacon interval = 102400 us, DTIM period = 1
I (4093) esp_netif_handlers: sta ip: 192.168.0.101, mask: 255.255.255.0, gw: 192.168.0.1
I (4093) example_connect: Got IP event!
</pre>
```

ESP-Marius, 2019-12-09:
Hi,
I tested your code with (with only the IP changed) and the Mosquitto docker container broker and it connects without problems. So this is probably not an issue with the code. 

Seems like there is a problem somewhere with your network setup or configuration on the PC running the broker. Are you able to ping the ESP32 from that PC? Do you get any feedback from the broker at all?

rebeliousconformist, 2019-12-12:
Hi @ESP-Marius: No LAN setting is changed. The code is working with the eclipse-moqsuito broker. Ill do further checks with rabbitmq and post here. Thanks for the prompt reply !

rebeliousconformist, 2020-01-08:
hi @ESP-Marius its working. thanks

closed 2020-01-08 by rebeliousconformist