espressif/esp-idf #4457: nvs_parition_gen.py DLL load failed for ffi, lib from openssl when creating encrypted NVS partition

jtrexr2, 2019-12-07:
## Environment
- Module or chip used:  ESP32-WROOM-32
- IDF version: v3.3-202-gefdddbb25
- Build System:  make
- Compiler version: xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Windows 10
- Power Supply:      External 3.3V

## Problem Description

While running the example to generate an encrypted NVS partition using the following command from the `esp-idf/components/nvs_flash/nvs_partition_generator/`:

`python nvs_partition_gen.py --input sample_singlepage_blob.csv --output sample_singlepage_blob.bin --size 0x3000 --encrypt true --keyfile testdata/sample_encryption_keys.bin`

### Expected Behavior
The encrypted NVS partition should be generated.
### Actual Behavior
Returns an ImportError: `ImportError: DLL load failed: The specified module could not be found.`
### Steps to repropduce

1. Downloaded a fresh copy of the pre-built msys environment for ESP-IDF 3v3 from: https://docs.espressif.com/projects/esp-idf/en/v3.3/get-started/windows-setup.html

2. Cloned internal repo containing `esp-idf release/3v3`.

3. Executed `./esp-idf/tools/windows/windows_install_prerequisites.sh`

4. Executed `python esp-idf/tools/check_python_dependencies.py` and all were satisfied.

5. Executed recommended example code from https://docs.espressif.com/projects/esp-idf/en/v3.3/api-reference/storage/nvs_partition_gen.html`esp-idf/components/nvs_flash/nvs_partition_generator` from within the `esp-idf/components/nvs_flash/nvs_partition_generator/` directory:

`python nvs_partition_gen.py --input sample_singlepage_blob.csv --output sample_si                                                   nglepage_blob.bin --size 0x3000 --encrypt true --keyfile testdata/sample_encryption                                                   _keys.bin`

## Debug Logs
```
$ python nvs_partition_gen.py --input sample_singlepage_blob.csv --output sample_singlepage_blob.bin --size 0x3000 --encrypt true --keyfile testdata/sample_encryption_keys.bin
Traceback (most recent call last):
  File "nvs_partition_gen.py", line 892, in <module>
    main()
  File "nvs_partition_gen.py", line 888, in main
    encr_keys_prefix, version_no, output_dir_path)
  File "nvs_partition_gen.py", line 763, in nvs_part_gen
    write_entry(nvs_obj, row["key"], row["type"], row["encoding"], row["value"])
  File "nvs_partition_gen.py", line 636, in write_entry
    nvs_instance.write_namespace(key)
  File "nvs_partition_gen.py", line 527, in write_namespace
    self.cur_page.write_primitive_data(key, self.namespace_idx, "u8", 0,self)
  File "nvs_partition_gen.py", line 454, in write_primitive_data
    self.write_entry_to_buf(entry_struct, 1,nvs_obj)
  File "nvs_partition_gen.py", line 218, in write_entry_to_buf
    encr_data_ret = self.encrypt_data(data, entrycount,nvs_obj)
  File "nvs_partition_gen.py", line 205, in encrypt_data
    encr_data_ret = self.encrypt_entry(data_val, tweak_val, encr_key_input)
  File "nvs_partition_gen.py", line 134, in encrypt_entry
    backend = default_backend()
  File "C:/Users/Jason/Documents/msys32/mingw32/lib/python2.7/site-packages/cryptography/hazmat/backends/__init__.py", line 15, in default_backend
    from cryptography.hazmat.backends.openssl.backend import backend
  File "C:/Users/Jason/Documents/msys32/mingw32/lib/python2.7/site-packages/cryptography/hazmat/backends/openssl/__init__.py", line 7, in <module>
    from cryptography.hazmat.backends.openssl.backend import backend
  File "C:/Users/Jason/Documents/msys32/mingw32/lib/python2.7/site-packages/cryptography/hazmat/backends/openssl/backend.py", line 53, in <module>
    from cryptography.hazmat.bindings.openssl import binding
  File "C:/Users/Jason/Documents/msys32/mingw32/lib/python2.7/site-packages/cryptography/hazmat/bindings/openssl/binding.py", line 13, in <module>
    from cryptography.hazmat.bindings._openssl import ffi, lib
ImportError: DLL load failed: The specified module could not be found.
```
## Other Info
I followed the suggestions from https://github.com/espressif/esp-idf/issues/2719, but still have the same error above. Those suggestions are:
1. `python -m pip install --user 'cryptography>=2.1.4'`, which was already satisfied.
2. `pacman -Sy mingw-w64-i686-openssl`, which updated the openssl library successfully but did not fix the problem.
3. `pacman -Syu` twice, did not fix the problem (command reports that entire system is up to date).

dobairoland, 2019-12-09:
Hello @jtrexr2. I tried to reproduce your issue but I was able to successfully run the above command with the same IDF commit and up to date MSYS environment as you've described.

Your log indicates that the cryptography package you are using is for `python2.7`. However, after running `pacman -Syu` the default python should be of version 3.8.

Please run the following command: `python --version`. The expected output should be `Python 3.8.0`.

Please run `which python` and the result should be `/mingw32/bin/python`.

You will need package `mingw-w64-i686-python3-cryptography` for python 3. You can install it by running `pacman -S mingw-w64-i686-python3-cryptography`.

If the above suggestions didn't help then please provide the output of the following commands:
```
pacman -Qs cryptography
pacman -Qs openssl
```

Thank you!

jtrexr2, 2019-12-10:
Ah okay, I didn't realize the system python needed to be Python 3. The output of `python --version` shows that I'm indeed on `Python 2.7.15` and the output of `which python` shows `/mingw32/bin/python`. 

Running `pacman -S mingw-w64-i686-python3-cryptography` yields the following:
```
$ pacman -S mingw-w64-i686-python3-cryptography
resolving dependencies...
looking for conflicting packages...
:: mingw-w64-i686-python3 and mingw-w64-i686-python2 are in conflict. Remove mingw-w64-i686-python2? [Y/n] Y
error: failed to prepare transaction (could not satisfy dependencies)
:: removing mingw-w64-i686-python2 breaks dependency 'mingw-w64-i686-python2' required by mingw-w64-i686-python2-appdirs
:: removing mingw-w64-i686-python2 breaks dependency 'mingw-w64-i686-python2' required by mingw-w64-i686-python2-cachecontrol
:: removing mingw-w64-i686-python2 breaks dependency 'mingw-w64-i686-python2' required by mingw-w64-i686-python2-certifi
:: removing mingw-w64-i686-python2 breaks dependency 'mingw-w64-i686-python2' required by mingw-w64-i686-python2-colorama
:: removing mingw-w64-i686-python2 breaks dependency 'mingw-w64-i686-python2' required by mingw-w64-i686-python2-distlib
:: removing mingw-w64-i686-python2 breaks dependency 'mingw-w64-i686-python2' required by mingw-w64-i686-python2-html5lib
:: removing mingw-w64-i686-python2 breaks dependency 'mingw-w64-i686-python2' required by mingw-w64-i686-python2-lockfile
:: removing mingw-w64-i686-python2 breaks dependency 'mingw-w64-i686-python2' required by mingw-w64-i686-python2-msgpack
:: removing mingw-w64-i686-python2 breaks dependency 'mingw-w64-i686-python2' required by mingw-w64-i686-python2-packaging
:: removing mingw-w64-i686-python2 breaks dependency 'mingw-w64-i686-python2' required by mingw-w64-i686-python2-ply
:: removing mingw-w64-i686-python2 breaks dependency 'mingw-w64-i686-python2' required by mingw-w64-i686-python2-progress
:: removing mingw-w64-i686-python2 breaks dependency 'mingw-w64-i686-python2' required by mingw-w64-i686-python2-pycparser
:: removing mingw-w64-i686-python2 breaks dependency 'mingw-w64-i686-python2' required by mingw-w64-i686-python2-pyparsing
:: removing mingw-w64-i686-python2 breaks dependency 'mingw-w64-i686-python2' required by mingw-w64-i686-python2-pytoml
:: removing mingw-w64-i686-python2 breaks dependency 'mingw-w64-i686-python2' required by mingw-w64-i686-python2-retrying
:: removing mingw-w64-i686-python2 breaks dependency 'mingw-w64-i686-python2>=2.7' required by mingw-w64-i686-python2-setuptools
:: removing mingw-w64-i686-python2 breaks dependency 'mingw-w64-i686-python2' required by mingw-w64-i686-python2-six
:: removing mingw-w64-i686-python2 breaks dependency 'mingw-w64-i686-python2' required by mingw-w64-i686-python2-urllib3
:: removing mingw-w64-i686-python2 breaks dependency 'mingw-w64-i686-python2' required by mingw-w64-i686-python2-webencodings
```

Was there a point in the installation of the msys environment where I could have selected a python version, or do I just need to upgrade my system to Python 3?

dobairoland, 2019-12-10:
No @jtrexr2, you don't need Python 3. It was just my (false) assumption based on your log that maybe you have Python 2 and 3 mixed up in MSYS. It seems that that isn't the case. IDF v3.3 should work with both Python 2 and 3 without any issue. As a matter of fact, I just confirmed that the command you are running can be successfully run with Python 2 and 3 as well.

Another thing I don't understand why `mingw-w64-i686-python3 and mingw-w64-i686-python2 are in conflict` in your system. It is possible to have both of them. My guess is that your update wasn't complete for some reason. For example, you have Python 2.7.15 and after upgrade it should be 2.7.17.

Could you please try to run `pacman -Syu` once more? If it isn't updating anything then I think `pacman` is broken in your MSYS. In that case please download again the pre-zipped MSYS environment which should work. Please DO NOT run `windows_install_prerequisites.sh` or `python -m pip install` because those might break something. The first one is used only when creating an environment from scratch (which is not your case if you are downloading the ZIP) and the second one is for Linux (which is also not your case).

jtrexr2, 2019-12-10:
`pacman -Syu` didn't update anything, so I downloaded a fresh copy of the of the msys2 environment. I also checked out a fresh `esp-idf` as well using `git clone -b v3.3 --recursive https://github.com/espressif/esp-idf.git`.

The logs below show my attempt at running the `nvs_partition_gen.py`, seeing the same error, running `pacman -Syu` twice, and then the output `which python` and `python --version`.

```
$ cd esp-idf/components/nvs_flash/nvs_partition_generator/

Jason@LAPTOP-MIBC0K21 MINGW32 /c/Users/Jason/Documents/Software/esp-idf/components/nvs_flash/nvs_partition_generator
$ python nvs_partition_gen.py --input sample_singlepage_blob.csv --output sample_singlepage_blob.bin --size 0x3000 --encrypt true --keyfile testdata/sample_encryption_keys.bin

Traceback (most recent call last):
  File "nvs_partition_gen.py", line 892, in <module>
    main()
  File "nvs_partition_gen.py", line 888, in main
    encr_keys_prefix, version_no, output_dir_path)
  File "nvs_partition_gen.py", line 763, in nvs_part_gen
    write_entry(nvs_obj, row["key"], row["type"], row["encoding"], row["value"])
  File "nvs_partition_gen.py", line 636, in write_entry
    nvs_instance.write_namespace(key)
  File "nvs_partition_gen.py", line 527, in write_namespace
    self.cur_page.write_primitive_data(key, self.namespace_idx, "u8", 0,self)
  File "nvs_partition_gen.py", line 454, in write_primitive_data
    self.write_entry_to_buf(entry_struct, 1,nvs_obj)
  File "nvs_partition_gen.py", line 218, in write_entry_to_buf
    encr_data_ret = self.encrypt_data(data, entrycount,nvs_obj)
  File "nvs_partition_gen.py", line 205, in encrypt_data
    encr_data_ret = self.encrypt_entry(data_val, tweak_val, encr_key_input)
  File "nvs_partition_gen.py", line 134, in encrypt_entry
    backend = default_backend()
  File "C:/Users/Jason/Documents/msys2/msys32/mingw32/lib/python2.7/site-packages/cryptography/hazmat/backends/__init__.py", line 15, in default_backend
    from cryptography.hazmat.backends.openssl.backend import backend
  File "C:/Users/Jason/Documents/msys2/msys32/mingw32/lib/python2.7/site-packages/cryptography/hazmat/backends/openssl/__init__.py", line 7, in <module>
    from cryptography.hazmat.backends.openssl.backend import backend
  File "C:/Users/Jason/Documents/msys2/msys32/mingw32/lib/python2.7/site-packages/cryptography/hazmat/backends/openssl/backend.py", line 53, in <module>
    from cryptography.hazmat.bindings.openssl import binding
  File "C:/Users/Jason/Documents/msys2/msys32/mingw32/lib/python2.7/site-packages/cryptography/hazmat/bindings/openssl/binding.py", line 13, in <module>
    from cryptography.hazmat.bindings._openssl import ffi, lib
ImportError: DLL load failed: The specified module could not be found.

Jason@LAPTOP-MIBC0K21 MINGW32 /c/Users/Jason/Documents/Software/esp-idf/components/nvs_flash/nvs_partition_generator
$

Jason@LAPTOP-MIBC0K21 MINGW32 /c/Users/Jason/Documents/Software/esp-idf/components/nvs_flash/nvs_partition_generator
$ pacman -Syu
:: Synchronizing package databases...
 mingw32 is up to date                                    0.0   B  0.00B/s 00:00 [---------------------------------------------]   0%
 mingw64 is up to date                                    0.0   B  0.00B/s 00:00 [---------------------------------------------]   0%
 msys is up to date                                       0.0   B  0.00B/s 00:00 [---------------------------------------------]   0%
:: Starting core system upgrade...
 there is nothing to do
:: Starting full system upgrade...
 there is nothing to do

Jason@LAPTOP-MIBC0K21 MINGW32 /c/Users/Jason/Documents/Software/esp-idf/components/nvs_flash/nvs_partition_generator
$ pacman -Syu
:: Synchronizing package databases...
 mingw32 is up to date                                    0.0   B  0.00B/s 00:00 [---------------------------------------------]   0%
 mingw64 is up to date                                    0.0   B  0.00B/s 00:00 [---------------------------------------------]   0%
 msys is up to date                                       0.0   B  0.00B/s 00:00 [---------------------------------------------]   0%
:: Starting core system upgrade...
 there is nothing to do
:: Starting full system upgrade...
 there is nothing to do

Jason@LAPTOP-MIBC0K21 MINGW32 /c/Users/Jason/Documents/Software/esp-idf/components/nvs_flash/nvs_partition_generator
$ which python
/mingw32/bin/python

Jason@LAPTOP-MIBC0K21 MINGW32 /c/Users/Jason/Documents/Software/esp-idf/components/nvs_flash/nvs_partition_generator
$ python --version
[... 62 lines cut ...]
    write_entry(nvs_obj, row["key"], row["type"], row["encoding"], row["value"])
  File "nvs_partition_gen.py", line 636, in write_entry
    nvs_instance.write_namespace(key)
  File "nvs_partition_gen.py", line 527, in write_namespace
    self.cur_page.write_primitive_data(key, self.namespace_idx, "u8", 0,self)
  File "nvs_partition_gen.py", line 454, in write_primitive_data
    self.write_entry_to_buf(entry_struct, 1,nvs_obj)
  File "nvs_partition_gen.py", line 218, in write_entry_to_buf
    encr_data_ret = self.encrypt_data(data, entrycount,nvs_obj)
  File "nvs_partition_gen.py", line 205, in encrypt_data
    encr_data_ret = self.encrypt_entry(data_val, tweak_val, encr_key_input)
  File "nvs_partition_gen.py", line 134, in encrypt_entry
    backend = default_backend()
  File "C:/Users/Jason/Documents/msys2/msys32/mingw32/lib/python2.7/site-packages/cryptography/hazmat/backends/__init__.py", line 15, in default_backend
[... cut ...]
  File "C:/Users/Jason/Documents/msys2/msys32/mingw32/lib/python2.7/site-packages/cryptography/hazmat/backends/openssl/backend.py", line 53, in <module>
    from cryptography.hazmat.bindings.openssl import binding
  File "C:/Users/Jason/Documents/msys2/msys32/mingw32/lib/python2.7/site-packages/cryptography/hazmat/bindings/openssl/binding.py", line 13, in <module>
    from cryptography.hazmat.bindings._openssl import ffi, lib
ImportError: DLL load failed: The specified module could not be found.
```

I appreciate your help!

dobairoland, 2019-12-10:
Thank you for your patience and the logs. It seems that the ZIP linked in the documentation is broken.
Please try this one https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20190611.zip which should work.

I will update the documentation.

I'm sorry for the caused inconvenience.

jtrexr2, 2019-12-10:
Awesome, thanks! That works now with no installs or updates out of the box, using python 2.7.16.

jtrexr2, 2019-12-11:
Also @dobairoland, I think the xtensa toolchain included with the link you provided is for version 8.2.0, which appears incompatible with `esp-idf release/v3.3`: 

```
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: esp32-2019r1
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 8.2.0
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
GIT_VERSION=bd25280
WARNING: Toolchain version is not supported: esp32-2019r1
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 8.2.0
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
```

I downloaded and replaced the `xtensa-esp32-elf` for v3.3 from here: https://docs.espressif.com/projects/esp-idf/en/v3.3/get-started/windows-setup-scratch.html and my project builds fine now.

dobairoland, 2019-12-11:
I'm happy that it works. Thank you @jtrexr2 for your reports which helped identify the problem. I keep open this issue until we add a new ZIP to the documentation.

closed 2020-02-26 by dobairoland