espressif/esp-idf #4458: No internet secure

supermikegee, 2019-12-07:
Additional info
Browser info
appCodeName
Mozilla
appName
Netscape
appVersion
5.0 (Linux; Android 7.0; H1711z) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/78.0.3904.108 Mobile Safari/537.36
cookieEnabled
true
onLine
true
platform
Linux armv7l
userAgent
Mozilla/5.0 (Linux; Android 7.0; H1711z) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/78.0.3904.108 Mobile Safari/537.36
userEmail
superbosssu@gmail.com
javaEnabled
false
pluginNames
Page info
URL
https://myaccount.google.com/locationsharing?hl=en
Page structure
<HTML> <HEAD> </HEAD> <BODY id="yDmH0d" class="tQj5Y ghyPEc IqBfM e2G3Fb b30Rkd EIlDfe d8Etdd LcUz9d"> <SCRIPT> </SCRIPT> <DIV class="VUoKZ"> </DIV> <DIV class="pGxpHc"> <HEADER id="gb" class="gb_sa gb_Za gb_Pe gb_Rd gb_ta"> <DIV class="gb_7d"> <DIV class="gb_yc gb_wc gb_ta"> </DIV> </DIV> <DIV class="gb_Md gb_3d gb_Ud gb_Ve"> <DIV class="gb_Rc gb_Zc"> <DIV class="gb_qc gb_Fa"> </DIV> <DIV class="gb_qc gb_tc gb_Fa"> </DIV> <DIV class="gb_qc gb_uc gb_Fa"> </DIV> <DIV class="gb_kc"> <DIV class="gb_lc"> <A id="sdgBod" class="gb_ie gb_mc gb_ge"> <SPAN class="gb_ua gb_Uc"> </SPAN> <SPAN class="gbpn gb_5d gb_Xc"> </SPAN> </A> </DIV> </DIV> <DIV class="gb_Rc gb_Fa gb_Wc gb_Xc"> </DIV> </DIV> <DIV class="gb_Rc gb_Xd gb_He gb_Me gb_Be gb_Ce gb_Ie"> <DIV class="gb_oe gb_ne"> </DIV> <DIV class="gb_De gb_Ha"> </DIV> <DIV class="gb_pe gb_ne"> <DIV class="gb_te gb_se gb_ve"> <DIV class="we87Vc fnnnWd"> <DIV id="ow9" class="U26fgb JRtysb WzwrXb Az1IYb zC7z7b Txs8ad"> <DIV class="NWlf3e MbhUzd"> </DIV> <SPAN class="MhXXcc oJeWuf"> <SPAN class="Lw7GHd snByac"> <SPAN class="DPvwYc b5edlf"> </SPAN> </SPAN> </SPAN> </DIV> </DIV> </DIV> </DIV> </DIV> <DIV class="gb_Sc gb_Ua gb_Rc gb_1d gb_Zd"> <DIV class="gb_Tc"> <DIV class="gb_Pa gb_dd gb_xg gb_i gb_Hf"> <DIV class="gb_Ff gb_Ta gb_xg gb_i"> <A class="gb_D gb_Oa gb_i"> <SPAN class="gb_Ia gbii"> </SPAN> </A> <DIV class="gb_2a"> </DIV> <DIV class="gb_1a"> </DIV> </DIV> </DIV> </DIV> </DIV> <DIV class="gb_H gb_F gb_U gb_X gb_Ue"> </DIV> <DIV class="gb_3a gb_F gb_l gb_ta gb_4a"> </DIV> </DIV> <DIV class="gb_Vd gb_3d"> </DIV> </HEADER> <SCRIPT> </SCRIPT> </DIV> <SCRIPT> </SCRIPT> <C-WIZ class="zQTmif SSPGKf O5K2Vd"> <DIV class="T4LgNb"> <DIV class="VjFXz"> </DIV> <DIV class="EWuRAb OUzXuf"> <DIV class="ikIPKb"> <DIV class="edoSyc"> <DIV class="U26fgb mUbCce fKz7Od UQO9ef zC7z7b iP4hIb"> <DIV class="VTBa7b MbhUzd"> </DIV> <SPAN class="xjKiLb"> <SPAN class="Ce1Y1c"> <SPAN class="DPvwYc sm8sCf ew338c"> </SPAN> </SPAN> </SPAN> </DIV> <DIV class="U26fgb mUbCce fKz7Od UQO9ef zC7z7b NQ3IFc"> </DIV> <H1 class="ZZ9xL"> </H1> </DIV> <DIV class="plevpc"> <DIV class="vCJLQ"> </DIV> </DIV> </DIV> <DIV class="MbGbAd"> </DIV> </DIV> <DIV> <C-WIZ> <DIV class="YwhQ0 "> <DIV class="yrQA5d"> <SPAN class="SM4h0e"> <A> </A> </SPAN> <H3> </H3> <SPAN> </SPAN> </DIV> <DIV class="T4sa0d"> </DIV> <DIV class="hyMrOd "> <DIV class="dEcOad a2Lyuc"> <DIV class="Kgyocb"> <DIV class="O32PKc wQT4Yd"> <DIV class="QnS3qe "> <DIV class="qt0M0c"> <DIV class="Yc2eif"> <DIV class="meUf6b"> <DIV class="eSa2Qd"> <SPAN class="DPvwYc Mmmyd"> </SPAN> </DIV> </DIV> </DIV> </DIV> </DIV> </DIV> </DIV> <DIV class="u2FzRc"> </DIV> </DIV> <DIV class="qNeFe RH9rqf"> </DIV> </DIV> </DIV> <DIV class="YwhQ0 "> <DIV class="yrQA5d"> <DIV class="FJWsrc"> <SPAN class="YzUys"> <A> </A> <A> </A> <A> </A> </SPAN> </DIV> </DIV> </DIV> <C-DATA id="i2"> </C-DATA> </C-WIZ> </DIV> <DIV class="vbHcXc gHBEZd"> </DIV> </DIV> <C-DATA id="i1"> </C-DATA> </C-WIZ> <SCRIPT> </SCRIPT> <SCRIPT> </SCRIPT> <SCRIPT id="_ij"> </SCRIPT> <SCRIPT> </SCRIPT> <SCRIPT> </SCRIPT> <DIV id="ZCHFDb" class="UsciIe gHBEZd"> <FOOTER class="e8yP4b gHBEZd"> <A class="StqqDd"> </A> <SPAN> </SPAN> <A class="StqqDd"> </A> <SPAN> </SPAN> <A class="StqqDd"> </A> </FOOTER> <NOSCRIPT> </NOSCRIPT> </DIV> <SCRIPT> </SCRIPT> <DIV class="gb_gf"> </DIV> <DIV class="gb_if"> </DIV> <DIV class="gb_hf"> </DIV> <DIV class="gb_Kd"> </DIV> <DIV class="gb_be"> </DIV> <DIV class="gb_sc"> </DIV> <DIV> </DIV> <SCRIPT> </SCRIPT> </BODY> </HTML>

supermikegee, 2019-12-07:
#4445

supermikegee, 2019-12-07:
#4457

supermikegee, 2019-12-07:
#4015

supermikegee, 2019-12-07:
#2980

supermikegee, 2019-12-07:
#2980 #3880

Alvin1Zhang, 2019-12-08:
Close due to not follow the issue template, feel free to reopen with more details. Thanks.

closed 2019-12-08 by Alvin1Zhang