espressif/esp-idf #4459: export.sh fail on shell fish

kylefelipe, 2019-12-08:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): cv4.1-dev-1086-g93a8603c5

- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): v4.1-dev-1086-g93a8603c5
- Operating System:     Linux (Debian Buster)
- Power Supply:         USB

## Problem Description

I'm using [FISH SHELL](https://fishshell.com/)

Following the steps from [Get Started](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#get-started-get-esp-idf), on Step 4 - Setup environment variables, on [Linux](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#id5) 
The follow line, from step above:

```
. $HOME/esp/esp-idf/export.sh
```

Issues a error
```Shell
. $HOME/esp/esp-idf/export_fish.sh                                       23:53
~/esp/esp-idf/export_fish.sh (linha 1): Unexpected ')' found, expecting '}'
# This script should be sourced, not executed.

function realpath_int() {
    wdir="$PWD"; [ "$PWD" = "/" ] && wdir=""
    arg=$1
    case "$arg" in
        "/"*) scriptdir="${arg}";;
^
do arquivo ~/esp/esp-idf/export_fish.sh
	chamado na linha 359 do arquivo “/usr/share/fish/config.fish”

na função “.”
	chamado na entrada padrão
```

igrr, 2019-12-09:
Hi @kylefelipe, so far we haven't added support for `fish` shell to the install/export scripts. These scripts are tested with `bash` only at the moment. If you are familiar with this shell and if you can send a PR to make the script compatible with `fish`, we will merge it. Otherwise we will treat this issue as a feature request for `fish` support.

eliabieri, 2020-02-26:
I used [bass](https://github.com/edc/bass) so source export.sh.
Install bass using fisherman, then run
$ bass source export.sh

stormwatch, 2020-03-13:
How should I call bass if I installed with `env IDF_TOOLS_PATH=/opt/espresif ./install.sh`?

closed 2020-03-31 by espressif-bot