espressif/esp-idf #4460: ULP stage_inc and jumps

chegewara, 2019-12-08:
Hi,
i am trying to write simple ULP app that would be running in deep sleep and i found that neither `sleep` can be use in deep sleep, which is clearly described in documentation and `jumps` in conjunction with `stage_inc`, which i didnt find any info about it.
https://docs.espressif.com/projects/esp-idf/en/v3.2/api-guides/ulp_instruction_set.html#stage-inc-increment-stage-count-register
Now is the question, is it bug or it is normal and jumps cant be used in deep sleep?
I have to add that ULP code works if is not used in deep sleep mode.
I am using arduino with ULP (esp-idf v3.2).
Thanks.

igrr, 2019-12-10:
@chegewara can you please clarify how you are using the stage counter? Are you counting within one run of the ULP program, or you `halt` the program and then use the counter when the ULP wakes up again? Also, please report which ULP assembler version you are using (run `esp32ulp-elf-as --version` to check).

chegewara, 2019-12-10:
I am using ULP in arduino with this ulp toolchain:
https://github.com/duff2013/ulptool

This is version output:
```
./esp32ulp-elf-as --version
GNU assembler (GNU Binutils) 2.28.51-esp-20190801
Copyright (C) 2017 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or later.
This program has absolutely no warranty.
This assembler was configured for a target of `esp32ulp-elf'.
```
```
xtensa-esp32-elf-as --version
GNU assembler (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 2.25.1
Copyright (C) 2014 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or later.
This program has absolutely no warranty.
This assembler was configured for a target of `xtensa-esp32-elf'.
```
During my learning process i found few issues. 
1) is with stage_inc and jumps, it is working as long as i start ulp program without entering deep sleep mode,  if i remember test code was something based on this sketch:
https://github.com/duff2013/ulptool/blob/master/src/ulp_examples/ulp_adc/adc.s

2) In my other test code i am reading `adc` in deep sleep ULP code and i am using `analogRead()` in main app code, but few hours ago i found this is wrong. It is causing some very odd behavior. And by odd i mean that i have ULP variable which is increased by 1 in every timer reset with halt, it is working until variable value is multiply of 32. This is ULP code and i know it is not perfect:
```
#include "soc/rtc_cntl_reg.h"
#include "soc/rtc_io_reg.h"
#include "soc/soc_ulp.h"

  /* Define variables, which go into .bss section (zero-initialized data) */
  .bss

  /* ADC1 channel 7, GPIO35 */
  .set adc_channel, 7

  .global toggle_counter
toggle_counter:
  .long 0

  .global voltage
voltage:
  .long 0

  /* Code goes into .text section */
  .text
  .global entry
entry:

  and r0, r0, 0
  /* Read toggle counter */
  move r3, toggle_counter
  ld r0, r3, 0
  /* Increment */
  add r0, r0, 1
  /* Save counter in memory */
  st r0, r3, 0

  and r0, r0, 31
  jump toggle_set, eq

  /* Set the RTC_GPIO27 output LOW (clear output) */
  WRITE_RTC_REG(RTC_GPIO_OUT_W1TC_REG,RTC_GPIO_OUT_DATA_W1TC_S+10,1,1)
  WRITE_RTC_REG(RTC_GPIO_OUT_W1TC_REG,RTC_GPIO_OUT_DATA_W1TC_S+13,1,1)
  jump toggle_complete

  .global toggle_set
toggle_set:
  move r2, voltage
  ld r0, r2, 0
  and r0, r0, 0
  /* read adc1 channel 7 + 1 to register R1*/
  adc r1, 0, adc_channel + 1
  /* store ADC value to voltage */
  st r1, r2, 0
  /* load R1 to R0 */
  ld r0, r1, 0
  jumpr alert, 3825, lt
  /* Set the RTC_GPIO27 output HIGH (set output) */
  WRITE_RTC_REG(RTC_GPIO_OUT_W1TS_REG,RTC_GPIO_OUT_DATA_W1TS_S+13,1,1)
  jump toggle_complete

  .global alert
alert:
  WRITE_RTC_REG(RTC_GPIO_OUT_W1TS_REG,RTC_GPIO_OUT_DATA_W1TS_S+10,1,1)

  .global toggle_complete
toggle_complete:

  /* Get ULP back to sleep */
  .global exit
exit:
  halt
```
To see when it stuck i start ulp program without entering deep sleep and printing `toggle_counter` value.

Summarize, i am more suspecting my lack of knowledge in using ULP rather than bug, but i thought i will report it. In worst case i will learn something.
Thanks.

chegewara, 2019-12-11:
I will close this issue for now, because it seems that i dont fully understand ULP logic. I will open it when i confirm problems.

closed 2019-12-11 by chegewara

chegewara, 2019-12-11:
Thanks @igrr for suggestion. Like i said, i am learning ULP hard way, and i thought that `stage` value is preserved after resting timer/program with halt, but its not.
Also other problems i mentioned here were caused by my lack of knowledge. For example ive been trying to use `ADC` in ULP and analogRead() in main app, then i found this:
```
This function reconfigures ADC1 to be controlled by the ULP.
 * Effect of this function can be reverted using adc1_get_raw function.
```
and i understood i have to call `adc1_ulp_enable()` each time before entering deep sleep and running ULP.