espressif/esp-idf #4464: Storing OTA URL in NVS : Fetching back for OTA update.

Yuvaraj4041, 2019-12-09:
**Environment**
    Module or chip used: ESP32-Wroom-32D Dev.kit
    IDF version 3.3

**Problem Description**
Whenever esp is started user has to enter OTA_server_url through UART and esp will store it in NVS. When OTA start interrupt occurred esp has to fetch stored url from nvs and has to connect with it and start OTA task. 
 
**Expected Behavior**
Once after fetching URL from NVS it has to connect with and start OTA.

**Actual Behavior**
Stored OTA URL is fetching form NVS and it is not connecting with it. It is giving Error. "HTTP_CLIENT: Error parse url"

**Code for storing URL in NVS:**
printf ("Entered OTA_Server_URL:%s", OTA_Server_URL);
i++;
esp_err = nvs_set_blob(handle, "OTA_URL", OTA_Server_URL, 90);
if(esp_err == ESP_OK) 
{
	printf("OTA_Server_URL is stroed in NVS: %s", OTA_Server_URL);
}

**Code for fetching url from NVS:**
esp_err = nvs_open("espmanudata", NVS_READWRITE, &handle);
    if (esp_err != ESP_OK){
		printf("\nError in open NVS for Fetching OTA Server URL\n");
	}	 
	char* parameter4 = "OTA_URL";
	size_t OTA_Server_URL_size;
	esp_err = nvs_get_blob(handle, parameter4, NULL, &OTA_Server_URL_size);
	if(esp_err != ESP_OK) 
	{
		printf("\nError in NVS get OTA_Server_URL data size! or No OTA_Server_URL in NVS\n");
	}
	char* EXAMPLE_SERVER_URL = malloc(OTA_Server_URL_size);
	esp_err = nvs_get_blob(handle, parameter4, EXAMPLE_SERVER_URL, &OTA_Server_URL_size);
	if(esp_err != ESP_OK) 
	{
		if(esp_err == ESP_ERR_NVS_NOT_FOUND) printf("\nKey %s not found\n", parameter4);
		printf("\nError in NVS get OTA_Server_URL data! or No Data in NVS\n");
	}
	else
	{
		printf("Data stored in NVS for key: %s is: %s\n", parameter4, EXAMPLE_SERVER_URL);
	}
	nvs_close(handle);

**Code for connecting to URL for OTA update.** 
esp_http_client_config_t config = {
	.url = EXAMPLE_SERVER_URL,
        .cert_pem = (char *)server_cert_pem_start,
    };

**Debug Logs:**
sc: BTN Pressed Down.[1B][0m
sc: Short Pressed : Starting OTA
 OTA_TASK: SHA-256 for the partition table: : d1c0e9d02fa9d26cd2e1984e7b5dd20157204f501ddc83ce82229e5f3175ee8b[1B][0m
 OTA_TASK: SHA-256 for bootloader: : 49ae113e266a2a32b84e98e22db97ebfcece4efdaa8d5361ceb8bc90453ff548[1B][0m
 OTA_TASK: SHA-256 for current firmware: : 009feb4412d3e5a37d2680056dfc016ef8c2447c6c06b45352a685175964aeeb[1B][0m
 OTA_TASK: Starting OTA example...[1B][0m
 OTA_TASK: Running partition type 0 subtype 0 (offset 0x00010000)[1B][0m
Data stored in NVS for key: OTA_URL is: https://switch.s3.ap-south-1.amazonaws.com/Switch.bin

HTTP_CLIENT: Error parse url https://switch.s3.ap-south-1.amazonaws.com/Switch.bin
HTTP_CLIENT: Error while setting default configurations[1B][0m
OTA_TASK: Failed to initialise HTTP connection[1B][0m
OTA_TASK: Exiting task due to fatal error...[1B][0m

Yuvaraj4041, 2019-12-10:
Hi @mahavirj 

Any updates?

mahavirj, 2019-12-10:
@Yuvaraj4041 If you log `strlen(url)` at following, does it match in working and non-working case?

https://github.com/espressif/esp-idf/blob/efdddbb253b0ceb9286a64d42ea42c9e9129f6b8/components/esp_http_client/esp_http_client.c#L677

Yuvaraj4041, 2019-12-10:
@mahavirj Thanks.

Below is the log
HTTP_CLIENT: URL String Length 78 - non-working case
HTTP_CLIENT: URL String Length 76 - working case. 

I check size of URL before giving it to uart. It was 76. and its is storing in nvs.
But if i fetch it from nvs and its giving URL String Length 78. I really dont know whats happening.

mahavirj, 2019-12-10:
@Yuvaraj4041 You may check contents of buffer using `ESP_LOG_BUFFER_HEX` and see what additional bytes corresponds to (might be `\r\n` sequence). Interim, if you repeat same exercise on `nvs_get_blob` and `nvs_set_blob` with some known data, do you still see inconsistency in returned length?

Yuvaraj4041, 2019-12-10:
@mahavirj  You are correct. Some additional bytes added with the URL stored in NVS (0d and 0a). I repeated for some known data, in this case i am getting proper length matching(without 0d and 0a). 
The data stored in nvs with length mismatching issue are obtained from UART. I am missing something in uart part.

closed 2019-12-10 by Yuvaraj4041