espressif/esp-idf #4465: Reset issues while after brownout detection

vivekalways, 2019-12-09:
In my application i am using ESP32 Wrover with ethernet ,In that after application firmware load, it does not execute on 1st two times hard reset. 
it shows following error...
esp rst:0x1 (POWER_ON_RESET) boot:0xf (DOWNLOAD_(UART0)
waiting for download.
 After two times of pressing reset button..it executes application firmware.
The problem is whenever the esp32 restarts, it hangs on running condition.
In my Application i am using following boot strapping pins in my case.
GPIO 0 up for clock & Down for Boot loader
GPIO 2 open condition
GPIO 5 TX 
GPIO 12 for clock enable 
GPIO 15 for I2C data pull up

Here is my schematic
[image]
[image]

Please help me out

Alvin1Zhang, 2019-12-10:
@vivekalways Thanks for reporting, it seems there is a post on our forum with same contents, https://esp32.com/viewtopic.php?f=2&p=53002#p53002, thanks.

suda-morris, 2019-12-10:
From your description, your design uses IO0 to input RMII clock from external. Keep in mind, IO0 is a special strap pin, ESP32 can go into different mode (download mode, flash load mode) according to the level on IO0 at the power on reset stage. If you're lucky, the external RMII clock is at low level, then program just run normally. However, if the external RMII clock is at high level, ESP32 will go into download mode, just like what you said `waiting for download.` 

In this case, the solution is to find some way to make sure the external RMII clock is not working at the reset stage (e.g. using one GPIO to control the power of external clock or enable pin).

vivekalways, 2019-12-10:
@suda-morris thanks for your valuable comment.i will check and let you know the result.

suda-morris, 2020-05-06:
We have new Ethernet doc here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_eth.html
Please read the guide before designing your own Ethernet board. Feel free to create a new post in our forum https://esp32.com/ to ask for help if you still have such hardware issue.

closed 2020-05-06 by suda-morris