espressif/esp-idf #4468: 如何使用RMT driver发送gap长度大于32767us的红外码

lxdemon, 2019-12-10:
受限于rmt_item32_t的限制，一个逻辑电平长度最长为32767us，然而有些红外码的stop bit长度大于32767us，这个时候该如何处理？

我现在测试的红外码stop bit低电平为54ms，我有试过将电平截断成两部分(32ms+22ms)，中间添加一个长度为0的高电平，item长度加1，但是我发现长度为0之后的电平会被丢掉。

期待您的回复，谢谢！

costaud, 2019-12-10:
请问具体是哪个协议。 对于超出范围时间，能否通过timer来辅助完成？

lxdemon, 2019-12-11:
我们是做全球红外码库的，所以看到其实有很多协议的码都有gap长度大于32767us的情况，我也有想到通过timer来辅助，只是想确认一下有没有更好的途径。我现在还有1个疑问，想请教一下：是否支持无载波发射

我之前有尝试用ESP8266来做红外发射，参考的是RTOS_SDK的peripherals里面有一个ir_tx的example，用的是i2s+hw_timer来实现的，但是当时发现连到wifi后，hw_timer偶尔会有一点抖动，造成时间不准确而导致有时候解码会失败，后来就没有做了。现在我们打算用ESP32来做，所以想到也可以使用类似的方法，我测过了timer group应该是没问题的，但是我现在不知道如何用i2s来产生载波，或者有更好的建议来做载波？需要的载波范围：21K-455KHz

costaud, 2019-12-11:
能否提供一个具体的协议名称，我们评估一下有没有更便捷的支持方式。

使用 esp32 的 RMT模块，是全硬件实现，不会有抖动。RMT模块本身可以选择是否打开载波，不需要再用其他模块来产生。建议使用 RMT 模块来实现。

如果必须兼容原代码，只要产生21K-455K的方波即可，ledc模块更加方波。

另外想请教一下，哪一个红外协议需要455KHz的载波频率 ？

lxdemon, 2019-12-11:
谢谢你的解答，我用解码器捕捉了2个红外码
[image]
第一个红外码的stop bit的gap长度是87.79ms，协议是ASTR01
第二个载波频率为455KHz，是美国一个BANG & OLUFSEN的TV，但是目前解码器还无法解码

lxdemon, 2019-12-17:
Hi, 我们现在想做红外学习功能，我看了rmt_nec_tx_rx的那个红外学习的demo，好像是只支持NEC的协议，而且没有学习到载波的频率？

我这边需要支持所有的红外协议以及载波频率，有没有建议的做法？

koobest, 2019-12-19:
简单的做法是使用 UART 模块的波特率检测功能, UART 波特率检测会将波形最窄的高电平和低电平的周期记录下来, 包括最窄高低电平的周期.比较容易实现

lxdemon, 2019-12-26:
Hi，我现在使用了GPIO中断加timer group来实现红外学习，具体为IO18下降沿触发中断(即一个载波)，然后timer记录下时间(5 tick为1us)，同时设置500ms超时作为学习结束标志，然后比较记录下的每个载波时间长度，小于100us视为红外高电平的载波时间段，并累加载波，否则视为红外低电平， 但是我发现有时候学习时记录到的电平时间有误。
[image]
图中绿色标记的第一条红外码是待学习的红外信号，第二条为学习完然后发出来的红外信号，
[image]
这是红外波形对应时间段的log，我怀疑IO中断被打断了，导致了1588这个错误的时间数据，以上只是我的猜测，不知道有没有这种可能，望解答。

附上红外接收IO口配置及中断代码片段
static void ir_rx_driver_init(void)
{
    esp_log_level_set(TAG, ESP_LOG_DEBUG);

    gpio_config_t io_conf;
    io_conf.intr_type = GPIO_PIN_INTR_NEGEDGE;
    io_conf.pin_bit_mask = 1ULL << GPIO_IR_RX;
    io_conf.mode = GPIO_MODE_INPUT;
    io_conf.pull_up_en = 0;
    gpio_config(&io_conf);

    gpio_install_isr_service(ESP_INTR_FLAG_IRAM);
    gpio_isr_handler_add(GPIO_IR_RX, ir_rx_gpio_isr_handler, (void*) GPIO_IR_RX);

    ir_rx_timer_init(TIMER_GROUP_IR_RX, TIMER_ID_IR_RX);
}

PRIVILEGED_DATA static portMUX_TYPE xTaskQueueMutex = portMUX_INITIALIZER_UNLOCKED;
static void IRAM_ATTR ir_rx_gpio_isr_handler(void* arg)
{
    if (!p_carrier->coming){
        p_carrier->coming = true;
        TIMERG0.hw_timer[TIMER_ID_IR_RX].load_high = 0ULL;
        TIMERG0.hw_timer[TIMER_ID_IR_RX].load_low = 0ULL;
        TIMERG0.hw_timer[TIMER_ID_IR_RX].reload = 1;

        TIMERG0.hw_timer[TIMER_ID_IR_RX].alarm_high = (uint32_t) (IR_RX_CARRIER_TIMEOUT >> 32);
        TIMERG0.hw_timer[TIMER_ID_IR_RX].alarm_low = (uint32_t) IR_RX_CARRIER_TIMEOUT;
        TIMERG0.hw_timer[TIMER_ID_IR_RX].config.alarm_en = TIMER_ALARM_EN;
        TIMER_IR_RX_START;
        ets_printf("Pulse coming\n");
        return;
    }

    taskENTER_CRITICAL_ISR(&xTaskQueueMutex);
    TIMERG0.hw_timer[TIMER_ID_IR_RX].update = 1;
    uint32_t timer_counter_value = TIMERG0.hw_timer[TIMER_ID_IR_RX].cnt_low;

    if ( ((p_carrier->w_index + 1) & (IR_RX_BUFF_CAPABILITY)) != p_carrier->r_index ){
        p_carrier->buff[p_carrier->w_index++] = timer_counter_value;
        p_carrier->w_index &= IR_RX_BUFF_CAPABILITY;
        p_carrier->capture_length ++;
    }
    else
        ets_printf("Carrier buff overflow!\n");

    TIMERG0.hw_timer[TIMER_ID_IR_RX].load_high = 0ULL;
    TIMERG0.hw_timer[TIMER_ID_IR_RX].load_low = 0ULL;
    TIMERG0.hw_timer[TIMER_ID_IR_RX].reload = 1;
    taskEXIT_CRITICAL_ISR(&xTaskQueueMutex);
}
如配置有不当之处忘指出，感谢！

costaud, 2019-12-26:
可以尝试，在Core1上运行ir_rx_driver_init，来初始化gpio和timer。

lxdemon, 2019-12-27:
Hi，请问是调用IPC的API吗，能具体说说嘛

我尝试用esp_ipc_call来调用ir_rx_driver_init，但是上电无限重启，控制台信息：overflow in task ipc1 has been detected，我换用esp_ipc_call_blocking又会一直阻塞。

附上相关代码，麻烦指正
static void ir_rx_driver_init(void *arg)
{
    esp_log_level_set(TAG, ESP_LOG_DEBUG);

    gpio_config_t io_conf;
    io_conf.intr_type = GPIO_PIN_INTR_NEGEDGE;
    io_conf.pin_bit_mask = 1ULL << GPIO_IR_RX;
    io_conf.mode = GPIO_MODE_INPUT;
    io_conf.pull_up_en = 0;
    gpio_config(&io_conf);

    gpio_install_isr_service(ESP_INTR_FLAG_IRAM);
    gpio_isr_handler_add(GPIO_IR_RX, ir_rx_gpio_isr_handler, (void*) GPIO_IR_RX);

    ir_rx_timer_init(TIMER_GROUP_IR_RX, TIMER_ID_IR_RX);
}

void ir_learn_start(bool learn_search, uint8_t *p_learn_buff, uint8_t buff_length, uint8_t *p_result)
{
    p_learn->learn_search = learn_search;
    p_learn->data = p_learn_buff;
    p_learn->length = buff_length;
    memset(p_learn->data, 0, p_learn->length);
    p_learn->result = p_result;
    if (p_learn->result != NULL)
        *p_learn->result = 0;
    
    xTaskCreate(ir_learn_task, "ir learn task", 3072, NULL, 10, &pTaskIrLearnHandle);
}

void ir_learn_driver_init(void)
{
    int val = 0x5a5a;
    esp_ipc_call(1, ir_rx_driver_init, (void *)&val);
}

上电会调用ir_learn_driver_init来初始化，调用ir_learn_start开始学习，我想问下只用把ir_rx_driver_init放在core1运行吗，ir_learn_start以及ir_learn_task需不需要也放在core1运行

costaud, 2019-12-27:
可以创建一个core1上的任务来进行初始化：
用    xTaskCreatePinnedToCore 来创建，最后一个参数指定哪个核(填1）。
目的是让实时性要求高的中断，由core1来响应和处理。

lxdemon, 2019-12-27:
感谢你的解答，问题已解决。

closed 2019-12-27 by costaud