espressif/esp-idf #4471: How to auto configure all node with same SSID without manual configuration

JDchew, 2019-12-11:
I am very new to mesh wifi network. As now, I did built up a mesh network by pre-configure each ESP32 SSID inside sdkconfig under example configuration section. Now I wish to move to next step, which is set only the central node, then the other ESP32 automatic connect to it without any configuration. Is it possible to do that ?

Alvin1Zhang, 2019-12-11:
@JDchew Thanks for reporting. Would you please help share which example configuration you are referring to? Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

Alvin1Zhang, 2020-03-10:
@JDchew Thanks for reporting. Will close due to short of feedback, feel free to reopen with more details. Thanks.

closed 2020-03-10 by Alvin1Zhang