espressif/esp-idf #4472: NTP issue ( year 2036 )

burkulesomesh43, 2019-12-11:
Hello,

I am using [esp-idf v3.3(LTS)](https://github.com/espressif/esp-idf/releases/tag/v3.3) and trying to get time from ntp server( asia.pool.ntp.org ). but sometimes I am getting wrong time from ntp server i.e. 07/02/2036. after doing R&D into this, I found that when there is no response from NTP server time is set to year 2036.
Please check following link for your reference.
[https://github.com/gmag11/NtpClient/issues/70](url)

I tried to recreate this but not found. I am using following sample code to resolve. please check for the same and let me know how to handle such problem.
`void get_time() {

sntp_setoperatingmode(SNTP_OPMODE_POLL);
sntp_setservername(0,"asia.pool.ntp.org");
sntp_init();

 time_t now;
 struct tm timeinfo;
 time(&now);
 setenv("TZ", "UTC-05:30", 1);
 tzset();
 localtime_r(&now, &timeinfo);

while( timeinfo.tm_year <= 70 )
{
	time(&now);
	localtime_r(&now, &timeinfo);
	ESP_LOGI(TAG, "Trying to reach NTP server...");
		
	vTaskDelay(5000 / portTICK_PERIOD_MS);
}
	
strftime(Timebuffer, sizeof(Timebuffer), "%d-%m-%Y %T", &timeinfo);
ESP_LOGI(TAG, "The current date/time is: %s", Timebuffer);
} `

we can handle it by checking year but not looks like its good solution. please suggest.

burkulesomesh43, 2019-12-18:
Awaiting reply...
Kindly help asap.

KonstantinKondrashov, 2019-12-21:
Hi @burkulesomesh43!

>  I found that when there is no response from NTP server time is set to year 2036.

You can use `sntp_get_sync_status()` function to get the moment when the system time was updated by SNTP.  https://github.com/espressif/esp-idf/blob/release/v3.3/examples/protocols/sntp/main/sntp_example_main.c#L151. Additional you can set a callback function for it `sntp_set_time_sync_notification_cb(time_sync_notification_cb);`
That SNTP API was ported to 3.3 not so far (2 months ago) in 07a3eca372c70c6c00d1b787f66f21efac24f015.

Seems into your app somehow the time is updated by other components? Is it possible?

burkulesomesh43, 2019-12-23:
Hi @KonstantinKondrashov,
I'll check for the NTP sync cb and get back to you.
I am updating time once when given power to esp32. after that I am calibrating time after every 24Hr's.

Alvin1Zhang, 2020-04-21:
@burkulesomesh43 Thanks for reporting and follow-up, will close this for now due to short of feedback, feel free to reopen if the issue still happens. Thanks.

closed 2020-04-21 by Alvin1Zhang