espressif/esp-idf #4473: Compile only necessary components for the project with CMake

mauriciofarina, 2019-12-11:
Hello,

I'm having some trouble to setup cmake in a way that only the necessary components for the it get compiled. From what I understand from the documentation, using `set(COMPONENTS main)`
in the project's `CMakeLists.txt` file would result in the wanted behaviour if all the components and main had their required components declared in their own `CMakeLists.txt` files (And, in my case they are). However, when I compile my project the whole IDF gets compiled.

I made a simple project to reproduce this:

- ESP-IDF: v4.1-dev-1086-g93a8603c5
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: macOS

File Structure:
```
- helloworld/
             - CMakeLists.txt
             - main/         - CMakeLists.txt
                             - theMain.c
```

helloworld/CMakeLists.txt
```
cmake_minimum_required(VERSION 3.5)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(helloworld)
set(COMPONENTS main)
```

helloworld/main/CMakeLists.txt
```
idf_component_register(SRCS "theMain.c"
                    INCLUDE_DIRS "")
```

helloworld/main/theMain.c
```
#include <stdio.h>

void app_main(void)
{
   printf("hello world");
}
```

renzbagaporo, 2019-12-12:
Hi, can you post the build log here?

mauriciofarina, 2019-12-17:
Hello @renzbagaporo 

Sorry for taking so long to replay,

Build Logs:

```
Executing action: all (aliases: build)
Running cmake in directory /Users/mauricio/esp/hello_world/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 /Users/mauricio/esp/hello_world"...
Warn about uninitialized values.
-- Found Git: /usr/bin/git (found version "2.21.0 (Apple Git-122.2)") 
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/mauricio/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /Users/mauricio/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /Users/mauricio/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Users/mauricio/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /Users/mauricio/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
-- Found PythonInterp: /Users/mauricio/.espressif/python_env/idf4.1_py2.7_env/bin/python (found version "2.7.16") 
-- Found Perl: /usr/bin/perl (found version "5.18.4") 
-- Adding linker script /Users/mauricio/esp/hello_world/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /Users/mauricio/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /Users/mauricio/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /Users/mauricio/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /Users/mauricio/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /Users/mauricio/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /Users/mauricio/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /Users/mauricio/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /Users/mauricio/esp/esp-idf/components/app_trace /Users/mauricio/esp/esp-idf/components/app_update /Users/mauricio/esp/esp-idf/components/asio /Users/mauricio/esp/esp-idf/components/bootloader /Users/mauricio/esp/esp-idf/components/bootloader_support /Users/mauricio/esp/esp-idf/components/bt /Users/mauricio/esp/esp-idf/components/cbor /Users/mauricio/esp/esp-idf/components/coap /Users/mauricio/esp/esp-idf/components/console /Users/mauricio/esp/esp-idf/components/cxx /Users/mauricio/esp/esp-idf/components/driver /Users/mauricio/esp/esp-idf/components/efuse /Users/mauricio/esp/esp-idf/components/esp-tls /Users/mauricio/esp/esp-idf/components/esp32 /Users/mauricio/esp/esp-idf/components/esp_adc_cal /Users/mauricio/esp/esp-idf/components/esp_common /Users/mauricio/esp/esp-idf/components/esp_eth /Users/mauricio/esp/esp-idf/components/esp_event /Users/mauricio/esp/esp-idf/components/esp_gdbstub /Users/mauricio/esp/esp-idf/components/esp_http_client /Users/mauricio/esp/esp-idf/components/esp_http_server /Users/mauricio/esp/esp-idf/components/esp_https_ota /Users/mauricio/esp/esp-idf/components/esp_https_server /Users/mauricio/esp/esp-idf/components/esp_local_ctrl /Users/mauricio/esp/esp-idf/components/esp_netif /Users/mauricio/esp/esp-idf/components/esp_ringbuf /Users/mauricio/esp/esp-idf/components/esp_rom /Users/mauricio/esp/esp-idf/components/esp_websocket_client /Users/mauricio/esp/esp-idf/components/esp_wifi /Users/mauricio/esp/esp-idf/components/espcoredump /Users/mauricio/esp/esp-idf/components/esptool_py /Users/mauricio/esp/esp-idf/components/expat /Users/mauricio/esp/esp-idf/components/fatfs /Users/mauricio/esp/esp-idf/components/freemodbus /Users/mauricio/esp/esp-idf/components/freertos /Users/mauricio/esp/esp-idf/components/heap /Users/mauricio/esp/esp-idf/components/idf_test /Users/mauricio/esp/esp-idf/components/jsmn /Users/mauricio/esp/esp-idf/components/json /Users/mauricio/esp/esp-idf/components/libsodium /Users/mauricio/esp/esp-idf/components/log /Users/mauricio/esp/esp-idf/components/lwip /Users/mauricio/esp/he [... cut ...]
├── [  87]  flash_bootloader_args
├── [  42]  flash_partition_table_args
├── [ 150]  flash_project_args
├── [ 799]  flasher_args.json
├── [144K]  helloworld.bin
├── [2.3M]  helloworld.elf
├── [1.6M]  helloworld.map
├── [2.5K]  kconfigs.in
├── [ 312]  kconfigs_projbuild.in
├── [4.1K]  ldgen_libraries
├── [1.8K]  ldgen_libraries.in
├── [  96]  partition_table
│   └── [3.0K]  partition-table.bin
├── [7.4K]  project_description.json
├── [   0]  project_elf_src.c
└── [ 68K]  rules.ninja

483 directories, 1094 files
```

renzbagaporo, 2019-12-18:
@mauriciofarina  Oh. You have to set `COMPONENTS` before the `project()` call.

mauriciofarina, 2019-12-18:
Great @renzbagaporo! Now things started to look more as they should!

I'm just having some trouble to figure out what are the minimal components needed for a project.

I got it to compile by using `set(COMPONENTS esptool_py main)`. However, I found that some people were recommending to use `set(COMPONENTS cxx app bootloader esptool_py main)`.
What should I use?

renzbagaporo, 2019-12-18:
@mauriciofarina Great question. 

I would say `bootloader` and `esptool_py` would be essential for a normal app. So in the top-level CMakeLists.txt, `main`, `bootloader`, `esptool_py` would have to be listed.

And in your `main` component's CMakeLists.txt, list its requirements explicitly using `REQUIRES`/`PRIV_REQUIRES`. The build system should figure out the rest of needed components based on this.

renzbagaporo, 2020-01-06:
Hi @mauriciofarina, have you had success with this issue?

mauriciofarina, 2020-01-08:
Hello @renzbagaporo!
Yes, my issue was solved!

closed 2020-01-08 by mauriciofarina