espressif/esp-idf #4474: Cant build Encryption flash, always partition orverlap or error

filzek, 2019-12-11:
- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: v4.1-dev-1086-g93a8603c5-dirty
-- Building ESP-IDF components for target esp32

-- Could NOT find Perl (missing: PERL_EXECUTABLE)
Traceback (most recent call last):
  File "C:/Users/dev/Desktop/esp-idf/components/partition_table/parttool.py", line 98, in __init__
    partition_table = gen.PartitionTable.from_binary(f.read())
  File "C:\Users\dev\Desktop\esp-idf\components\partition_table\gen_esp32part.py", line 231, in from_binary
    result.append(PartitionDefinition.from_binary(data))
  File "C:\Users\dev\Desktop\esp-idf\components\partition_table\gen_esp32part.py", line 383, in from_binary
    raise InputError("Invalid magic bytes (%r) for partition definition" % magic)
gen_esp32part.InputError: Invalid magic bytes (b'nv') for partition definition

seens that partitions overlap and problems, trying to do everything but always return erro.

projectgus, 2019-12-12:
Hi @filzek ,

Can you please give a little more detail: which configuration do you have set for flash encryption? what partition configuration do you have set, if it's a custom partition table then you share the contents of the CSV please? Also, what is the full output from the build both before and after the error output?

Thanks,

Angus

filzek, 2019-12-12:
send in private e-mail the files you requested

chegewara, 2019-12-12:
Maybe this is the case:
https://docs.espressif.com/projects/esp-idf/en/latest/security/secure-boot.html#bootloader-size

projectgus, 2019-12-16:
Hi @filzek,

Thanks for sending those files and logs. Will reply here as this may help other people who find similar problems.

> gen_esp32part.InputError: Invalid magic bytes (b'\r\n') for partition definition

You can ignore this error, it's a shortcoming in `parttool.py`  that this is logged. Will fix.

> gen_esp32part.InputError: CSV Error: First partition offset 0x9000 overlaps end of partition table 0xa000

If partition table offset is set to 0x9000, then the custom partition file (partitions.csv in your case) can't specify any partition at an offset less than 0xa000 - otherwise the partition collides with the partition table (which is 0x1000 bytes long).

Please the contents of your partitions.csv, and the partition offset value in sdkconfig.

filzek, 2019-12-16:
this is not the true problem, as even if try to build the example in esp-idf\examples\security\flash_encryption  it return the same error.

we did try to full remove esp-idf, uninstall esp-idf tool, and we reinstalled everything.

still same error.

now also, the idf.py menuconfig doesnt work.

C:\Users\filze\Desktop\esp-idf\examples\security\flash_encryption>idf.py menuconfig
Executing action: menuconfig
Traceback (most recent call last):
  File "C:\Users\filze\Desktop\esp-idf\tools\idf.py", line 764, in <module>
    main()
  File "C:\Users\filze\Desktop\esp-idf\tools\idf.py", line 682, in main
    cli(sys.argv[1:], prog_name=PROG)
  File "C:\Users\filze\.espressif\python_env\idf4.1_py3.7_env\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\filze\.espressif\python_env\idf4.1_py3.7_env\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "C:\Users\filze\.espressif\python_env\idf4.1_py3.7_env\lib\site-packages\click\core.py", line 1164, in invoke
    return _process_result(rv)
  File "C:\Users\filze\.espressif\python_env\idf4.1_py3.7_env\lib\site-packages\click\core.py", line 1102, in _process_result
    **ctx.params)
  File "C:\Users\filze\.espressif\python_env\idf4.1_py3.7_env\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "C:\Users\filze\Desktop\esp-idf\tools\idf.py", line 614, in execute_tasks
    task.run(ctx, global_args, task.action_args)
  File "C:\Users\filze\Desktop\esp-idf\tools\idf.py", line 207, in run
    self.callback(self.name, context, global_args, **action_args)
  File "C:\Users\filze\Desktop\esp-idf\tools\idf_py_actions\core_ext.py", line 40, in menuconfig
    os.environ['MENUCONFIG_STYLE'] = style.encode(sys.getfilesystemencoding() or 'utf-8')
  File "C:\Users\filze\.espressif\python_env\idf4.1_py3.7_env\lib\os.py", line 683, in __setitem__
    value = self.encodevalue(value)
  File "C:\Users\filze\.espressif\python_env\idf4.1_py3.7_env\lib\os.py", line 739, in check_str
    raise TypeError("str expected, not %s" % type(value).__name__)
TypeError: str expected, not bytes

C:\Users\filze\Desktop\esp-idf\examples\security\flash_encryption>

filzek, 2019-12-16:
Problem still, when following the example in flash_encryption the problem stills happen.

C:\Users\filze\Desktop\esp-idf\examples\security\flash_encryption>idf.py build flash monitor
Executing action: all (aliases: build)
Running ninja in directory c:\users\filze\desktop\esp-idf\examples\security\flash_encryption\build
Executing "ninja all"...
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Project version: v4.1-dev-1533-g9dbb7526c-dirty
-- Building ESP-IDF components for target esp32
Loading defaults file C:/Users/filze/Desktop/esp-idf/examples/security/flash_encryption/sdkconfig.defaults...
Traceback (most recent call last):
  File "C:/Users/filze/Desktop/esp-idf/components/partition_table/parttool.py", line 98, in __init__
    partition_table = gen.PartitionTable.from_binary(f.read())
  File "C:\Users\filze\Desktop\esp-idf\components\partition_table\gen_esp32part.py", line 231, in from_binary
    result.append(PartitionDefinition.from_binary(data))
  File "C:\Users\filze\Desktop\esp-idf\components\partition_table\gen_esp32part.py", line 383, in from_binary
    raise InputError("Invalid magic bytes (%r) for partition definition" % magic)
gen_esp32part.InputError: Invalid magic bytes (b'# ') for partition definition

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:/Users/filze/Desktop/esp-idf/components/partition_table/parttool.py", line 338, in <module>
    main()
  File "C:/Users/filze/Desktop/esp-idf/components/partition_table/parttool.py", line 311, in main
    target = ParttoolTarget(**target_args)
  File "C:/Users/filze/Desktop/esp-idf/components/partition_table/parttool.py", line 102, in __init__
    partition_table = gen.PartitionTable.from_csv(f.read())
  File "C:\Users\filze\Desktop\esp-idf\components\partition_table\gen_esp32part.py", line 122, in from_csv
    % (e.offset, last_end))
gen_esp32part.InputError: CSV Error: First partition offset 0x9000 overlaps end of partition table 0xa000
Traceback (most recent call last):
  File "C:/Users/filze/Desktop/esp-idf/components/partition_table/parttool.py", line 98, in __init__
    partition_table = gen.PartitionTable.from_binary(f.read())
  File "C:\Users\filze\Desktop\esp-idf\components\partition_table\gen_esp32part.py", line 231, in from_binary
    result.append(PartitionDefinition.from_binary(data))
  File "C:\Users\filze\Desktop\esp-idf\components\partition_table\gen_esp32part.py", line 383, in from_binary
    raise InputError("Invalid magic bytes (%r) for partition definition" % magic)
gen_esp32part.InputError: Invalid magic bytes (b'# ') for partition definition

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:/Users/filze/Desktop/esp-idf/components/partition_table/parttool.py", line 338, in <module>
    main()
  File "C:/Users/filze/Desktop/esp-idf/components/partition_table/parttool.py", line 311, in main
    target = ParttoolTarget(**target_args)
  File "C:/Users/filze/Desktop/esp-idf/components/partition_table/parttool.py", line 102, in __init__
    partition_table = gen.PartitionTable.from_csv(f.read())
  File "C:\Users\filze\Desktop\esp-idf\components\partition_table\gen_esp32part.py", line 122, in from_csv
    % (e.offset, last_end))
gen_esp32part.InputError: CSV Error: First partition offset 0x9000 overlaps end of partition table 0xa000
-- Adding linker script C:/Users/filze/Desktop/esp-idf/examples/security/flash_encryption/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/filze/Desktop/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Users/filze/Desktop/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/Users/filze/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/filze/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/filze/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/filze/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/filze/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
Traceback (most recent call last):
  File "C:/Users/filze/Desktop/esp-idf/components/partition_table/parttool.py", line 98, in __init__
    partition_table = gen.PartitionTable.from_binary(f.read())
  File "C:\Users\filze\Desktop\esp-idf\components\partition_table\gen_esp32part.py", line 231, in from_binary
    result.append(PartitionDefinition.from_binary(data))
  File "C:\Users\filze\Desktop\esp-idf\components\partition_table\gen_esp32part.py", line 383, in from_binary
    raise InputError("Invalid magic bytes (%r) for partition definition" % magic)
gen_esp32part.InputError: Invalid magic bytes (b'# ') for partition definition

[... 31 lines cut ...]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:/Users/filze/Desktop/esp-idf/components/partition_table/parttool.py", line 338, in <module>
    main()
  File "C:/Users/filze/Desktop/esp-idf/components/partition_table/parttool.py", line 311, in main
    target = ParttoolTarget(**target_args)
  File "C:/Users/filze/Desktop/esp-idf/components/partition_table/parttool.py", line 102, in __init__
    partition_table = gen.PartitionTable.from_csv(f.read())
[... cut ...]
See also "C:/Users/filze/Desktop/esp-idf/examples/security/flash_encryption/build/CMakeFiles/CMakeOutput.log".
FAILED: build.ninja
C:\Users\filze\.espressif\tools\cmake\3.13.4\bin\cmake.exe -SC:\Users\filze\Desktop\esp-idf\examples\security\flash_encryption -BC:\Users\filze\Desktop\esp-idf\examples\security\flash_encryption\build
ninja: error: rebuilding 'build.ninja': subcommand failed
ninja failed with exit code 1

C:\Users\filze\Desktop\esp-idf\examples\security\flash_encryption>

cant solve the problem

filzek, 2019-12-16:
We figure out all the problem, its beacuse its missing a basic information about the true offset for the initial partition table in menuconfig and the correct initial in the partition file, so, if fix both it will work great.

To solve the problem, the example  partitions_example.csv need to be fixed

# Name,   Type, SubType, Offset,  Size, Flags
nvs,        data, nvs,      0xaa00,  0x6000,  #the 0xaa00 example was incorrect related to 0x9000
# Extra partition to demonstrate reading/writing of encrypted flash
storage,    data, 0xff,     0x10A00,  0x1000, encrypted
factory,    app,  factory,  0x20000, 1M, encrypted

So, a mistake to increase both numbers in CSV and in menuconfig. Menuconfig 0x9000 is perfect, so, but in CSX it need to be at least 0xaa00 to do not colide.

Example and messages are showing incorrect. doing it will build and boot.

closed 2019-12-16 by filzek

projectgus, 2019-12-16:
@filzek Glad you figured out. Suggest aligning your partition starts to multiples of 0x1000, ie use 0xa000 instead of 0xaa00 and 0x10000 instead of 0x10a00.

Having first partition at 0xa000 with partition table at 0x9000 will not collide.