espressif/esp-idf #4475: esp_mesh_send is always blocking

sinope-db, 2019-12-11:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROVER
- IDF version: v3.2.3
- Build System:         Make
- Compiler version: 1.22.0-80-g6c4433a5
- Operating System:     Windows
- Power Supply:         external 3.3V

## Problem Description

Calling esp_mesh_send() with the flag MESH_DATA_NONBLOCK set (actual value used 0x32), the function is blocking if the device has left the mesh but the child disconnected event has not occurred yet.

I'm using ESP-MDF to send, but setting breakpoints in the mwifi_subcontract_write() function shows that esp_mesh_send() is blocking.

### Expected Behavior

esp_mesh_send() returns without waiting for send to complete.

### Actual Behavior

esp_mesh_send() blocks until the device is removed (inactive timer expire)

### Steps to repropduce

1. Setup mesh network with a fixed root.
2. connect device to mesh (type set to node)
3. Send and receive data to/from device successfully.
4. remove power from device added in step above
5. From the root node, send data to the device removed before the child disconnected event.

## Debug Logs

```
I (747576) [Mesh, 797]: Send, addr: 30:ae:a4:d3:80:74, size: 28, data:
D (747586) [mwifi, 848]: esp_mesh_send dest_addr: 30:ae:a4:d3:80:74, mesh_data.size: 28, data:
W (765816) wifi: inactive timer: now=2e7dc04e last_rx_time=2ca48196 diff=7926, aid[1]30:ae:a4:d3:80:74 leave
I (765816) wifi: station: 30:ae:a4:d3:80:74 leave, AID = 1, bss_flags is 134243, bss:0x3fffe870
I (765826) wifi: n:1 0, o:1 1, ap:1 1, sta:0 0, prof:1
WIFI event 17 -
D (765836) [mwifi, 131]: esp_mesh_event_cb event.id: 4
I (765856) [Mesh, 205]: event_loop_cb, event: 4
D (765856) [mwifi, 131]: esp_mesh_event_cb event.id: 6
I (765866) [mwifi, 226]: Routing table is changed by removing leave children remove_num: 1, total_num: 1
I (765876) [Mesh, 205]: event_loop_cb, event: 6
I (765876) mesh: [scan]new scanning time:300ms
I (765886) [Mesh, 280]: Routing table update; total_num: 1
W (765896) wifi: [beacon]new interval:100ms
W (765846) [mwifi, 644]: <ESP_ERR_MESH_DISCONNECTED> Node failed to send packets, dest_addr: 30:ae:a4:d3:80:74, flag: 0x32, opt->type: 0x08, opt->len: 13, data->tos: 0, data: 0x3fffe0fe, size: 28
W (765926) [mwifi, 898]: <ESP_ERR_MESH_DISCONNECTED> Node failed to send packets, data_flag: 0x32, dest_mac: 30:ae:a4:d3:80:74
E (765946) Mesh: Error sending to mesh (16395) ESP_ERR_MESH_DISCONNECTED
I (775906) mesh: [scan]new scanning time:600ms
W (775906) wifi: [beacon]new interval:300ms```

szmodz, 2019-12-11:
it is. see: https://github.com/espressif/esp-idf/issues/4407

szmodz, 2019-12-11:
https://github.com/espressif/esp-idf/issues/3694

szmodz, 2019-12-12:
Here's a test case:
https://github.com/szmodz/esp-idf/commit/7cf87821dfa37523ff8e3e0d979f7662f8b804eb#diff-cca269835969cdb944b869727126c9a7L191

dongdong004, 2019-12-12:
As far as I know, `MESH_DATA_NONBLOCK` only for  `ESP_ERR_MESH_QUEUE_FULL` and `ESP_ERR_MESH_XON_NO_WINDOW`.

This means that esp_mesh_send() behaves the same way without this flag unless it returns `ESP_ERR_MESH_QUEUE_FULL` or `ESP_ERR_MESH_XON_NO_WINDOW`.

sinope-db, 2019-12-12:
I have the same behavior on the nodes when sending to root and the task calling esp_mesh_recv_toDS() is inactive. The nodes get stuck on esp_mesh_send().

dongdong004, 2019-12-13:
There's no way to make sure that esp_mesh_send() doesn't block. But if you send packets downstream with `data.tos=MESH_TOS_DEF` and flag `MESH_DATA_NONBLOCK`, it can be non-block. Otherwise, it can block.

Alvin1Zhang, 2020-03-10:
@sinope-db Thanks for reporting. Will close due to short of feedback, feel free to reopen with more details. Thanks.

closed 2020-03-10 by Alvin1Zhang