espressif/esp-idf #4476: ledc set freq when LEDC_APB_CLK

chegewara, 2019-12-12:
I found this issue in AFR, which is copied from esp-idf:
https://github.com/espressif/esp-idf/blob/release/v3.3/components/driver/ledc.c#L443

This is output with additional logging:
```
W (329) ledc_set_freq: FREQ = 200, divider => 800000, precision = 128, REF => 256000000, APB => 20480000000
```

projectgus, 2019-12-16:
Hi @chegewara,

It's hard to understand all these incomplete bug reports in different places. Please use our issue template to explain the issue - it's there for a reason, even if you think the fields are not relevant they're helpful to the people who are triaging the issue and try to help you with it. 

Do I understand the issue correctly:

In ESP-IDF v3.3, calling `ledc_set_freq()` with clock source `LEDC_APB_CLK` causes the clock divider setting to be incorrect?

projectgus, 2019-12-16:
Sorry, some of what above is based on an incorrect understanding of the ledc driver.

In v3.x:

* If you need to pass low frequencies to `ledc_set_freq`, call `ledc_timer_set` first and pass `LEDC_REF_TICK` as the `clk_src` parameter (or set a low frequency in the initial call to `ledc_timer_config`

In v4.x:

* Set the clk_cfg parameter of the timer_conf structure `to LEDC_USE_AUTO_CLK` (default value) and then call `ledc_set_freq` with any values and the clock source will automatically switch depending on the frequency.

The `LEDC_USE_AUTO_CLK` is a breaking API behaviour change so it can't be backported to v3.x

chegewara, 2019-12-16:
I am using exactly the same code on both, v3.3 and master. On master it works perfectly, on v3.3 i am having problems. 
I will try to update A4988 driver according to your suggestions and i am accepting that API cant be backported. 
This is current log output with v3.3:
```
I (180) boot: Loaded app from partition at offset 0x10000
I (180) boot: Disabling RNG early entropy source...
I (181) cpu_start: Pro cpu up.
I (184) cpu_start: Application information:
I (189) cpu_start: Project name:     esp-idf
I (194) cpu_start: App version:      1
I (198) cpu_start: Compile time:     Dec 16 2019 08:12:32
I (205) cpu_start: ELF file SHA256:  3d6b0ac08f04f5c1...
I (211) cpu_start: ESP-IDF:          v3.3.1-4-g659ccfa99
I (217) cpu_start: Starting app cpu, entry point is 0x40080f04
0x40080f04: call_start_cpu1 at /home/chegewara/esp/esp-idf/components/esp32/cpu_start.c:269

I (0) cpu_start: App cpu up.
I (227) heap_init: Initializing. RAM available for dynamic allocation:
I (234) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (240) heap_init: At 3FFB2F30 len 0002D0D0 (180 KiB): DRAM
I (246) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (252) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (259) heap_init: At 40088714 len 000178EC (94 KiB): IRAM
I (265) cpu_start: Pro cpu start user code
I (283) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (285) ledc: ledc_set_duty_with_hpoint(383): hpoint argument is invalid
I (285) gpio: GPIO[14]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (295) stepper_a4988: <200> (1) [0] set freq: 200

E (305) setSpeedRpmCur: 0
I (305) stepper_a4988: <200> (1) [0] set freq: 600

E (315) setSpeedRpmCur: 0
I (335) stepper_a4988: <200> (1) [0] set freq: 800

E (335) setSpeedRpmCur: 0
I (355) stepper_a4988: <200> (1) [0] set freq: 1000

E (355) setSpeedRpmCur: 0
```

Will get back here if i have some progress. Thanks.

projectgus, 2019-12-16:
@chegewara Try passing speed_mode `LEDC_LOW_SPEED_MODE` here: https://github.com/espressif/esp-iot-solution/blob/master/components/motor/stepper/a4988/include/iot_a4988.h#L95

chegewara, 2019-12-16:
Ive been trying to reproduce working setup in past 2 hours and i failed with release/v3.3, release/v4.0 and master. It was my fault, because i forgot i have esp-idf-master folder with esp-idf. Here is output from `git show`:
```
commit 93a8603c545fb8e54741d6685146e2f3b874378d (HEAD -> master, origin/master, origin/HEAD)
Merge: a0aab6650 cd730b94b
Author: Ivan Grokhotkov <ivan@espressif.com>
Date:   Mon Nov 18 19:34:05 2019 +0800

    Merge branch 'feature/builtin_type_exceptions_test' into 'master'
    
    C++: Throw builtin type exception test
    
    See merge request espressif/esp-idf!6570
```

Here is my test code and as you can see i dont have to use LOW_SPEED mode:
```
void test() {
    int steps = 800*3;
    int rpm = 60*5;

    CA4988Stepper stepper(STEPPER_STEP, STEPPER_DIR, 200);

    stepper.setSpeedRpm(rpm);
    stepper.step(steps);
    vTaskDelay(1000/portTICK_RATE_MS);
    stepper.step(-1*steps);
    vTaskDelay(1000/portTICK_RATE_MS);

    stepper.setSpeedRpm(60);
    stepper.run(1);
    vTaskDelay(5000/portTICK_RATE_MS);
    stepper.run(-1);
    vTaskDelay(5000/portTICK_RATE_MS);
    stepper.stop();
}
```

Thanks for your help @projectgus

projectgus, 2019-12-16:
@chegewara On master you don't need any changes because the LEDC driver uses the name LEDC_USE_AUTO_CLK configuration by default.

On v3.x you will need to make some changes in the driver to set low PWM frequences.

chegewara, 2019-12-16:
> CA4988Stepper stepper(STEPPER_STEP, STEPPER_DIR, 200, LEDC_LOW_SPEED_MODE);

This is not working at all, its even worse than with HIGH SPEED. 

With normal use (HIGH SPEED) it is working partially, with some specific parameters set (RPM 200 for example). So, maybe its not issue with ledc and set freq, but i cant adjust motor speed.

Its a bit disappointing it is working with that commit 93a8603c545fb8e54741d6685146e2f3b874378d, with master it wont even compile, but i cant make it work with v3.3 which is LTS and A4988 driver is from espressif.

projectgus, 2019-12-16:
@chegewara Why did you close the issue if the problem still exists?

chegewara, 2019-12-16:
Because i cant find the problem. 
It is not working like expected, i cant control RPM, but i feel like every day i am performing tests it behaves differently. 
I accept that working solution cant be backported to v3.3, so i have to decide if i will accept that i cant control RPM or i will try to find/write other driver, because arduino drivers works great.

koobest, 2019-12-17:
Hi, chegewara, 
Did you initialize the `hpoint` member of ledc_`channel_config_t`, maybe you need to initialize to 0, otherwise it will be an indeterminate value.
```
typedef struct {
    int gpio_num;                   /*!< the LEDC output gpio_num, if you want to use gpio16, gpio_num = 16 */
    ledc_mode_t speed_mode;         /*!< LEDC speed speed_mode, high-speed mode or low-speed mode */
    ledc_channel_t channel;         /*!< LEDC channel (0 - 7) */
    ledc_intr_type_t intr_type;     /*!< configure interrupt, Fade interrupt enable  or Fade interrupt disable */
    ledc_timer_t timer_sel;         /*!< Select the timer source of channel (0 - 3) */
    uint32_t duty;                  /*!< LEDC channel duty, the range of duty setting is [0, (2**duty_resolution)] */
    int hpoint;                     /*!< LEDC channel hpoint value, the max value is 0xfffff */
} ledc_channel_config_t;
```
thanks !!

chegewara, 2019-12-17:
Yes, i saw error log about hpoint and i added initialization to A4988 driver. It did not help.

Maybe its just over usage to use ledc to control stepper motor, but i was happy when i found espressif A4988 driver. Now i feel like i was fooled. 
I am really big fan of esp32, but now i am just sad.

xiongyumail, 2019-12-30:
@chegewara 
At such a low frequency, I think you should use `MCPWM`, LEDC is usually used for lighting control.
Or LEDC's peripheral clock uses `LEDC_USE_RTC8M_CLK`

chegewara, 2019-12-31:
Thanks for constructive answers, but i dont understand why should i change LEDC to MCPWM since it is espressif driver for A4988 and i know it should works, because it is working with some esp-idf commits. This particular commit i am testing against works exactly how it should (motor speed control etc) 93a8603c545fb8e54741d6685146e2f3b874378d

xiongyumail, 2019-12-31:
In my opinion, iot-solution just provides a solution idea, it may not cover all application scenarios, but it can better let us understand how to do it.

chegewara, 2019-12-31:
Ok, that is good enough explanation, but the problem is it is working with some esp-idf commits and does not work with esp-idf v3.3 or master.

songruo, 2022-02-24:
@chegewara I wasn't able to fully understand the conversation going-on here, but for the initial post, the warning

W (329) ledc_set_freq: FREQ = 200, divider => 800000, precision = 128, REF => 256000000, APB => 20480000000

can be eliminated if you increase the ledc duty resolution in the initial call to` ledc_timer_config`. From the warning, it looks like `LEDC_TIMER_7_BIT` is used. If you select `LEDC_TIMER_13_BIT`, then 200Hz signal can be generated even with `LEDC_APB_CLK`. 

I have checked the clock divider calculation on the v3.3 ledc driver, and didn't find anything suspicious. Maybe the issue is on the A4988 driver side? Also IDF v3.3 and v4.0 are end of life now. We are going to close this issue, if you think this problem does exist in the later IDF versions, please feel free to reopen this issue with some more details.

closed 2022-02-24 by songruo