espressif/esp-idf #4477: esp_wifi_set_max_tx_power(82)

AddisAir, 2019-12-12:
According to the [docs](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/network/esp_wifi.html#_CPPv425esp_wifi_set_max_tx_power6int8_t), esp_wifi_set_max_tx_power(82) can be set up to 82.

However, when you do that, it just stays on 78. 
Is this a bug? or an error in the documentation? 

I understand there are legal limits for TX power, but there are situations where it doesn't apply.

chegewara, 2019-12-12:
https://github.com/espressif/esp-idf/issues/4223

esp-HarveyRong, 2020-01-19:
Hi @AddisAir  At present, the maximum value can only be 78, and values greater than 78 will also be set to 78.

Alvin1Zhang, 2020-05-27:
@AddisAir Thanks for reporting. Would you please help share if any updates for the issue? Has your issue been resolved? Thanks.

Alvin1Zhang, 2020-07-30:
Thanks for reporting and sorry for slow turnaround, the fix is available at https://github.com/espressif/esp-idf/commit/695ad84654188ffa970760fa1c307e119178ba8c. Feel free to reopen if the issue still happens. Thanks.

closed 2020-07-30 by Alvin1Zhang